espressif/arduino-esp32 #4715: I2C pin 18 and 5 Not working with Arduino-ESP32

AtefMohamed, 2021-01-14:
Hello,
Unfouranatly I finished my PCB and got manufactured. connecting the i2c bus of the TPS88004 Ti sensor. with IO18 and IO5 pins as they said that almost all pins can be configured as I2C.

with scanner not device is detected. 
with read, data cannot connect to the device. 

----------------------------- Remove above -----------------------------

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32-WROOM-32 Chip
|<B>Version/Date</B>|1.0.4|2.0.0|0badbeef|11/jul/2017|today's master|
|<B>IDE name</B>|Arduino IDE
|<B>Flash Frequency</B>|40Mhz
|<B>PSRAM enabled</B>|no|
|<B>Upload Speed</B>|115200|
|<B>Computer OS</B>|Windows 10|

### Description:
Describe your problem here

/*********
  Rui Santos
  Complete project details at https://randomnerdtutorials.com  
*********/

#include <Wire.h>
#define I2C_SDA 18
#define I2C_SCL 5
void setup() {
  Wire.begin(I2C_SDA, I2C_SCL);
  Serial.begin(9600);
  Serial.println("\nI2C Scanner");
}
 
void loop() {
  byte error, address;
  int nDevices;
  Serial.println("Scanning...");
  nDevices = 0;
  for(address = 1; address < 127; address++ ) {
    Wire.beginTransmission(address);
    error = Wire.endTransmission();
    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
      nDevices++;
    }
    else if (error==4) {
      Serial.print("Unknow error at address 0x");
      if (address<16) {
        Serial.print("0");
      }
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  }
  else {
    Serial.println("done\n");
  }
  delay(5000);          
}

felmue, 2021-01-14:
Hello @AtefMohamed 

unless GPIO18 and GPIO5 are used otherwise in your setup I don't see an issue using them for I2C. I ran a test with GPIO18 and GPIO5 on my ESP32 board and I2C works just fine.

- did you add the required pull-up resistors on the SDA and SCL lines?
- have you tested your Ti sensor to be working ok?
- do you have another I2C slave device you could try?

Thanks
Felix

chegewara, 2021-01-14:
You have to be more accurate what arduino-esp32 version you are using, v1.0.4 or master branch.
With master branch this may fail:
`Wire.begin(I2C_SDA, I2C_SCL);`
so my advice is to add check if returns true or not.

AtefMohamed, 2021-01-14:
> Hello @AtefMohamed
>
> unless GPIO18 and GPIO5 are used otherwise in your setup I don't see an issue using them for I2C. I ran a test with GPIO18 and GPIO5 on my ESP32 board and I2C works just fine. [...]

yes i used 33K ohm resistor as stated in the data sheet of TPS8804.

. No i don't know how to test this sensor.

yes i will try with another i2c device.

AtefMohamed, 2021-01-14:
> You have to be more accurate what arduino-esp32 version you are using, v1.0.4 or master branch.
> With master branch this may fail:
> `Wire.begin(I2C_SDA, I2C_SCL);`
> so my advice is to add check if returns true or not.

Can you clear how to use check returns using this library?  
if you need arduino IDE version i use 1.8.13 .

AbdallahAwdalla, 2021-01-14:
> Can you clear how to use check returns using this library?
> if you need arduino IDE version i use 1.8.13 .

Hi Atef 

you could use 
`Serial.println(Wire.begin(I2C_SDA, I2C_SCL);`
and check the return value (0 or 1 );

AtefMohamed, 2021-05-25:
ESP32 burnt.

closed 2021-05-25 by AtefMohamed