espressif/arduino-esp32 #4716: With ESP32S2, SD card read works in main loop, but will crash when called in a subroutine.

RudyFiero, 2021-01-14:
### Hardware:
Board:							ESP32-S2-Saola-1M
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						Arduino IDE 1,8.13
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I can read a file on a SD card if I try and read it from Setup() or in Loop(). But if I have the same routine in a subroutine I get a crash. 

Sorry if I have not done a good enough job with the formatting. I always have difficulty with it, but I can't put the time in to fix it. I should have been in bed a couple of hours ago. 

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
/*
   Connect the SD card to the following pins:

   SD Card | ESP32
      D2       -
      D3       SS
      CMD      MOSI
      VSS      GND
      VDD      3.3V
      CLK      SCK
      VSS      GND
      D0       MISO
      D1       -
*/
#include "FS.h"
#include "SD.h"
#include "SPI.h"

int    DACvalue = 0;
int count = 0;

//-------------------------------------------
#define VSPI FSPI
static const int spiClk = 1000000; // 1 MHz

//uninitalised pointers to SPI objects
SPIClass * vspi = NULL;

//-------------------------------------------
void doFile() {

  File profile = SD.open("/foo.txt", FILE_READ);

  if (!profile) {
    Serial.println("Opening file to read failed");
    return;
  }

  Serial.println("File Content:");

  while (profile.available()) {
    Serial.write(profile.read());
    delay(150);
  }
  profile.close();

  Serial.println("");
  Serial.println("File read done");
  Serial.println("=================");
}

//===========================================
void setup() {
  Serial.begin(115200);

  //------------------------------------
  SPIClass SDSPI(HSPI);

#define SD_miso  17
#define SD_mosi  16
#define SD_sck   15
#define SD_ss    14

  SDSPI.begin(SD_sck, SD_miso, SD_mosi, -1);
  pinMode(SD_ss, OUTPUT);        //HSPI SS  set slave select pins as output

  if (!SD.begin(SD_ss, SDSPI)) {
    Serial.println("Card Mount Failed");
    return;
  }
  //------------------------------------

  File profile = SD.open("/foo.txt", FILE_READ);

  if (!profile) {
    Serial.println("Opening file to read failed");
    return;
  }

  Serial.println("File Content:");

  while (profile.available()) {

    while (profile.available()) {
      Serial.write(profile.read());
      delay(50);
    }8.13

    Serial.println("");
    Serial.println("File read done");
    Serial.println("=================");
  }
}
  //===========================================
  void loop() {
    count++;

    if (count == 10) {
        doFile();      // This will cause a crash, but the same routine below works.
/*
//----------- 
  File profile = SD.open("/foo.txt", FILE_READ);

  if (!profile) {
    Serial.println("Opening file to read failed");
    return;
  }

  Serial.println("File Content:");

  while (profile.available()) {
    Serial.write(profile.read());
    delay(50);
  }
  profile.close();

  Serial.println("");
  Serial.println("File read done");
  Serial.println("=================");

//-----------
*/
    }
    Serial.println(count);
    delay(1000);

  }
```
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x608
load:0x4004c000,len:0xa38
load:0x40050000,len:0x2848
entry 0x4004c190

```

7
8
9
File Content:
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c:1440 (xQueueSemaphoreTake)- assert failed!

abort() was called at PC 0x40029da5 on core 0

Backtrace:0x40027642:0x3ffc6ac0 0x40027f29:0x3ffc6ae0 0x4002c8ed:0x3ffc6b00 0x40029da5:0x3ffc6b80 0x40084858:0x3ffc6bc0 0x40083aa4:0x3ffc6be0 0x40082e92:0x3ffc6c10 0x4008cafd:0x3ffc6c40 0x4008eb2f:0x3ffc6c60 0x40090308:0x3ffc6ca0 0x40089bf2:0x3ffc6cd0 0x4001a1cb:0x3ffc6cf0 |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40025f79
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x8
load:0x3ffe6108,len:0x608
load:0x4004c000,len:0xa38
load:0x40050000,len:0x2848
entry 0x4004c190
File Content:
1000
1100

```

me-no-dev, 2021-01-14:
that `}8.13` in the subroutine I would guess is a typo? Can you decode the backtrace please?

RudyFiero, 2021-01-14:
Yeah. I was trying to enter the Arduino version and I was starting up a new session with the IDE and it changed the focus. Something like that. 

I have not done much with ESP32s yet. I don't know how to do a backtrace yet, I have not installed and decoder. I'm assuming it is like the ESP8266, if so then I should have no problem. I can't look at this until tonight. Right now I need to get this working in loop(). I was hoping to demo something to my managers today.

RudyFiero, 2021-01-14:
In my application I tried moving the file read routine into loop() and have a flag to trigger it. It doesn't work. So maybe it working in loop() with the above code isn't triggering the problem. 

This is from the code above. 

`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400a43ee  PS      : 0x00060e30  A0      : 0x80082fd0  A1      : 0x3ffc6e00  
A2      : 0x0064732f  A3      : 0x00000000  A4      : 0x3ffc775c  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x3ffc75e0  A8      : 0x8002a513  A9      : 0x3ffc6de0  
A10     : 0x3ffc775c  A11     : 0x00000000  A12     : 0x3ffc2574  A13     : 0x00000080  
A14     : 0xfffffffc  A15     : 0x3ffc78e4  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00647343  LBEG    : 0x3ffc2574  LEND    : 0x00000080  LCOUNT  : 0x40026e75  

Backtrace:0x400a43eb:0x3ffc6e00 0x40082fcd:0x3ffc6e20 0x400826ca:0x3ffc6e50 0x4008b72d:0x3ffc6e80 0x4008d38c:0x3ffc6ea0 0x4008eafd:0x3ffc6ee0 0x40086906:0x3ffc6f10 0x4001a1cb:0x3ffc6f30 |<-CORRUPTED

`

`PC: 0x400a43ee: spiGetClockDiv at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-spi.c line 427
EXCVADDR: 0x00647343

Decoding stack results
0x400a43eb: spiGetClockDiv at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-spi.c line 427
0x40082fcd: SPIClass::beginTransaction(SPISettings) at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src\SPI.cpp line 137
0x400826ca: ff_sd_read(unsigned char, unsigned char*, unsigned int, unsigned int) at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SPI\src/SPI.h line 32
0x4008b72d: ff_disk_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/diskio/diskio.c line 70
0x4008d38c: f_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/src/ff.c line 3811
0x4008eafd: vfs_fat_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/vfs/vfs_fat.c line 358
0x40086906: esp_vfs_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/vfs/vfs.c line 446
`

RudyFiero, 2021-01-14:
The following is from the crash of my application code. In this I have the file read in loop() and triggered with a flag.

`Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400af8d1  PS      : 0x00060430  A0      : 0x80083e30  A1      : 0x3ffca500  
A2      : 0x0000ffff  A3      : 0x3ffca9d0  A4      : 0x3ffcaa44  A5      : 0x3fd80024  
A6      : 0x0000000b  A7      : 0x3ffc6120  A8      : 0x800943e7  A9      : 0x3ffca610  
A10     : 0x3ffca62c  A11     : 0x3ffca8b4  A12     : 0x3ffca860  A13     : 0x3fd81080  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000013  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000ffff  LBEG    : 0x3ffca860  LEND    : 0x3fd81080  LCOUNT  : 0x40027405  

Backtrace:0x400af8ce:0x3ffca500 0x40083e2d:0x3ffca520 0x4008352a:0x3ffca550 0x40090d01:0x3ffca580 0x40091041:0x3ffca5a0 0x40092278:0x3ffca5c0 0x40092395:0x3ffca5e0 0x40093002:0x3ffca610 0x400943e4:0x3ffca880 0x4008a2fe:0x3ffca9f0 0x400a20f9:0x3ffcaa10 0x40082be5:0x3ffcaa30 0x40082248:0x3ffcaaa0 0x40081f11:0x3ffcaad0 0x400888c9:0x3ffcab20 0x4002b30d:0x3ffcab40
`

`PC: 0x400af8d1
EXCVADDR: 0x0000ffff

Decoding stack results
0x40083e2d: spiTransaction at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-spi.c line 906
0x4008352a: psramInit at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\cores\esp32\esp32-hal-psram.c line 71
0x40090d01: uw_init_context_1 at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32s2-elf/src/gcc/libgcc/config/xtensa/unwind-dw2-xtensa.c line 473
0x40091041: _Unwind_GetLanguageSpecificData at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32s2-elf/src/gcc/libgcc/config/xtensa/unwind-dw2-xtensa.c line 178
0x40092278: __cxxabiv1::__gxx_personality_v0(int, _Unwind_Action, _Unwind_Exception_Class, _Unwind_Exception*, _Unwind_Context*) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32s2-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc line 320
0x40092395: __cxxabiv1::__cxa_call_unexpected(void*) at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32s2-elf/src/gcc/libstdc++-v3/libsupc++/eh_personality.cc line 731
0x40093002: fread at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/fread.c line 266
0x400943e4: _strtoul_l at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdlib/strtoul.c line 153
0x4008a2fe: esp_spiram_get_size at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp32s2/spiram.c line 312
0x400a20f9: __ssvfiscanf_r at /builds/idf/crosstool-NG/.build/HOST-i686-w64-mingw32/xtensa-esp32s2-elf/src/newlib/newlib/libc/stdio/vfscanf.c line 893
0x40082be5: ff_sd_initialize(unsigned char) at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SD\src\sd_diskio.cpp line 526
0x40082248: sdDeselectCard(unsigned char) at E:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\libraries\SD\src\sd_diskio.cpp line 89
0x40081f11: VFSImpl::open(char const*, char const*) at e:\arduino-1.8.13-windows\arduino-1.8.13\portable\packages\esp32\hardware\esp32\1.0.4\tools\xtensa-esp32s2-elf\xtensa-esp32s2-elf\include\c++\8.2.0\bits/shared_ptr_base.h line 121
0x400888c9: nvs::PageManager::load(unsigned int, unsigned int) at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/nvs_flash/src/intrusive_list.h line 120
0x4002b30d: verify_allocated_region at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c line 116
`

RudyFiero, 2021-01-14:
In my application (the above post) I have it do (inline) the file read in setup() without a crash. But in loop() it causes the crash in the above post.

RudyFiero, 2021-01-15:
The decoded stack trace I provided above was from an older portable install I was using for the ESP32S2. I had done a more current download and had verified that it also behaved the same as the older code. I have now decoded my test code with the the newer ESP32S2 code. 

`Backtrace:0x40027642:0x3ffc6a80 0x40027f29:0x3ffc6aa0 0x4002c8ed:0x3ffc6ac0 0x40029da5:0x3ffc6b40 0x40083c14:0x3ffc6b80 0x40082fcc:0x3ffc6ba0 0x400826aa:0x3ffc6bd0 0x4008bbcd:0x3ffc6c00 0x4008dbff:0x3ffc6c20 0x4008f3d8:0x3ffc6c60 0x40088cc2:0x3ffc6c90 0x4001a1cb:0x3ffc6cb0 |<-CORRUPTED
`
`0x40027642: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c line 361
0x40027f29: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/system_api.c line 106
0x4002c8ed: abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/abort.c line 46
0x40029da5: xQueueSemaphoreTake at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/queue.c line 1440
0x40083c14: spiTransaction at F:\arduino-1.8.13-windows\arduino-1.8.13\portable\sketchbook\hardware\espressif\esp32\cores\esp32\esp32-hal-spi.c line 897
0x40082fcc: SPIClass::beginTransaction(SPISettings) at F:\arduino-1.8.13-windows\arduino-1.8.13\portable\sketchbook\hardware\espressif\esp32\libraries\SPI\src\SPI.cpp line 142
0x400826aa: ff_sd_read(unsigned char, unsigned char*, unsigned int, unsigned int) at F:\arduino-1.8.13-windows\arduino-1.8.13\portable\sketchbook\hardware\espressif\esp32\libraries\SPI\src/SPI.h line 32
0x4008bbcd: ff_disk_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/diskio/diskio.c line 70
0x4008dbff: f_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/src/ff.c line 3811
0x4008f3d8: vfs_fat_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/fatfs/vfs/vfs_fat.c line 358
0x40088cc2: esp_vfs_read at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/vfs/vfs.c line 448
`

`/*
   Connect the SD card to the following pins:

   SD Card | ESP32
      D2       -
      D3       SS
      CMD      MOSI
      VSS      GND
      VDD      3.3V
      CLK      SCK
      VSS      GND
      D0       MISO
      D1       -
*/
#include "FS.h"
#include "SD.h"
#include "SPI.h"

int    DACvalue = 0;
int count = 0;

//-------------------------------------------
#define VSPI FSPI
static const int spiClk = 1000000; // 1 MHz

//uninitalised pointers to SPI objects
SPIClass * vspi = NULL;

//-------------------------------------------
void doFile() {

  File profile = SD.open("/foo.txt", FILE_READ);

  if (!profile) {
    Serial.println("Opening file to read failed");
    return;
  }

  Serial.println("File Content:");

  while (profile.available()) {
    Serial.write(profile.read());
    delay(150);
  }
  profile.close();

  Serial.println("");
  Serial.println("File read done");
  Serial.println("=================");
}

//===========================================
void setup() {
  Serial.begin(115200);

  //------------------------------------
  SPIClass SDSPI(HSPI);

#define SD_miso  17
#define SD_mosi  16
#define SD_sck   15
#define SD_ss    14

  SDSPI.begin(SD_sck, SD_miso, SD_mosi, -1);
  pinMode(SD_ss, OUTPUT);        //HSPI SS  set slave select pins as output

  if (!SD.begin(SD_ss, SDSPI)) {
    Serial.println("Card Mount Failed");
    return;
  }
  //------------------------------------

  File profile = SD.open("/foo.txt", FILE_READ);

  if (!profile) {
    Serial.println("Opening file to read failed");
    return;
  }

  Serial.println("File Content:");

  while (profile.available()) {

    while (profile.available()) {
      Serial.write(profile.read());
      delay(50);
    }

    Serial.println("");
    Serial.println("File read done");
    Serial.println("=================");
  }
}
  //===========================================
  void loop() {
    count++;

    if (count == 10) {
        doFile();      // This will cause a crash, but the same routine below works.
/*
//----------- 
  File profile = SD.open("/foo.txt", FILE_READ);

  if (!profile) {
    Serial.println("Opening file to read failed");
    return;
  }

  Serial.println("File Content:");

  while (profile.available()) {
    Serial.write(profile.read());
    delay(50);
  }
  profile.close();

  Serial.println("");
  Serial.println("File read done");
  Serial.println("=================");

//-----------
*/
    }
    Serial.println(count);
    delay(1000);

  }`

venice1200, 2021-03-13:
I had a similar issue.
Move the „SPIClass SDSPI(HSPI);“ to the top of the sketch.

See:
https://github.com/venice1200/MiSTer_tty2oled/blob/main/MiSTer_SSD1322_SD/MiSTer_SSD1322_SD.ino

Maybe it helps.

closed 2021-07-09 by stale[bot]

AllanOricil, 2024-03-11:
Could someone help me to fix this error https://github.com/AllanOricil/esp32-mfa-totp-generator/pull/9
I cant read/write/append files anywhere in the loop. It only works inside the setup. I have no idea why this happens.