espressif/arduino-esp32 #4727: Why can't you close the file handle opened with SPIFFS.open("/") without doing damage?

scubachristopher, 2021-01-17:
Hi,

I am simply traversing the SPIFFS file system, as per your example, and printing the heap before and after.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			Core 5.0.4·Home 3.3.1
IDE name:						VS Code
Flash Frequency:					40Mhz
PSRAM enabled:                                      dunno
Upload Speed:					115200
Computer OS:                                          Mac big sur

### Description:
Once I close the file handle associated with SPIFFS.open("/"), heap_caps_print_heap_info(MALLOC_CAP_DEFAULT) never reports heap information.  If you don't close the file, it will leak.

```
#include "FS.h"
#include "SPIFFS.h"

#define FORMAT_SPIFFS_IF_FAILED true

void setup(){
    Serial.begin(115200);
    if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
        Serial.println("SPIFFS Mount Failed");
        return;
    }

    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    File root = SPIFFS.open("/");
    File file = root.openNextFile();
  
    while (file) {
        Serial.printf("FILE: %18.18s, size: %6d bytes\n", file.name(), file.size());
        file = root.openNextFile();
    }
  
    //
    // If this close is called, heap_caps_print_heap_info will never print anything
    //
    close(root);

    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    Serial.printf("after close\n");
}

void loop(){
}
```

### Debug Messages:
```
15:08:06.338 ->  At 0x3ffb8000`⸮⸮eɕ⸮⸮916
15:08:06.338 ->     largest_free_block 916 alloc_blocks 18 free_blocks 1 total_blocks 19
15:08:06.338 ->   At 0x3ffb0000 len 25480 free 17172 allocated 8228 min_free 17172
15:08:06.338 ->     largest_free_block 17172 alloc_blocks 2 free_blocks 1 total_blocks 3
15:08:06.369 ->   At 0x3ffae6e0 len 6192 free 6092 allocated 36 min_free 6092
15:08:06.369 ->     largest_free_block 6092 alloc_blocks 1 free_blocks 1 total_blocks 2
15:08:06.369 ->   At 0x3ffaff10 len 240 free 0 allocated 120 min_free 0
15:08:06.369 ->     largest_free_block 0 alloc_blocks 5 free_blocks 1 total_blocks 6
15:08:06.369 ->   At 0x3ffb6388 len 7288 free 0 allocated 6920 min_free 0
15:08:06.369 ->     largest_free_block 0 alloc_blocks 21 free_blocks 0 total_blocks 21
15:08:06.407 ->   At 0x3ffb9a20 len 16648 free 6100 allocated 9832 min_free 600
15:08:06.407 ->     largest_free_block 4980 alloc_blocks 41 free_blocks 4 total_blocks 45
15:08:06.407 ->   At 0x3ffc17d8 len 124968 free 124920 allocated 0 min_free 124920
15:08:06.407 ->     largest_free_block 124920 alloc_blocks 0 free_blocks 1 total_blocks 1
15:08:06.407 ->   At 0x3ffe0440 len 15072 free 15024 allocated 0 min_free 15024
15:08:06.407 ->     largest_free_block 15024 alloc_blocks 0 free_blocks 1 total_blocks 1
15:08:06.441 ->   At 0x3ffe4350 len 113840 free 113792 allocated 0 min_free 113792
15:08:06.441 ->     largest_free_block 113792 alloc_blocks 0 free_blocks 1 total_blocks 1
15:08:06.441 ->   Totals:
15:08:06.441 ->     free 284020 allocated 30572 min_free 278520 largest_free_block 124920
15:08:06.517 -> FILE:  /httpcert.pem.der, size:   1011 bytes
15:08:06.517 -> FILE:       /index.lrigz, size:  52446 bytes
15:08:06.517 -> FILE:    /client.key.der, size:   1191 bytes
15:08:06.517 -> FILE:        /client.crt, size:   1265 bytes
15:08:06.551 -> FILE:            /ca.crt, size:   1424 bytes
15:08:06.551 -> FILE:        /ca.crt.der, size:   1011 bytes
15:08:06.551 -> FILE:    /client.crt.der, size:    894 bytes
15:08:06.551 -> FILE:      /httpcert.pem, size:   1424 bytes
15:08:06.551 -> FILE:        /client.key, size:   1675 bytes
15:08:06.584 -> after close
```

Here is a run with close(root) commented out:
```
15:12:50.634 -> e_block 4 alloc_bl⸮⸮⸮⸮⸮⸮ffb8000 len 6688 free 916 allocated 5436 min_free 916
15:12:50.634 ->     largest_free_block 916 alloc_blocks 18 free_blocks 1 total_blocks 19
15:12:50.634 ->   At 0x3ffb0000 len 25480 free 17172 allocated 8228 min_free 17172
15:12:50.634 ->     largest_free_block 17172 alloc_blocks 2 free_blocks 1 total_blocks 3
15:12:50.634 ->   At 0x3ffae6e0 len 6192 free 6092 allocated 36 min_free 6092
15:12:50.634 ->     largest_free_block 6092 alloc_blocks 1 free_blocks 1 total_blocks 2
15:12:50.669 ->   At 0x3ffaff10 len 240 free 0 allocated 120 min_free 0
15:12:50.669 ->     largest_free_block 0 alloc_blocks 5 free_blocks 1 total_blocks 6
15:12:50.669 ->   At 0x3ffb6388 len 7288 free 0 allocated 6920 min_free 0
15:12:50.669 ->     largest_free_block 0 alloc_blocks 21 free_blocks 0 total_blocks 21
15:12:50.669 ->   At 0x3ffb9a20 len 16648 free 6100 allocated 9832 min_free 596
15:12:50.669 ->     largest_free_block 4980 alloc_blocks 41 free_blocks 4 total_blocks 45
15:12:50.702 ->   At 0x3ffc17d8 len 124968 free 124920 allocated 0 min_free 124920
15:12:50.702 ->     largest_free_block 124920 alloc_blocks 0 free_blocks 1 total_blocks 1
15:12:50.702 ->   At 0x3ffe0440 len 15072 free 15024 allocated 0 min_free 15024
15:12:50.702 ->     largest_free_block 15024 alloc_blocks 0 free_blocks 1 total_blocks 1
15:12:50.702 ->   At 0x3ffe4350 len 113840 free 113792 allocated 0 min_free 113792
15:12:50.702 ->     largest_free_block 113792 alloc_blocks 0 free_blocks 1 total_blocks 1
15:12:50.740 ->   Totals:
15:12:50.740 ->     free 284020 allocated 30572 min_free 278516 largest_free_block 124920
15:12:50.816 -> FILE:  /httpcert.pem.der, size:   1011 bytes
15:12:50.816 -> FILE:       /index.lrigz, size:  52446 bytes
15:12:50.816 -> FILE:    /client.key.der, size:   1191 bytes
15:12:50.816 -> FILE:        /client.crt, size:   1265 bytes
15:12:50.816 -> FILE:            /ca.crt, size:   1424 bytes
15:12:50.816 -> FILE:        /ca.crt.der, size:   1011 bytes
15:12:50.816 -> FILE:    /client.crt.der, size:    894 bytes
15:12:50.854 -> FILE:      /httpcert.pem, size:   1424 bytes
15:12:50.854 -> FILE:        /client.key, size:   1675 bytes
15:12:50.854 -> Heap summary for capabilities 0x00001000:
15:12:50.854 ->   At 0x3ffbdb28 len 52 free 0 allocated 4 min_free 0
15:12:50.892 ->     largest_free_block 0 alloc_blocks 1 free_blocks 0 total_blocks 1
15:12:50.892 ->   At 0x3ffb8000 len 6688 free 340 allocated 5940 min_free 0
15:12:50.892 ->     largest_free_block 264 alloc_blocks 22 free_blocks 3 total_blocks 25
15:12:50.892 ->   At 0x3ffb0000 len 25480 free 16624 allocated 8736 min_free 16292
15:12:50.892 ->     largest_free_block 16400 alloc_blocks 4 free_blocks 3 total_blocks 7
15:12:50.892 ->   At 0x3ffae6e0 len 6192 free 6092 allocated 36 min_free 6092
15:12:50.892 ->     largest_free_block 6092 alloc_blocks 1 free_blocks 1 total_blocks 2
15:12:50.930 ->   At 0x3ffaff10 len 240 free 0 allocated 120 min_free 0
15:12:50.930 ->     largest_free_block 0 alloc_blocks 5 free_blocks 1 total_blocks 6
15:12:50.930 ->   At 0x3ffb6388 len 7288 free 0 allocated 6920 min_free 0
15:12:50.930 ->     largest_free_block 0 alloc_blocks 21 free_blocks 0 total_blocks 21
15:12:50.930 ->   At 0x3ffb9a20 len 16648 free 6100 allocated 9832 min_free 596
15:12:50.967 ->     largest_free_block 4980 alloc_blocks 41 free_blocks 4 total_blocks 45
15:12:50.967 ->   At 0x3ffc17d8 len 124968 free 124920 allocated 0 min_free 124920
15:12:50.967 ->     largest_free_block 124920 alloc_blocks 0 free_blocks 1 total_blocks 1
15:12:50.967 ->   At 0x3ffe0440 len 15072 free 15024 allocated 0 min_free 15024
15:12:50.967 ->     largest_free_block 15024 alloc_blocks 0 free_blocks 1 total_blocks 1
15:12:50.967 ->   At 0x3ffe4350 len 113840 free 113792 allocated 0 min_free 113792
15:12:50.967 ->     largest_free_block 113792 alloc_blocks 0 free_blocks 1 total_blocks 1
15:12:51.000 ->   Totals:
15:12:51.000 ->     free 282892 allocated 31588 min_free 276716 largest_free_block 124920
15:12:51.000 -> after close
```

lbernstone, 2021-01-17:
I don't think there is a leak.  There will be "delayed reclamation", but you get it back when the destructor runs.  Here's an example using the arduino functions.  Unremark the close, and you can see that you get the memory back right away.
```
#include <SPIFFS.h>
const char* fmtMemCk = "Free: %d\tMaxAlloc: %d\t PSFree: %d\n";
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())

void setup() {
  Serial.begin(115200);
  SPIFFS.begin(true);
}

void loop() {
  delay(1000);
  MEMCK;
  File root = SPIFFS.open("/","r");
  File file = root.openNextFile();
  //root.close();
  //file.close();
  MEMCK;
}
```

scubachristopher, 2021-01-18:
@lbernstone -- a few questions:

1).  How long does it take to reclaim?  See below...
2).  If you do close, it doesn't seem to free -- why?
3).  Original issue -- why does closing cause the issue with heap_caps_print_heap_info(MALLOC_CAP_DEFAULT)?

```
#include "FS.h"
#include "SPIFFS.h"

#define FORMAT_SPIFFS_IF_FAILED true
#define SLEEP_TIME              60

const char* fmtMemCk = "\nFree: %d\tMaxAlloc: %d\t PSFree: %d\n\n";
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())

int totalFree, maxAlloc;

void setup()
{
    
    Serial.begin(115200);
    if(!SPIFFS.begin(FORMAT_SPIFFS_IF_FAILED)){
        Serial.println("SPIFFS Mount Failed");
        return;
    }
    Serial.printf("\n\n");
}

void test(bool closeIt)
{
  
    File root = SPIFFS.open("/");
    File file = root.openNextFile();
  
    while (file) {
        Serial.printf("FILE: %18.18s, size: %6d bytes\n", file.name(), file.size());
        file = root.openNextFile();
    }
  
    if (closeIt) {
      close(root);
    }
}

void loop()
{
    int seconds = 0;

    heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
    MEMCK;
    totalFree = ESP.getFreeHeap();
    maxAlloc  = ESP.getMaxAllocHeap();
    test(true);

    while (true) {
      Serial.printf("\n%5d seconds in --> Total free delta is %d bytes less than at start.  Max allocation delta is %d bytes less than at start.\n", \
        seconds, totalFree - ESP.getFreeHeap(), maxAlloc - ESP.getMaxAllocHeap());
      heap_caps_print_heap_info(MALLOC_CAP_DEFAULT);
      delay(SLEEP_TIME * 1000);
      seconds += SLEEP_TIME;
    }
}
```

Output:

```
15:17:34.980 -> Heap summary for capabilities 0x00001000:
15:17:34.980 ->   At 0x3ffbdb28 len 52 free 4 allocated 0 min_free 4
[... 37 lines cut ...]
15:19:35.248 -> 
15:19:35.248 ->   120 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:20:35.253 -> 
15:20:35.253 ->   180 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:21:35.226 -> 
15:21:35.226 ->   240 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:22:35.239 -> 
15:22:35.239 ->   300 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:23:35.246 -> 
15:23:35.246 ->   360 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:24:35.258 -> 
15:24:35.258 ->   420 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:25:35.232 -> 
15:25:35.232 ->   480 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:26:35.237 -> 
15:26:35.237 ->   540 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:27:35.240 -> 
15:27:35.240 ->   600 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:28:35.246 -> 
15:28:35.246 ->   660 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:29:35.237 -> 
15:29:35.237 ->   720 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
15:30:35.249 -> 
15:30:35.249 ->   780 seconds in --> Total free delta is 648 bytes less than at start.  Max allocation delta is 0 bytes less than at start.
```

closed 2021-09-06 by stale[bot]

VojtechBartoska, 2022-04-06:
Hello, marking as expired. If you still face this issue on latest v2.0.3-RC1, please reopen it. Thanks for understanding and your contribution.

Lagunaxx, 2023-12-23:
file memleak still persists. latest downloaded version of arduino
```
#include <Arduino.h>
#include "FS.h"
#include "SPIFFS.h"

const char* fmtMemCk = "Free: %d\tMaxAlloc: %d\t PSFree: %d\n";
#define MEMCK Serial.printf(fmtMemCk,ESP.getFreeHeap(),ESP.getMaxAllocHeap(),ESP.getFreePsram())
#define SLEEP_TIME              60

#if CONFIG_AUTOSTART_ARDUINO

void setup() {

	Serial.begin(115200);
	SPIFFS.begin(true);
	delay(100);

	Serial.println("[APP] Free memory(1): " + String(esp_get_free_heap_size()) + " bytes");
	MEMCK;
	File *fp=new(File);
	Serial.println("[APP] Free memory(2): " + String(esp_get_free_heap_size()) + " bytes");
	MEMCK;
	*fp = SPIFFS.open("/wifioff.png", "r"); //(char *)icodata_WiFi.fname[0]
	Serial.println("[APP] Free memory(3): " + String(esp_get_free_heap_size()) + " bytes");
	MEMCK;
	fp->close();
	Serial.println("[APP] Free memory(4): " + String(esp_get_free_heap_size()) + " bytes");
	MEMCK;
	delete(fp);
	Serial.println("[APP] Free memory(5): " + String(esp_get_free_heap_size()) + " bytes");
	MEMCK;

}

void loop() {

    int seconds = 0;

    while (true) {
      MEMCK;
      delay(SLEEP_TIME * 1000);
      seconds += SLEEP_TIME;
    }

	delay(25);
}

```

and result is 
```
[APP] Free memory(1): 278652 bytes
Free: 353116    MaxAlloc: 147456         PSFree: 0
[APP] Free memory(2): 278624 bytes
Free: 353088    MaxAlloc: 147456         PSFree: 0
[APP] Free memory(3): 274188 bytes
Free: 348652    MaxAlloc: 139264         PSFree: 0
[APP] Free memory(4): 278536 bytes
Free: 353000    MaxAlloc: 147456         PSFree: 0
[APP] Free memory(5): 278564 bytes
Free: 353028    MaxAlloc: 147456         PSFree: 0
Free: 353028    MaxAlloc: 147456         PSFree: 0
Free: 353028    MaxAlloc: 147456         PSFree: 0
```

Lagunaxx, 2023-12-26:
for future askers: SPIFFS in first call allocate a lil bit ram, so on next open/close this first time allocated memory will not be freed, but other opened files on close will free ram normaly (except this first time allocated part)