espressif/arduino-esp32 #4728: No USB Serial port available on Mac OS

slaftos, 2021-01-17:
Make your question, not a Statement, inclusive. Include all pertinent information:

### Hardware:
Board:							MAGTAG29_ESP32S2
Core Installation version:			idf-release/v4.2
IDE name:						Arduino CLI or Arduino IDE 8.1.13
Flash Frequency:					build.flash_freq=80m
PSRAM enabled:                                    Same results enabled or disabled
Upload Speed:					921600
Computer OS:                                        Mac OS Mohave 10.14.6

### Description:
After upload and reset USB serial port is not identified/found by OS.
Happens on a simple blink sketch, Serial.begin is called first line in setup.
Cloned a fresh release today where this was happening.  Commit prior to below did not have this problem:
```
ESP-IDF update to 2bfdd036b and ESP-DSP

 idf-release/v4.2 (#4485) 
```

Code operates correctly (blinks)  but no USB Serial port available.
Booting up ROM bootloader serial port comes up just fine /dev/cu.usbmodem01

### Sketch: 
```cpp

// the setup function runs once when you press reset or power the board
void setup() {

  // initialize USB serial converter so we have a port created
  Serial.begin();

  // initialize built in LED pin as an output.
  pinMode(LED_BUILTIN, OUTPUT);}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);              // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);              // wait for a second
}
```

### Debug Messages:
```
USB serial port not available
```

SvenArke, 2021-01-18:
Syntax: Serial.begin(speed)

Serial.begin(9600);

Already tried ?

slaftos, 2021-01-18:
Yes even with a speed set it has the same result.  The USB device is just not recognized, not a communication issue.  It is more like a USB handshake issue, since neither windows nor Macos can recognize the device.  But switching back to the previous commit of idf-release/v4.2 and it works correctly.

rdmcmurray, 2021-01-29:
I can confirm the issue. My Metro ESP32-S2 that will only appear as an available port when reset in boot loader mode. 

-
Board: Adafruit Metro ESP32-S2
Core Installation version: idf-release/v4.2
IDE name: Arduino CLI or Arduino IDE 8.1.13
Flash Frequency: build.flash_freq=80m
PSRAM enabled: Same results enabled or disabled
Upload Speed: 921600
Computer OS: Mac OS Big Sur

dwriggles, 2021-02-25:
I am having this issue as well.

friggeri, 2021-03-07:
I am having the same issue as @rdmcmurray, same OS and configuration

StarWitch, 2021-03-09:
Having the same issue with the UnexpectedMaker Feather S2 (ESP32-S2). Switching back to the stale `esp32s2` branch fixes it.

shlomozippel, 2021-03-16:
Finally figured this out. The bug is in IDF 4.2, not the arduino core. This is just a bandaid until espressif fix it for real

https://github.com/shlomozippel/arduino-esp32/commit/593515c1662f433cbf707f23cf6b81ed679b3d88

ssilverman, 2021-04-17:
This fix looks like it's in the latest `master`, but I still don't see a serial port over USB. Is there a new fix?

ssilverman, 2021-06-18:
Is this fixed? Pinging to keep the issue alive.

ssilverman, 2021-08-21:
Pinging to keep the issue alive.

thedevleon, 2022-01-22:
Having the same issue. Did this ever get fixed or did you find a fix @ssilverman?

VojtechBartoska, 2022-04-07:
Hello folks, are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? 

Take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

slaftos, 2022-04-10:
This is resolved.  I tested with 2.0.2 and 2.0.3-RC1. both work correctly.

closed 2022-04-10 by slaftos