espressif/arduino-esp32 #4729: ESP32-S2 I2C does not honor repeated-start (no stop) in API?

ladyada, 2021-01-18:
hey @me-no-dev i verified someone trying an ESP32-S2 with an LC907 and i _am pretty sure_ that the issue is that when we ask for no-stop in `uint8_t TwoWire::endTransmission(bool sendStop)` its sending a stop anyways.

This is the correct I2C transaction to read the ID code (on avr chipset)
[image]
we write 0x11 register command to i2c device 0xB, don't stop, then read 3 bytes of data

if we run the same code on ESP32-S2 I get:
[image]
(there's a delay due to serial printf)
with wrong data 0xFF

if we go back to avr and turn ON stop (which is incorrect), we get the same resulting trace:
[image]
same wrong data

this could be the root cause for
https://github.com/espressif/arduino-esp32/issues/4589
https://github.com/espressif/arduino-esp32/issues/4561
https://github.com/espressif/arduino-esp32/issues/4375
https://github.com/espressif/arduino-esp32/issues/4403

those are also repeated-start devices. not all i2c devices require repeated start

me-no-dev, 2021-01-27:
@ladyada sorry for the delay. I2C for S2 uses ESP-IDF's I2C driver. I did not find a way to ensure no-stop there and remember that in order for that to work, write and read need to happen in one bus transaction (not really Arduino-friendly and with possible multitasking issues). Unfortunately we have not heard from @stickbreaker in a loong time (he was maintaining the driver) but we are looking into getting some help internally to be able to handle that in near future.

ladyada, 2021-01-27:
hihi thanks for replying - there's quite a few chips that require write-then-read in one no-stop transaction. we haven't heard of this problem before for ESP32 (just S2) so it could be a regression in v4.2 or for S2

either way its OK if there's a separate call just for write-then-read added, we have abstracted away the API for all our sensors so we could easily add that exception - just let me know :)

erichiller, 2021-06-09:
Is there a surround workaround for this? Perhaps @ladyada you know of one?
I'm encountering this with:
* Adafruit ESP32-S2 Metro
* Adafruit MLX90640 (55°)

ladyada, 2021-06-09:
there is no work-around, it must be fixed here/IDF :)

me-no-dev, 2021-06-10:
we are aware of the issue, it will be fixed before 2.0.0 is released :)

ladyada, 2021-06-10:
awesome thank you @me-no-dev  :)

VojtechBartoska, 2021-06-24:
This issue is solved internally in ESP-IDF ([IDF-3460](https://jira.espressif.com:8443/browse/IDF-3460)

hallard, 2021-07-05:
ahah Just came up here after scratching my head for 2 hours trying to have battery monitoring on Metro ESP32 S2. Glad to see it's not on our side. Just let us know if there is a fixed branch to test before official release of 2.0.0

SuGlider, 2021-09-14:
@ladyada , I just submitted a PR #5664 that shall fix the issue you reported.
Could you please apply it to your environment and verify if it solves this issue?

Thanks!

ladyada, 2021-09-14:
@SuGlider LC709 works right now, so it seems to have solved at least that issue!

me-no-dev, 2021-09-20:
Hi @ladyada could you please re-test with this new PR: https://github.com/espressif/arduino-esp32/pull/5683
I have only MLX90614 as sensor that requires repeated start and that worked without issues.

PR #5683 (2021-10-01): Implement Thread-Safe I2C based on ESP-IDF API
## Summary
This PR is a complete refactoring of  Wire Library and I2C HAL in order to use IDF instead of current Register manipulation approach.

It implements Thread-safe Arduino Wire.

Fix #5478 
Fix #5635 
Fix #5636 
Fix #4729 
Fix #5470 

## Impact
It solves many reported issues related to I2C.
closed 2021-10-01 by me-no-dev