espressif/arduino-esp32 #4730: WebServer.h POST receive example

redakker, 2021-01-18:
Hi,

Could you provide me a POST receiving example for ESP32?
Maybe it is my fault but I have not found it yet.

I would like to get the **body** of the POST request (**not** a file upload). How can I do that?

Thank you!

mehrdad987, 2021-01-19:
[https://github.com/G6EJD/ESP32-ESP8266-File-Download-Upload-Delete-Stream-and-Directory](url)

redakker, 2021-01-19:
> [https://github.com/G6EJD/ESP32-ESP8266-File-Download-Upload-Delete-Stream-and-Directory](url)
@mehrdad987 I have no result under that link

redakker, 2021-01-19:
@mehrdad987 Sorry, got it. I needed to copy-paste instead of clicking on it.
Thx

StevenDeRoover, 2021-02-07:
> @mehrdad987 Sorry, got it. I needed to copy-paste instead of clicking on it.
> Thx

Did you manage to get the values of a POST form?

I'm too struggling with this.

redakker, 2021-02-17:
> > @mehrdad987 Sorry, got it. I needed to copy-paste instead of clicking on it.
> > Thx
> 
> Did you manage to get the values of a POST form?
> 
> I'm too struggling with this.

I've solved with GET request for the time being. I have not tried it yet.

StevenDeRoover, 2021-02-17:
I did find a solution:
`server.on("<post uri>", HTTP_POST, [] {
    server.arg("<input from HTML form>").toCharArray(<variable>, <length of variable>);
    server.sendHeader("Connection", "close");
    server.send(200, "text/plain", "OK");
});`

closed 2021-07-08 by stale[bot]

rferrese, 2022-12-02:
An example of this would be really helpful.

lbernstone, 2022-12-02:
I don't know if the cert validation still works, but the POST args definitely do.
https://github.com/lbernstone/setCerts/blob/master/setCerts.ino

VoidVolker, 2026-10-13:
To get raw data using post method I found next solution:
```cpp
server.on("/post", HTTP_POST, [] { 
    String body = server.arg(0);
});
```
Looks like it also supports form-data and values can be obtained using it's names. Here the source code link: [`String arg(const String &name)`](https://github.com/espressif/arduino-esp32/blob/902745611a69350d7cb33d146a2d9d53f69ac21d/libraries/WebServer/src/WebServer.h#L244)