espressif/arduino-esp32 #4732: WiFi.localIP() returns 255.255.255.255 after connect

SvenArke, 2021-01-18:
### Hardware:
Board:							ESP32-WROOM-32
Core Installation version:			        1.0.4 & 1.0.5 rc6
IDE name:						Arduino 1.8.13

### Description:

With 1.0.6 - any version i get always 255.255.255.255 as IP address from WiFi.localIP()

Code:
      log_i("Connected to: ");
      log_i( "%s", WiFi.SSID() );
      log_i( "%s", WiFi.localIP().toString().c_str() );

Core 1.0.4 returns:
[I][Test-2020.ino:85] setup(): Connected to: 
[I][Test-2020.ino:86] setup(): NET1
[I][Test-2020.ino:87] setup(): 192.168.0.169

Core 1.0.5 rc1 to rc6 returns:
[I][Test-2020.ino:85] setup(): Connected to: 
[I][Test-2020.ino:86] setup(): NET1
[I][Test-2020.ino:87] setup(): 255.255.255.255

Tested all rc versions - no luck. Code runs only on 1.04 flawless. Any suggestions ?

Thank you !

SvenArke, 2021-01-20:
Found the problem !
Sketch contained a bugfix for "hostname not sent via dhcp" - >https://github.com/espressif/arduino-esp32/issues/2537

WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE); 

This kills WiFi.localIP() on Core 1.0.5 rc6 ...

me-no-dev, 2021-02-03:
done :) thanks for finding this

gerab, 2021-12-21:
wow, surprisingly it's still there!

thomastech, 2021-12-23:
Yes, I ran into this bug too. As mention before, no problem if I remove the bugfix for "hostname not sent via dhcp" - >#2537. Is that the final/only solution? Hoping for some words of wisdom from the devs.

VojtechBartoska, 2022-01-04:
@me-no-dev Can you please test this again?

YordanYanakiev, 2022-04-15:
Same issue here in all around the projects. :'(

VojtechBartoska, 2022-04-20:
@YordanYanakiev tested on v2.0.3-rc1?

YordanYanakiev, 2022-04-20:
yes, also the previous ones downto 2.0.0

SuGlider, 2022-04-21:
This issue is linked to #6610

SuGlider, 2022-04-21:
The main issue is that <lwip/inet.h> declares ```#define INADDR_NONE IPADDR_NONE``` and it includes <lwip/ip4_addr.h> that declares ```#define IPADDR_NONE ((u32_t)0xffffffffUL)``` (IP 255.255.255.255)

When <lwip/inet.h> is included, it makes the Arduino declaration of "IPAddress INADDR_NONE(0,0,0,0)" invalid. 

This process occurs in all the WiFi/ETH Arduino Libraries.

SuGlider, 2022-04-29:
This is the test case used to verify/test this issue and PR Fix:

``` cpp
#include <Arduino.h>
#include <WiFi.h>

void setup() {
  WiFi.begin("Your_SSID", "Your_wifi_password");
  String nodeName = "NODE-" + WiFi.macAddress();
  nodeName.replace(":", "");
  char _nodeName[20]; nodeName.toCharArray(_nodeName, 20);
  WiFi.setHostname(_nodeName);
  WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
  WiFi.mode(WIFI_STA);

  log_i("Connected to: ");
  log_i( "%s", WiFi.SSID() );
  log_i( "%s", WiFi.localIP().toString().c_str() );
  log_i( "%s", WiFi.getHostname());
}

void loop() {
}
```

closed 2022-04-29 by me-no-dev

rysiulg, 2022-05-04:
In This arduino libraries bugs and workaround for hostname on esp32 ...
any version i get always 255.255.255.255 as IP address from WiFi.localIP()

use:
 WiFi.config(((u32_t)0x0UL),((u32_t)0x0UL),((u32_t)0x0UL)); 
or 
#ifdef INADDR_NONE
#undef INADDR_NONE
#define INADDR_NONE ((u32_t)0x0UL)
#endif
WiFi.config(INADDR_NONE ,INADDR_NONE ,INADDR_NONE );

to get config definitions of ip 0.0.0.0 and be able to obtain dhcp values. 
By default arduino resolves definitions INADDR_NONE -> IPADDR_NONE -> ((u32_t)0xffffffff0UL) which gives 255.255.255.255 ip and dhcp resolution doesn't work.

best regards

> ### Hardware:
> Board: ESP32-WROOM-32 Core Installation version: 1.0.4 & 1.0.5 rc6 IDE name: Arduino 1.8.13
>
> ### Description:
> With 1.0.6 - any version i get always 255.255.255.255 as IP address from [...]

SuGlider, 2022-05-04:
> In This arduino libraries bugs and workaround for hostname on esp32 ... any version i get always 255.255.255.255 as IP address from WiFi.localIP()
>
> use: [...]

@rysiulg - The PR #6659 works for Arduino Core 2.0.3 onward.
In your post you say it doesn't work for Arduino 1.0.4 and 1.0.5, right?
In that case I suggest not using `INADDR_NONE` within these core versions, as you said.

`INADDR_NONE` can be replaced by `IPAddress(0,0,0,0)` or `(u32_t)0UL` as you wrote.

PR #6659 (2022-04-29): Fixes INADDR_NONE 
## Description of Change

Fixes ```IPAddress INADDR_NONE``` declaration when using Arduino WiFi or ETH.
This symbol was defined as ```0xffffffff``` by ```lwip /inet.h```, making it impossible to use INADDR_NONE correctly.

This PR only works when ```<wifi-provisioning/wifi_config.h>``` has a modification to include ```<lwip/ip4_addr.h>``` instead of ```<lwip/inet.h>```. This will be done directly to the sdk folder in the github structure and it has been fixed in IDF by a separated Merge Request. This will be reflected in the future, for good.

## Tests scenarios
This PR was  [... cut]
commit ce2cd111 (2022-04-29): Fixes INADDR_NONE  (#6659)

Description of Change

Fixes IPAddress INADDR_NONE declaration when using Arduino WiFi or ETH.
This symbol was defined as 0xffffffff by lwip /inet.h, making it impossible to use INADDR_NONE correctly.

This PR only works when <wifi-provisioning/wifi_config.h> has a modification to include <lwip/ip4_addr.h> instead of <lwip/inet.h>. This will be done directly to the sdk folder in the github structure and it has been fixed in IDF by a separated Merge Request. This will be reflected in the future, for good.

Tests scenarios

This PR was tested with all Arduino WiFi exa [... cut]