espressif/arduino-esp32 #4733: ESP-WROOM-32 bluetooth drops out after connecting

caspersauce, 2021-01-19:
I am having an issue with the esp-wroom-32. I am trying very basic examples to connect to the esp-wroom-32 over bluetooth. A basic example like this one:

[SerialToSerialBT.ino](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/SerialToSerialBT/SerialToSerialBT.ino)

This example works perfectly until a device actually connects. It connects and then within seconds it gets kicked off. This happens with any client: computer, phone or tablet. I found this issue from 6 months ago that describes the issue perfectly:

[github issue](https://github.com/espressif/arduino-esp32/issues/4092)

The only solution that it mentions is a better power supply. I have tried multiple power supplies and cables with no success. Has anyone had any problems getting a simple example with classic bluetooth working on a esp-wroom device?

caspersauce, 2021-01-20:
Just to add to this, I was able to get a windows pc to connect over bluetooth to the SerialToSerialBT.ino example. This is the only device that will successfully connect to to the bluetooth classic of a esp-wroom-32. Is there some kind of configuration that is making the wroom kick off certain clients like a mac or android device?

Has anyone ever connected to the esp-wroom classic bluetooth using an android device? or is this something that it is unable to do?

lbernstone, 2021-01-20:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BluetoothSerial

caspersauce, 2021-01-20:
The windows connection working led me down an hours long path to finally solving this problem. The problem lies with the clients. Some devices will instantly disconnect and bluetooth classic connection if data is not be transmitted. Some clients must have some kind of idle dropout setting.

If people in the future have this issue, ensure that in your example you are transmitting data, just connecting to a device may not be enough to hold a connection

closed 2021-01-20 by caspersauce

Bhaktixcode, 2025-07-04:
If you got the solution caspersauce, can you help out please, this is like really urgent!

Bhaktixcode, 2025-07-04:
Can you tell me what kind of data you are talking about exactly and how to solve this? Elaborate please!!

[image]