espressif/arduino-esp32 #4736: WiFiClient::flush() can get stuck indefinitely

tko, 2021-01-20:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
Core Installation version:			1.0.4
IDE name:						Platform.io
Flash Frequency:					Can't tell (Crystal is 40MHz?)
PSRAM enabled:                                    Can't tell
Upload Speed:					921600
Computer OS:                                        Mac OSX

### Description:
After reading the http response line with `WiFiClient::readBytesUntil()` calling `WiFiClient::flush()` hangs indefinitely. Doesn't always happen, best guess is it depends on response length. First noticed with 204 No Content response from an influxdb server.

### Sketch:
```cpp
#include <WiFi.h>
#include <WiFiClient.h>

const char ssid[] = "FIXME";
const char pass[] = "FIXME";

void setup()
{
  Serial.begin(115200);

  WiFiClient client;
  WiFi.begin(ssid, pass);
  WiFi.waitForConnectResult();
  client.connect("httpbin.org", 80);
  client.printf("GET /status/200 HTTP/1.0\r\nHost: httpbin.org\r\n\r\n");

  auto res = client.readStringUntil('\n');
  Serial.println(res);

  client.flush();
  Serial.println("Done.");
}

void loop()
{
}
```

### Debug Messages:
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.1.2, MASK: 255.255.255.0, GW: 192.168.1.1
HTTP/1.1 204 NO CONTENT
```

enwi, 2021-01-23:
Having the same issue. The while loop inside `WiFiClient::flush()` hangs indefinitely due to `recv` returning 0, but the previous call to `available()` returning a value larger 0. Adding some sort of timeout would definitely help.

https://github.com/espressif/arduino-esp32/blob/3236358dedd91c44c7ff32f7c23f384c59c8368b/libraries/WiFi/src/WiFiClient.cpp#L483

enwi, 2021-01-23:
Ahh I think I found the culprit. The call to `WiFiClient::available()` inside `WiFiClient::flush()` refers to the available data inside `WiFiClientRxBuffer` which was already received from the socket. In that case the socket can't receive anymore data, since it was already received from the RXBuffer thus looping forever waiting for data.

I fixed the code by changing the implementation of `WiFiclient::flush()` to 
```c++
void WiFiClient::flush() {
    if(available()){
       while(read() >= 0) {};
    }
}
```

This way the `WiFiClientRxBuffer` is in charge of handling the data and receiving from  the socket.

enwi, 2021-06-20:
I dare you to close the issue 🤣

Darktemp, 2021-08-15:
I had the same issue, that would hand PubSubClient::connected() if the connection was lost and it lead to the same spot in WiFiClient::flush(), where available() reported non-zero byte but recv always returned 0. Checking errno anyways shows this:
`[E][WiFiClient.cpp:491] flush(): fail on fd 58, errno: 128, "Socket is not connected"`
For now, I just changed `if(res < 0) {` to `if(res <= 0) {` but I'm not sure if there are valid scenarios, where res==0 is fine.

VojtechBartoska, 2022-03-30:
Hello @tko, sorry for late reply. Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

tko, 2022-04-02:
Now I get
```
...
[  2617][D][WiFiGeneric.cpp:852] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  2620][D][WiFiGeneric.cpp:914] _eventCallback(): STA IP: 192.168.1.2, MASK: 255.255.255.0, GW: 192.168.1.1
HTTP/1.1 200 OK
[  3127][E][WiFiClient.cpp:499] flush(): fail on fd 48, errno: 128, "Socket is not connected"
Done.
```
I suppose it's better for not hanging anymore though the error seems little out of place.

Adding bit more logging I see before `flush()` calling `available()` returns 214, and checking `connected()` returns 0 and logs additionally
```
[  3164][D][WiFiClient.cpp:528] connected(): Disconnected: RES: 0, ERR: 128
```

VojtechBartoska, 2022-04-04:
Thanks for answer, we will test this.

lemonkey, 2022-08-27:
I also see this when there are remaining bytes to be read for an HTTPClient request but the client has since disconnected "gracefully". recv returns 0 and flush() gets stuck in a loop since the original value returned by available() (used to set the local variable 'a') is non-zero.

lemonkey, 2022-08-27:
@Darktemp 

> For now, I just changed if(res < 0) { to if(res <= 0) { but I'm not sure if there are valid scenarios, where res==0 is fine.

When res == 0, you can check to see if errno is set. In my testing errno is always 128 when this happens ("Socket is not connected".

michael-land, 2023-08-05:
The problem persists in version 2.0.11

Shaquum, 2023-10-02:
please look morę for the issues

xreef, 2023-10-17:
esp32 core 2.0.14 fixes the problem.
Thanks to all

Parsaabasi, 2025-01-09:
Hello, 

I close this because the issue seems fixed. Also, this report contains the release we no longer support. Please try the new versions and in case the issue persists, feel free to reopen it. 

Thanks

closed 2025-01-09 by Parsaabasi