espressif/arduino-esp32 #4740: ESP32-S2 Software Serial compile error

kevin192291, 2021-01-23:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do? Use Software Serial with the ESP32-S2
Describe your system( Hardware, computer, O/S, core version, environment). Linux Mint (latest) Arduino IDE v 1.8.13
Describe what is failing. I am getting the error:
```
/home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/esp32s2/lib/librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0xa9): could not decode instruction; possible configuration mismatch
/home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: libraries/EspSoftwareSerial/SoftwareSerial.cpp.o:(.literal._ZN14SoftwareSerial6rxBitsEv+0x8): undefined reference to `__atomic_exchange_1'
/home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: libraries/EspSoftwareSerial/SoftwareSerial.cpp.o: in function `SoftwareSerial::rxBits()':
/home/kevin/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/c++/8.4.0/bits/atomic_base.h:413: undefined reference to `__atomic_exchange_1'
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board ESP32S2 Dev Module.
```
Show the shortest possible code that will duplicate the error.
```
#include <TinyGPS.h>
#include <SoftwareSerial.h>

TinyGPS gps;
SoftwareSerial ss(32, 33);
int gpsLockPin = 25;
int gpsEnPin = 15;
int gpsLock = 0;

void GpsSetup()
{
  ss.begin(9600);
}
```
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

I have attempted to install the https://github.com/plerup/espsoftwareserial library with no luck.
I suspected that it woudn't work with the 32-S2, but due to the lack of an extra UART port, I thought This problem may have been fixed by now.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32-s2 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Ubuntu (linux mint)

```

ByronAP, 2021-01-24:
same issue here... librtc.a exists in the location

> ld.exe: C:\Users\bapen\AppData\Local\arduino15\packages\esp32\hardware\esp32\2.0.0\tools\sdk\esp32s2\lib\librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0xa9): could not decode instruction; possible configuration mismatch
The system cannot find the file specified

seackone, 2021-01-24:
Hi,
I have exactly the same issue as kevin:

```/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: /Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/sdk/esp32s2/lib/librtc.a(rtc.o)(.text.rtc_pad_gpio_wakeup+0xa9): could not decode instruction; possible configuration mismatch
/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: libraries/EspSoftwareSerial/SoftwareSerial.cpp.o:(.literal._ZN14SoftwareSerial6rxBitsEv+0xc): undefined reference to `__atomic_exchange_1'
/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld: libraries/EspSoftwareSerial/SoftwareSerial.cpp.o: in function `SoftwareSerial::rxBits()':
/Users/Name/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/xtensa-esp32s2-elf/xtensa-esp32s2-elf/include/c++/8.4.0/bits/atomic_base.h:413: undefined reference to `__atomic_exchange_1'
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board ESP32S2 Dev Module.
```

And here is the code:

```
#include <SoftwareSerial.h>

SoftwareSerial SoftSerial(32, 33);

void setup() {
}
void loop() {
}
```

Im using the Arduino IDE 1.8.13 as well on a Macbook Pro.

kevin192291, 2021-01-24:
Is there any chance that it is a conflict between SoftwareSerial libraries?

me-no-dev, 2021-01-25:
S2's toolchain and maybe newer ones for ESP32 do not have `__atomic_exchange_1` implemented. I have no idea if that will be done or if the SoftSerial lib will need adjustment. Submit an issue here for more info: https://github.com/espressif/crosstool-NG

kevin192291, 2021-01-25:
Posted it now, Link is [Here](https://github.com/espressif/crosstool-NG/issues/10)

vautieri, 2021-03-07:
Any progress on getting a new build with this fixed?

kevin192291, 2021-03-08:
I am sorry, that I haven't updated this ticket, It was moved and the resolution is posted here:
https://github.com/espressif/esp-idf/issues/6463#issuecomment-778392769

closed 2021-03-08 by kevin192291