espressif/arduino-esp32 #4743: [v4.2] Hardware timer functions not working, interrupts fail - known issue?

taligentx, 2021-01-25:
Hi folks,

The hardware timer functions seem to have an issue in the IDF 4.2 branch.  The following sketch correctly blinks the onboard LED in 1.0.5-rc6 - under 4.2, this sketch compiles but fails to fire the interrupt:
```cpp
hw_timer_t * timer = NULL;
#define ledPin LED_BUILTIN

void onTimer() {
  if (digitalRead(ledPin)) digitalWrite(ledPin, 0);
  else digitalWrite(ledPin, 1);
}

void setup() {
  pinMode(ledPin, OUTPUT);
  timer = timerBegin(0, 80, true);
  timerAttachInterrupt(timer, &onTimer, true);
  timerAlarmWrite(timer, 1000000, true);
  timerAlarmEnable(timer);
}

void loop() {}
```

The native IDF functions do work in the 4.2 branch, as tested in this example:
```cpp
#include "esp_timer.h"
#define ledPin LED_BUILTIN

void onTimer() {
  if (digitalRead(ledPin)) digitalWrite(ledPin, 0);
  else digitalWrite(ledPin, 1);
}

esp_timer_handle_t timer;
const esp_timer_create_args_t timerParameters = { .callback = reinterpret_cast<esp_timer_cb_t>(&onTimer) };

void setup() {
  pinMode(ledPin, OUTPUT);
  esp_timer_create(&timerParameters, &timer);
  esp_timer_start_periodic(timer, 1000000);
}

void loop() {}
```    

    
Is this currently the expected behavior?  Per `esp32-hal-timer.c`:
```cpp
void timerAttachInterrupt(hw_timer_t *timer, void (*fn)(void), bool edge){
	// EDGE DOES NOT WORK CURRENTLY
	edge = false;
```

Thanks!

Environment: NodeMCU ESP-32S generic dev module, ESP32-S2 official dev board, macOS 11.1, Arduino IDE 1.8.13, core 1.0.5-rc6 and idf-release/v4.2.

ClemensAtElektor, 2021-03-16:
To me, the second parameter in the call to timerAttachInterrupt should be onTimer without the '&', like so:

`timerAttachInterrupt(timer, onTimer, true);`

I think this is a bug in the timer examples RepeatTime and WatchdogTimer, but I may be mistaken.

RoscoeTheDog, 2021-03-19:
I am having this issue as well on my ESP32-S2-Saola-1 board. I have taken the [direct example](https://github.com/espressif/arduino-esp32/blob/master/libraries/ESP32/examples/Timer/RepeatTimer/RepeatTimer.ino), and even tried taking out the pass by reference (&) from the code. When running the code, the serial monitor prints nothing. I have tried printing the incrimenter from the ISR and it stays 0, indicating the ISR is not running.

Any updates or solutions on this? Many libraries seem to rely on the Arduino framework instead of the ESP-IDF framework. However, when there is timers involved then I need to adjust some code for the libraries to compensate for this. As an example, I am trying to use a stepper driver library which relies on the hardware timers so I need this resolved in order for it to work.

lbernstone, 2021-03-19:
I am fairly certain the Ticker library still works for a simple timer.  If you need more control, the code in the Ticker library can be used as a template.

taligentx, 2021-03-19:
> I am fairly certain the Ticker library still works for a simple timer. If you need more control, the code in the Ticker library can be used as a template.

[Ticker](https://github.com/espressif/arduino-esp32/blob/idf-release/v4.2/libraries/Ticker/src/Ticker.cpp) uses the ESP-IDF functions as in my second example, and should be fine for milliseconds and seconds-level timers.  Unfortunately it currently doesn't handle microsecond timers and wouldn't work for my library ([dscKeybusInterface](https://github.com/taligentx/dscKeybusInterface)), which requires a 250 microsecond timer.  IR libraries also use microsecond level timers.

me-no-dev, 2021-03-19:
I've tested and I am using the timer driver for Marlin. Please provide a full minimum sketch to reproduce :)

RoscoeTheDog, 2021-03-23:
Here is full example sketch, taken from the OP. Neither attaching the interrupt pass-by-reference or pass-by-copy does anything.

I also tried using this board with the [ESP32TimerInterrupt library](https://github.com/khoih-prog/ESP32TimerInterrupt). The author confirmed this library does not work with the S2 variant of the ESP32. It would not be a big deal as long as I had _some other_ way of utilizing Timer/ISR functionality. As it stands currently, this board which has been out for almost 2 years now still does not seem to have timer functionality/support under the Arduino framework whatsoever, and there is no way to combine the frameworks together in a "compatibility" mode, as the compiler complains that the board is unsupported for the `esp-idf+` framework. I have tried almost everything now. It seems as though you have to choose one framework or the other at the moment which sucks because most of the available libraries out there right now are mostly supported under the Arduino framework.

```
#include <Arduino.h>
#include "esp_timer.h"
hw_timer_t *timer = NULL;
#define ledPin 17

void onTimer()
{
	Serial.println("ISR Running");
}

void setup()
{
	pinMode(ledPin, OUTPUT);
	timer = timerBegin(0, 80, true);
	timerAttachInterrupt(timer, onTimer, true);
	timerAlarmWrite(timer, 1000000, true);
	timerAlarmEnable(timer);
}

void loop() {}
```

PlatformIO config:
```
[env:esp32-s2-saola-1]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-idf-v4.2
// you can try adding platform_packages="idf-framework git URL here"
board = esp32-s2-saola-1
framework = arduino // if you add "esp-idf+" after arduino with the included platform packages, compiler will complain "this board is unsupported" Note that it does work if you only choose one or the other regardless depending on which one you choose.
lib_deps = 
	https://github.com/khoih-prog/ESP32TimerInterrupt.git
monitor_speed = 115200
upload_port = COM7
board_build.f_cpu = 240000000L
```

taligentx, 2021-03-24:
Hi @me-no-dev,

> I've tested and I am using the timer driver for Marlin. Please provide a full minimum sketch to reproduce :)

I've re-tested the first example on a freshly installed macOS 10.14 system, new Arduino install, new checkout of the 'idf-releases/v4.2' branch with the following steps - only modified to remove `--depth 1` from `git clone` to get all branches:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/mac.md
```
mkdir -p ~/Documents/Arduino/hardware/espressif && \
cd ~/Documents/Arduino/hardware/espressif && \
git clone https://github.com/espressif/arduino-esp32.git esp32 && \
cd esp32 && \
git submodule update --init --recursive --depth 1 && \
cd tools && \
python get.py 
```
Compiled and uploaded to a NodeMCU esp32 board and again confirmed that the timer interrupt is not firing.  Checked out the master branch, re-launched Arduino, re-compiled and uploaded to the board and the timer interrupt is working correctly.

Let me know if there's any additional info I can provide to help replicate this issue - thanks!

-Nikhil

me-no-dev, 2021-03-25:
@taligentx I'll have a look in a few days, but in general the timer is hardware and Arduino has it's own low-level driver which have not changed, so it is really strange how the hardware would not work between versions :)

blackhack, 2021-06-16:
Bumping, right now on master branch (https://github.com/espressif/arduino-esp32/commit/cf43d174b7f64c20bb3499c82739b7240742df8b) the problem still persist, I even tried with https://github.com/khoih-prog/ESP32TimerInterrupt/commits/master but to no avail. I check-out to idf-release/v3.3 and the same code works without problem.

I tried both examples RepeatTimer.ino from ESP32/Timer and TimerInterruptTest.ino from the library.

breedx2, 2021-06-20:
Same thing here on a recent install. Timer code from the examples does nothing, cannot get it to fire. Tried:
* different divider frequencies (80/240)
* println in the ISR directly vs. just incrementing a value and printing that value from `loop()`
* with/without a mutex
* ARDUINO_ISR_ATTR vs IRAM_ATTR on the isr callback

What a drag/showstopper. Happy to try other things and to troubleshoot/experiment, because this project is dead without a working hardware timer.

breedx2, 2021-06-26:
> To me, in the function setup() the second parameter in the call to timerAttachInterrupt should be onTimer without the '&', like so:
> 
> `timerAttachInterrupt(timer, onTimer, true);`
> 
> I think this is a bug in the timer examples RepeatTime and WatchdogTimer, but I may be mistaken.

I don't think this is true. It did not help in my case anyway. I tried it. Does it work for you @ClemensAtElektor or were you just speculating?

blackhack, 2021-06-26:
> I don't think this is true. It did not help in my case anyway. I tried it. Does it work for you @ClemensAtElektor or were you just speculating?

`&` operator is optional when addressing a function pointer, so for the compiler is irrelevant which format is used, the bug is in the esp32-arduino core somewhere.

JohnMacrae, 2021-06-27:
I took the Timer example code on a know good board and compiled and uploaded it using the Boards manager release version (not Alpha).  It runs fine.  I then took the same code example and compiled it using the latest git Alpha (on both Win 10 and Ubuntu) and uploaded it. The timer does not work.

closed 2021-06-28 by me-no-dev

RoscoeTheDog, 2021-06-28:
@me-no-dev 

Why was this closed? It is clear that people are still encountering this bug/issue and there is no working solution.

me-no-dev, 2021-06-29:
it was closed automatically by the commit linked above, because it contains the fix

RoscoeTheDog, 2021-06-29:
Apologies.. I did not see the hyperlink to the solution. Great news. Many thanks

breedx2, 2021-07-01:
I confirmed tonight that this fix also works for me. Thanks for the work! Very appreciated. :pray:

spotrevoc, 2021-07-06:
I have updated to the latest versions of Arduino (1.8.15) and arduino-esp32 (2.0.0-alpha1), verified that esp32-hal-timer.c includes the fixes in the commit above, recompiled, and I still cannot get the timers to work on the esp32-s2.

blackhack, 2021-07-06:
> 
> 
> I have updated to the latest versions of Arduino (1.8.15) and arduino-esp32 (2.0.0-alpha1), verified that esp32-hal-timer.c includes the fixes in the commit above, recompiled, and I still cannot get the timers to work on the esp32-s2.

That release is from April 17, for this fix to be merge you need to use directly git's latest version.
https://github.com/espressif/arduino-esp32/archive/refs/heads/master.zip

spotrevoc, 2021-07-06:
I have the most recent version. I nuked my entire Arduino install and got fresh hardware cores and libs today. Installed straight from the .git and went into esp32-hal-timer.c and physically verified that it contains the fix in the commit above (it does). See the date modified on the file in the screen snip below, it's literally from an hour ago.
[image]

blackhack, 2021-07-06:
> 
> 
> I have the most recent version. I nuked my entire Arduino install and got fresh hardware cores and libs today. Installed straight from the .git and went into esp32-hal-timer.c and physically verified that it contains the fix in the commit above (it does). See the date modified on the file in the screen snip below, it's literally from an hour ago.
> [image]
> 
> ESP_IDF native code works just fine.

Sadly I don't have an esp32-s2 available to do some testing, the fix works for sure on esp32. Try looking for any residual Arduino folder on %userprofile%\AppData\Local\Temp and delete it (They shouldn't be there anyway after closing the IDE).

Or %userprofile%\AppData\Local\Temp\VMBuilds if using VisualMicro

alireza369, 2021-10-19:
> I confirmed tonight that this fix also works for me. Thanks for the work! Very appreciated. 🙏

hi how did you solved this problem ? 
plz share whit us 😘