espressif/arduino-esp32 #4745: Connection error with HTTPS OTA Firmware Upgrade for Arduino

v11, 2021-01-26:
### Hardware:
Board:							ESP32 Lolin D32
Core Installation version:			arduino-esp32 1.0.5-rc6
IDE name:						vscode with arduino-plugin
Flash Frequency:					80Mhz
PSRAM enabled:                                    Not sure
Upload Speed:					115200
Computer OS:                                        Mac OSX

### Description:
I am trying to run the HTTPS_OTA_Update.ino demo. Unfortunately without success. 

### Sketch: 
I am using the example provided in arduino-esp32/libraries/Update/examples/HTTPS_OTA_Update/
I just added a simple function to test HTTPS connection to my server.

```
// This sketch provide the functionality of OTA Firmware Upgrade
#include "Arduino.h"
#include "WiFi.h"
#include <WiFiClientSecure.h>
#include "HttpsOTAUpdate.h"
// This sketch shows how to implement HTTPS firmware update Over The Air.
// Please provide your WiFi credentials, https URL to the firmware image and the server certificate.

static const char *ssid = "*********";       // your network SSID (name of wifi network)
static const char *password = "*********"; // your network password

static const char *url = "https://vision11.ch/inklay/bin/inklay.bin"; //state url of your firmware image

// Certificate Vision11
const char *cert_vision11 =
    "-----BEGIN CERTIFICATE-----\n"
    "MIIEZTCCA02gAwIBAgIQQAF1BIMUpMghjISpDBbN3zANBgkqhkiG9w0BAQsFADA/\n"
    "MSQwIgYDVQQKExtEaWdpdGFsIFNpZ25hdHVyZSBUcnVzdCBDby4xFzAVBgNVBAMT\n"
    "DkRTVCBSb290IENBIFgzMB4XDTIwMTAwNzE5MjE0MFoXDTIxMDkyOTE5MjE0MFow\n"
    "MjELMAkGA1UEBhMCVVMxFjAUBgNVBAoTDUxldCdzIEVuY3J5cHQxCzAJBgNVBAMT\n"
    "AlIzMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAuwIVKMz2oJTTDxLs\n"
    "jVWSw/iC8ZmmekKIp10mqrUrucVMsa+Oa/l1yKPXD0eUFFU1V4yeqKI5GfWCPEKp\n"
    "Tm71O8Mu243AsFzzWTjn7c9p8FoLG77AlCQlh/o3cbMT5xys4Zvv2+Q7RVJFlqnB\n"
    "U840yFLuta7tj95gcOKlVKu2bQ6XpUA0ayvTvGbrZjR8+muLj1cpmfgwF126cm/7\n"
    "gcWt0oZYPRfH5wm78Sv3htzB2nFd1EbjzK0lwYi8YGd1ZrPxGPeiXOZT/zqItkel\n"
    "/xMY6pgJdz+dU/nPAeX1pnAXFK9jpP+Zs5Od3FOnBv5IhR2haa4ldbsTzFID9e1R\n"
    "oYvbFQIDAQABo4IBaDCCAWQwEgYDVR0TAQH/BAgwBgEB/wIBADAOBgNVHQ8BAf8E\n"
    "BAMCAYYwSwYIKwYBBQUHAQEEPzA9MDsGCCsGAQUFBzAChi9odHRwOi8vYXBwcy5p\n"
    "ZGVudHJ1c3QuY29tL3Jvb3RzL2RzdHJvb3RjYXgzLnA3YzAfBgNVHSMEGDAWgBTE\n"
    "p7Gkeyxx+tvhS5B1/8QVYIWJEDBUBgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEE\n"
    "AYLfEwEBATAwMC4GCCsGAQUFBwIBFiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2Vu\n"
    "Y3J5cHQub3JnMDwGA1UdHwQ1MDMwMaAvoC2GK2h0dHA6Ly9jcmwuaWRlbnRydXN0\n"
    "LmNvbS9EU1RST09UQ0FYM0NSTC5jcmwwHQYDVR0OBBYEFBQusxe3WFbLrlAJQOYf\n"
    "r52LFMLGMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDAjANBgkqhkiG9w0B\n"
    "AQsFAAOCAQEA2UzgyfWEiDcx27sT4rP8i2tiEmxYt0l+PAK3qB8oYevO4C5z70kH\n"
    "ejWEHx2taPDY/laBL21/WKZuNTYQHHPD5b1tXgHXbnL7KqC401dk5VvCadTQsvd8\n"
    "S8MXjohyc9z9/G2948kLjmE6Flh9dDYrVYA9x2O+hEPGOaEOa1eePynBgPayvUfL\n"
    "qjBstzLhWVQLGAkXXmNs+5ZnPBxzDJOLxhF2JIbeQAcH5H0tZrUlo5ZYyOqA7s9p\n"
    "O5b85o3AM/OJ+CktFBQtfvBhcJVd9wvlwPsk+uyOy2HI7mNxKKgsBTt375teA2Tw\n"
    "UdHkhVNcsAKX1H7GNNLOEADksd86wuoXvg==\n"
    "-----END CERTIFICATE-----\n"
    "";

static HttpsOTAStatus_t otastatus;

void setup()
{

  Serial.begin(115200);
  Serial.print("Attempting to connect to SSID: ");
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  testConnection("www.vision11.ch");

  HttpsOTA.onHttpEvent(HttpEvent);
  Serial.println("Starting OTA");
  HttpsOTA.begin(url, cert_vision11);

  Serial.println("Please Wait it takes some time ...");
}

void testConnection(char *host)
{

  // This function loads device config data from Firebase (Deep Sleep in Seconds, GMT Time Zone Offset in Seconds)

  Serial.println("");
  Serial.println("connecting to vision11");

  // Wifi Client for secure connection (HTTPS)
  WiFiClientSecure client;

  // Add Certificate
  client.setCACert(cert_vision11);

  // Connect to Host
  Serial.println("connect to server via port 443");

  // HTTPS Connection: Check if Secure Connection is successfull
  if (!client.connect(host, 443))
  {
    // Connection Failed
    Serial.println("connection failed!");

    char err_buf[100];
    if (client.lastError(err_buf, 100) < 0)
    {
      // Show the last Error-Message when something goes wrong
      Serial.println(err_buf);
    }
    else
    {
      Serial.println("connection error");
    }

    return;
  }
  else
  {
  }

  Serial.println("connection success");

  // close
  client.stop();
}

void HttpEvent(HttpEvent_t *event)
{
  switch (event->event_id)
  {
  case HTTP_EVENT_ERROR:
    Serial.println("Http Event Error");
    break;
  case HTTP_EVENT_ON_CONNECTED:
    Serial.println("Http Event On Connected");
    break;
  case HTTP_EVENT_HEADER_SENT:
    Serial.println("Http Event Header Sent");
    break;
  case HTTP_EVENT_ON_HEADER:
    Serial.printf("Http Event On Header, key=%s, value=%s\n", event->header_key, event->header_value);
    break;
  case HTTP_EVENT_ON_DATA:
    break;
  case HTTP_EVENT_ON_FINISH:
    Serial.println("Http Event On Finish");
    break;
  case HTTP_EVENT_DISCONNECTED:
    Serial.println("Http Event Disconnected");
    break;
  }
}

void loop()
{
  otastatus = HttpsOTA.status();
  if (otastatus == HTTPS_OTA_SUCCESS)
  {
    Serial.println("Firmware written successfully. To reboot device, call API ESP.restart() or PUSH restart button on device");
  }
  else if (otastatus == HTTPS_OTA_FAIL)
  {
    Serial.println("Firmware Upgrade Fail");
  }
  delay(1000);
}
```

### Problem:
HTTPS Connection to my server is working but the OTA is not working? Any help is welcome. 

### Debug Messages:
```
Connected to Lukas WiFi
connecting to vision11
connect to server via port 443
connection success

Starting OTA
Please Wait it takes some time ...
E (2365) esp-tls: mbedtls_ssl_handshake returned -0x2700
E (2368) esp-tls: Failed to open new connection
E (2368) TRANS_SSL: Failed to open a new connection
E (2368) HTTP_CLIENT: Connection failed, sock < 0
E (2373) esp_https_ota: Failed to open HTTP connection: ESP_ERR_HTTP_CONNECT
E (2379) esp_https_ota: Failed to establish HTTP connection
Http Event Disconnected
Http Event Disconnected
Firmware Upgrade Fail
```

v11, 2021-01-26:
with
`HttpsOTA.begin(url, cert_vision11, false);`

i get
```
Attempting to connect to SSID: .Connected to Lukas WiFi
Starting OTA
Please Wait it takes some time ...
Http Event On Connected
Http Event Header Sent
Http Event On Header, key=Connection, value=Keep-Alive
Http Event On Header, key=Content-Type, value=application/octet-stream
Http Event On Header, key=Last-Modified, value=Mon, 25 Jan 2021 20:52:29 GMT
Http Event On Header, key=Etag, value="1812b0-600f2f8d-cb4e1db8c82e7411;;;"
Http Event On Header, key=Accept-Ranges, value=bytes
Http Event On Header, key=Content-Length, value=1577648
Http Event On Header, key=Date, value=Tue, 26 Jan 2021 20:48:35 GMT
Http Event On Header, key=Alt-Svc, value=quic=":443"; ma=2592000; v="43,46", h3-Q043=":443"; ma=2592000, h3-Q046=":443"; ma=2592000, h3-Q050=":443"; ma=2592000, h3-25=":443"; ma=2592000, h3-27=":443"; ma=2592000
E (7413) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (7450) task_wdt:  - IDLE0 (CPU 0)
E (7582) task_wdt: Tasks currently running:
E (7582) task_wdt: CPU 0: ipc0
E (7582) task_wdt: CPU 1: IDLE1
E (7582) task_wdt: Aborting.
abort() was called at PC 0x401085f0 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088214:0x3ffbe2c0 0x40088491:0x3ffbe2e0 0x401085f0:0x3ffbe300 0x40082e65:0x3ffbe320 0x40085485:0x3ffb9d60 0x40082541:0x3ffb9d80 0x40085203:0x3ffb9da0 0x4008942a:0x3ffb9dc0

Rebooting...
```

Can somebody at least confirm that this code was tested successfully?

arkhipenko, 2021-03-03:
I am having the same issue with core 1.0.5
The OTA code that was successfully updating both SPIFFS and BINARY from the HTTPS location when compiled with core 1.0.4 stopped working with 1.0.5. Not a single line of code changed. It's a plain-vanilla update. 

```
bool checkOTA() {
  Log.notice("checkOTA()" CR);
  WiFiClientSecure espClient;  

  bool        updateHappened = false;
  int         rc;

  //  do not reboot automatically when update is over (since we need to display message)
  httpUpdate.rebootOnUpdate(false);

  //  this callback is used to draw a horizontal progress line on the OTA screen
  Update.onProgress(handleOTAProgress);

  t_httpUpdate_return ret = httpUpdate.updateSpiffs( espClient, d["ota url base"] + OTA_URI, appVersion );
  switch (ret) {
    case HTTP_UPDATE_FAILED:
      Log.warning("HTTP_UPDATE_FAILED: Error code=%d (%S)" CR, httpUpdate.getLastError(), httpUpdate.getLastErrorString());
      break;

    case HTTP_UPDATE_NO_UPDATES:
      Log.notice("HTTP_UPDATE_NO_UPDATES" CR);
      break;

    case HTTP_UPDATE_OK:
      Log.notice ("HTTP_UPDATE_OK");
      updateHappened = true;
      break;
  }
```

failed with "Connection refused" message.

arkhipenko, 2021-03-03:
Hmm, interesting - it did work OK with the explicit root CA certificate, but not without it as it used to be. Is that a new feature?

me-no-dev, 2021-03-04:
yes, you need to now specify that you will not verify the certificate in order to connect without it :)

https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFiClientSecure/src/WiFiClientSecure.h#L60

arkhipenko, 2021-03-04:
Thank you for letting me know!

closed 2021-07-09 by stale[bot]

aminbaig, 2021-08-11:
@v11 were you able to make it work. I am having the same problem and would really appreciate if you can share the modification you might have made to make it work.

v11, 2021-08-11:
@aminbaig I was using another (http only) OTA example and made it compatible with Certs. But in the end i have my binaries on a HTTP server, so i don't need  https at all.

aminbaig, 2021-08-12:
@v11 Thanks for your response. Would it be possible for you to share that example which you are running? Would really appreciate it.

v11, 2021-08-12:
There is some other stuff you might want to get rid of first (led-controls, error-messages etc). But in general, this works flawless for me.

```
void execOTA(String serverPath, String filename, String serverVersion)
{

  // This function performs an over the air update (HTTP).
  // The binary on the server will be downloaded and installed

  // The url structure to the binary: [host]/[path]/[channel]/[displayType]/[filename]
  // Example XXX

  Serial.println("");
  Serial.println("exec ota");

  long contentLength = 0;
  bool isValidContentType = false;

  // Host
  const char *host = updateServerHost;

  // URL to the binary folder
  String path = updateServerBinFolder;

  // Activate the LED
  ledcWrite(0, 255);
  ledcWrite(1, 255);
  ledcWrite(2, 255);

  // Render Debug Page
  renderDisplayOTA(serverVersion);

  // Wifi Client (HTTP)
  WiFiClient client;

  Serial.println("connect to server via port 80");

  // Connect the update
  if (client.connect(host, 80))
  {

    Serial.print("requesting url: ");
    Serial.println(host + path + clientChannel + "/" + clientDisplayType + "/" + filename);

    // Get the contents of the bin file
    client.print(String("GET ") + path + clientChannel + "/" + clientDisplayType + "/" + filename + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "Cache-Control: no-cache\r\n" +
                 "Connection: close\r\n\r\n");

    unsigned long timeout = millis();

    while (client.available() == 0)
    {
      if (millis() - timeout > 5000)
      {

        // Server not available
        Serial.println("update server timeout!");

        // Show error page
        setError("Update failed. Server not responding.");

        // Stop the client
        client.stop();
        return;
      }
    }

    while (client.available())
    {

      // read headers
      String line = client.readStringUntil('\n');

      // remove space, to check if the line is end of headers
      line.trim();

      // If the the line is empty, this is end of headers
      // break the while and feed the remaining "client" to the Update.writeStream();
      if (!line.length())
      {
        // Headers ended
        // Get the OTA started
        break;
      }

      // Check if the HTTP Response is 200 (OK)
      // else break and Exit Update
      if (line.startsWith("HTTP/1.1"))
      {
        if (line.indexOf("200") < 0)
        {
          Serial.println("update failed: server not responding");

          // Show error page
          setError("Update failed. Server not responding.");

          break;
        }
      }

      // extract headers: Get content length
      if (line.startsWith("Content-Length: "))
      {
        contentLength = atol((getHeaderValue(line, "Content-Length: ")).c_str());
        Serial.println("got " + String(contentLength) + " bytes from server");
      }

      // extract headers: Get content type
      if (line.startsWith("Content-Type: "))
      {
        String contentType = getHeaderValue(line, "Content-Type: ");
        Serial.println("got " + contentType + " payload");
        if (contentType == "application/octet-stream")
        {
          isValidContentType = true;
        }
      }
    }
  }
  else
  {

    // Server not available
    Serial.println("connection to " + String(host) + " failed.");

    // Show error page
    setError("Update failed. Server not responding.");
  }

  // Check what is the contentLength and if content type is "application/octet-stream"
  Serial.println("contentLength : " + String(contentLength) + ", isValidContentType : " + String(isValidContentType));

  // check contentLength and content type
  if (contentLength && isValidContentType)
  {
    // Check if there is enough space to OTA update
    bool canBegin = Update.begin(contentLength);

    if (canBegin)
    {
      Serial.println("begin update. this may take 2-5 mins to complete. please wait.");

      // No activity would appear on the Serial monitor
      size_t written = Update.writeStream(client);

      if (written == contentLength)
      {
        Serial.println("update sucessfull: written : " + String(written));
      }
      else
      {
        Serial.println("update failed: written only : " + String(written) + "/" + String(contentLength));

        // Show error page
        setError("Update failed.");
      }

      if (Update.end())
      {

        if (Update.isFinished())
        {
          // Sucessfull Update
          Serial.println("update successfull: rebooting");
          reset();
        }
        else
        {
          Serial.println("update failed: something went wrong");
        }
      }
      else
      {
        Serial.println("update failed: error #: " + String(Update.getError()));
      }
    }
    else
    {

      // Not enough space to begin OTA
      Serial.println("Not enough space to begin OTA");

      // Show error page
      setError("Update failed. Not enough space to begin OTA");

      client.flush();
    }
  }
  else
  {
    Serial.println("there was no content in the response");

    // Show error page
    setError("Update failed. There was no content in the response");

    client.flush();
  }
}
```