espressif/arduino-esp32 #4746: xTaskGetHandle is not available to be used

VoLinhTruc, 2021-01-26:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.4
IDE name:						Arduino IDE and Platform.io
Flash Frequency:					80Mhz?
PSRAM enabled:                                    yes
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I think all API supported by SDK can be used in Arduino core.
I found xTaskGetHandle API is described here but I cannot used it in Arduino IDE and in PlatformIO IDE (in VSCode).
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#include "Arduino.h"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"

void setup() {
  // put your setup code here, to run once:

	TaskHandle_t task_handle_from_name = xTaskGetHandle(key_ptr);

}

void loop() {
  // put your main code here, to run repeatedly:

}

```

### Debug Messages:
```
undefined reference to `xTaskGetHandle'
```

RafigRzayev, 2021-01-27:
If I am not mistaken, there are other features as well which are available in RTOS API and are not usable in Arduino by default.  As I remember there is a file where you can specifically edit which RTOS features you want to be enabled. Though maybe you will need to compile new libraries. Here are the notes from https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos.html#_CPPv414xTaskGetHandlePKc :

> TaskHandle_txTaskGetHandle(const char *pcNameToQuery)
Note
This function takes a relatively long time to complete and should be used sparingly.
Return
The handle of the task that has the human readable name pcNameToQuery. NULL is returned if no matching name is found. **INCLUDE_xTaskGetHandle must be set to 1 in FreeRTOSConfig.h for pcTaskGetHandle() to be available.**

Kiznoh, 2021-01-27:
Hi @VoLinhTruc,
unfortunetly the function `xTaskGetHandle` is not avaible in ESP-IDF v3.3 which Arduino-esp32 is currently based on.

VoLinhTruc, 2021-01-28:
> Hi @VoLinhTruc,
> unfortunetly the function `xTaskGetHandle` is not avaible in ESP-IDF v3.3 which Arduino-esp32 is currently based on.

Thank you, sir.
I hope I can use that function in future. That function is the very useful function to get the Task handle by string from user.

VoLinhTruc, 2021-01-28:
I got the answer, so the issue is closed now.

stodge, 2021-02-20:
The version of FreeRTOS provided is V8.2.0 from 2015. The latest version of FreeRTOS is V10.4.3. Will FreeRTOS ever be upgraded? Or can't the ESP32 support it?

capedra, 2021-07-17:
@ivankravets @me-no-dev 
That's exactly what I have been looking for......
https://www.freertos.org/vTaskGetInfo.html

maxgerhardt, 2021-09-07:
Compiling the sketch

```cpp
#include <Arduino.h>

void setup(){
    Serial.begin(115200);
    TaskStatus_t xTaskDetails;
     // Use the handle to obtain further information about the task.
    vTaskGetInfo( NULL, // get info for *this* thread
               &xTaskDetails,
               pdTRUE, // Include the high water mark in xTaskDetails.
               eInvalid ); // Include the task state in xTaskDetails.
}

void loop(){}
```

with Arduino-ESP32 2.0.0 still leads to the result

>C:\Users\Max\Documents\Arduino\sketch_sep07a/sketch_sep07a.ino:7: undefined reference to `vTaskGetInfo'
collect2.exe: error: ld returned 1 exit status
exit status 1

Any update on enabling that in precompiled ESP-IDF libraries?

Maksons, 2021-11-12:
I do not use Arduino but in plane ESP-IDF it is still problem .
I find https://github.com/espressif/esp-idf/issues/5519 which report that with FreeRTOS 10  xTaskGetHandle should be by default, but do not work form me, not found still.

I overcome problem by solution given on ESP forum : 

    INCLUDE_xTaskGetHandle must be set to 1 in FreeRTOSConfig.h for xTaskGetHandle() to be available.

For that I need copy and modify freertos component in local code base.
@igrr  could  `INCLUDE_xTaskGetHandle 1`  be by default for FreeRTOS or possible to selected in configuration ?

mishaturnbull, 2021-11-12:
Same here.  I've been modifying the value in the FreeRTOS header to use the function, but this means anybody else who wants to compile my code has to then go do the same.  Any chance we could get it to be a menuconfig item?

p.s. @Maksons, how you are copying & modifying FreeRTOS component?  Do you just copy the component from the esp-idf components folder into your project's components -- does that override the default component from esp-idf?

Thanks!!

Maksons, 2021-11-15:
@mishaturnbull 
Yes, I copy components from esp-idf folder to project's components .
Yes, local one override original one. It is described in : 
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#multiple-components-with-the-same-name

igrr, 2022-04-05:
INCLUDE_xTaskGetHandle  is set to 1 in IDF release/v4.4, so this feature should be availble in arduino-esp32 v2.0.0 and later:
 https://github.com/espressif/esp-idf/blob/c29343eb94d2f2ca17b3a5b38c82452e556147f2/components/freertos/include/esp_additions/freertos/FreeRTOSConfig.h#L255

tshcherban, 2023-01-02:
Its still not available, im on 2.0.5 core

gfvalvo, 2023-01-26:
> @igrr Its still not available, im on 2.0.5 core, updating define:
> `#define configUSE_TRACE_FACILITY 1`
> does not help

Agree. As of Core 2.0.6, trying to call vTaskGetInfo() still reports "undefined reference to `vTaskGetInfo'". This function, like xTaskGetHandle() depends on configUSE_TRACE_FACILITY. In FreeRTOSConfig.h we have:

```
#ifdef CONFIG_FREERTOS_USE_TRACE_FACILITY
#define configUSE_TRACE_FACILITY                        1       /* Used by uxTaskGetSystemState(), and other trace facility functions */
#endif
```
But, I can't find 'CONFIG_FREERTOS_USE_TRACE_FACILITY' defined anywhere. So, that could be the problem.

bobemoe, 2023-02-22:
Trying to use this in 2.0.6 too, but getting similar errors. Can this be re-opened or should we create new issue?

VojtechBartoska, 2023-02-23:
Can you please retest this under 2.0.7 version please?

me-no-dev, 2023-02-23:
@VojtechBartoska no need to retest. This has not changed, but will probably be part of the v3.0.0 update

Jason2866, 2023-04-17:
Not a good idea to add always.
```
NOTE: This function is intended for debugging use only as its use results in the scheduler remaining suspended for an extended period.
```
from https://www.freertos.org/vTaskGetInfo.html

Maksons, 2023-04-17:
> Not a good idea to add always.
> 
> ```
> NOTE: This function is intended for debugging use only as its use results in the scheduler remaining suspended for an extended period.
> ```
> 
> from https://www.freertos.org/vTaskGetInfo.html

Initial subject is about `xTaskGetHandle` ...

me-no-dev, 2023-07-18:
Actually I would say, we will not be providing this feature, because it will cause performance degradation for every user. People that really need this, can use the lib-builder and recompile their own libs with that enabled.

zekageri, 2023-10-13:
It could be added with a define or a compiler flag. :/ At least give us the choice to choose!

VoLinhTruc, 2024-09-22:
> It could be added with a define or a compiler flag. :/ At least give us the choice to choose!

I agree with your idea. this feature should be an option by enable MACRO

Jason2866, 2024-09-22:
@VoLinhTruc Not possible, since Arduino is using static compiled libs. There is no way to change. Only way is to compile this lib with the feature enabled. This will not be done in general since it degrees performance. The other option is to use Arduino as a component of IDF and set all settings like you want/need.
There is no other choice. Like being pregnant. Either you are or not. There is no in between.

igrr, 2024-09-22:
I would suggest revisiting this question. AFAICT, simply enabling `configUSE_TRACE_FACILITY` adds very little run-time overhead — just a few places where additional structure members are set when FreeRTOS objects are initialized, and given that these functions also allocate memory, this overhead is probably insignificant.

closed 2025-01-16 by Parsaabasi