espressif/arduino-esp32 #4749: Static IP of ESP32 in AP mode (AP+STA) has different value for clients (incremented by one)

RafigRzayev, 2021-01-27:
I am using ESP32 as an access point, and set its IP to a predefined static value through `softAPConfig().` Clients connect to ESP32 AP without problem. However, later when clients send HTTP requests to the static IP, the IP is not reachable to them. 

After checking ESP32 network parameters both on mobile phones and PC, it was visible that the IP address of network is different from the one we have originally set in the ESP32. Its value is incremented by one. 

For example, static IP is **100.100.100.100**, 
 - if client sends request to this IP, requests fails
 - if client sends the request to **100.100.100.101**, everything works well.

The screenshot below shows the network information for ESP32 AP with static IP setting of **192, 168, 1, 160**. IP4 addresses don't match. 
[image]

You can observe the behaviour with the sketch below: 
```cpp
#include "WiFi.h"

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.print("Setting AP (Access Point)…");
  WiFi.mode(WIFI_AP_STA);  // AP + STA required because of wifi scan during AP
  const char* ssid = "MyESP32";
  const char* password = "abcdefghijk";
  if (!WiFi.softAP(ssid, password)) {
    Serial.println("AP Failed");
    return;
  }
  delay(2000);
  // Set static IP
  IPAddress AP_LOCAL_IP(192, 168, 1, 160);
  IPAddress AP_GATEWAY_IP(192, 168, 1, 4);
  IPAddress AP_NETWORK_MASK(255, 255, 255, 0);
  if (!WiFi.softAPConfig(AP_LOCAL_IP, AP_GATEWAY_IP, AP_NETWORK_MASK)) {
    Serial.println("AP Config Failed");
    return;
  }
  // Show the local IP
  Serial.print("AP IP address: ");
  IPAddress IP = WiFi.softAPIP();
  Serial.println(IP);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Am I doing something wrong or is it normal behavior?

Kiznoh, 2021-01-27:
Hi, 
the static IP address `AP_LOCAL_IP` is for the ESP, your client can't have the same IP otherwise there would be an address colision.
The client obtains its address using DHCP which the ESP provides.

To connect to the ESP from the client, you'd send the request to the ESP's IP address. That would be the address printed to the serial in your sketch. (in this case 192.168.1.160)

RafigRzayev, 2021-01-27:
@Kiznoh Hi, thank you for your message. 

I think you misunderstood the issue. By "client", I mean PC, phone and etc. Client is a device independent from the ESP32. I am not aiming to assign an IP address to the client. 

I am using ESP as a server in AP mode. Phone checks available WiFi connections, and selects the WiFi broadcasted by ESP. Connects successfully. Later, phone tries to send a GET request to the ESP server. Phone looks into IP address of ESP WiFi to infer the IP address of the server. After identifying the IP, phone tries to send HTTP request to this IP (to ESP). But HTTP request fails, because IP address of server that the phone inferred doesn't match with the actual IP address of ESP. E.g. if I set static IP for ESP32 as 1.1.1.1, phone should be able to extract this IP from WiFi information in order to send the HTTP requests. Instead phone sees an IP address 1.1.1.2, sends a request to non-existing IP, and thus HTTP fails. The issue here is that IP value we set to ESP32 is not usable by the "clients" that connect to AP. The screenshot that I shared above, demonstrates it. The PC (client) connects to ESP WiFi. Checks its IP address in the settings. The IP is not the one I set in the sketch. I hope was able to clarify.

Kiznoh, 2021-01-27:
Hi @RafigRzayev,
by client I also ment a wireless device (phone, PC...) connecting to the ESP in softAP mode. When the clients connects to the wireless network, it does not have an IP address. The client sends a DHCP request which the ESP responds to. The ESP has its own IP address `AP_LOCAL_IP` and can't offer that to the client so it offers the next avaible address, which is "one higher". 

I suggest you remove the static IP part and use the default like the [WiFiAccessPoint](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino) example. In that case, the ESP always has an address of 192.168.4.1 (which you can get as the gateway or DNS server value in the network properties)

RafigRzayev, 2021-01-28:
@Kiznoh Hi, Thank you very much for your help. I guess I understand the problem now. The IP address that the client sees in the WiFi settings is not the IP of ESP32, but IP given to the client by ESP32. Naturally, sending the HTTP request to this IP would mean sending it to yourself(clients sends to client). I though that IP address visible in the settings belongs to ESP, and wondered why is it incorrect :)
[image]

closed 2021-01-28 by RafigRzayev