espressif/arduino-esp32 #4750: Esp not working when serial disconnected from computer

Elsaligi, 2021-01-27:
Hi..am facing a weird problem with esp32 and arduino am working with.. when debugging port connected to my computer everything works well.. if I disconnect ftdi cable everything stops working... I tried with different baud rates too . But same issue...

Any body faces similar issues?
I am sure it is nothing to do with hardware as same hardware is working with different codes.

chegewara, 2021-01-27:
Did you check power supply?

Kiznoh, 2021-01-27:
Hi, can you post a picture of you setup?

pipi61, 2021-01-27:
Enable pin pull up resistor?
https://electronics.stackexchange.com/questions/428531/minimal-hardware-requirements-esp32-wroom/445811

Elsaligi, 2021-01-28:
> Did you check power supply?

Actually supply vcc is provided from external supply even during debugging .only tx,rx, gnd lines were used for debugging also

Elsaligi, 2021-01-28:
> Enable pin pull up resistor?
> https://electronics.stackexchange.com/questions/428531/minimal-hardware-requirements-esp32-wroom/445811

Thank you so much..I will check this.. but my problems seems to be weird as same setup working with another code ..will it happen?

Elsaligi, 2021-01-28:
> Enable pin pull up resistor?
> https://electronics.stackexchange.com/questions/428531/minimal-hardware-requirements-esp32-wroom/445811

Hi,

I checked with my schematic, I pulled up EN with a 10K resistor...

closed 2021-07-09 by stale[bot]