espressif/arduino-esp32 #4751: AsyncUDPMulticastServer.ino not responding on Multicast-IP

fliesskomma, 2021-01-28:
Hi there,
AsyncUDPMulticastServer.ino from AsyncUDP examples does not respond in Arduino-Serial-Monitor on Multicast-IP and Port.
Responding works as expected on all broadcasts and local IPs, but not on defined Multicast IP.
These are the adresses I'm sending to:
Broadcast: 192.168.178.51:1234 (responding)
Unicast: 192.168.178.51:1234 (responding)
Multicast: 239.1.2.3:1234 (not responding)

See here:
[image]

Here's the example code:
```
#include "WiFi.h"
#include "AsyncUDP.h"

const char * ssid = "SSID";
const char * password = "PASSWORD";

AsyncUDP udp;

void setup()
{
    Serial.begin(115200);
    delay(200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.println("WiFi Failed");
        while(1) {
            delay(1000);
        }
    }
    if(udp.listenMulticast(IPAddress(239,1,2,3), 1234)) {
        Serial.print("UDP Listening on IP: ");
        Serial.println(WiFi.localIP());
        udp.onPacket([](AsyncUDPPacket packet) {
            Serial.print("UDP Packet Type: ");
            Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
            Serial.print(", From: ");
            Serial.print(packet.remoteIP());
            Serial.print(":");
            Serial.print(packet.remotePort());
            Serial.print(", To: ");
            Serial.print(packet.localIP());
            Serial.print(":");
            Serial.print(packet.localPort());
            Serial.print(", Length: ");
            Serial.print(packet.length());
            Serial.print(", Data: ");
            Serial.write(packet.data(), packet.length());
            Serial.println();
            //reply to the client
            //packet.printf("Got %u bytes of data", packet.length());
        });
        //Send multicast
        udp.print("Hello!");
    }
}

void loop()
{
    delay(1000);
    //Send multicast
    udp.print("Anyone here?");
}
```
As I understand this part of the above code, the first line distinguishes the packet type, and the 6th line prints the packet type to Serial Monitor:
```
if(udp.listenMulticast(IPAddress(239,1,2,3), 1234)) {
        Serial.print("UDP Listening on IP: ");
        Serial.println(WiFi.localIP());
        udp.onPacket([](AsyncUDPPacket packet) {
            Serial.print("UDP Packet Type: ");
            Serial.print(packet.isBroadcast()?"Broadcast":packet.isMulticast()?"Multicast":"Unicast");
            Serial.print(", From: ");
            Serial.print(packet.remoteIP());
            Serial.print(":");
            Serial.print(packet.remotePort());
            Serial.print(", To: ");
            Serial.print(packet.localIP());
            Serial.print(":");
            Serial.print(packet.localPort());
            Serial.print(", Length: ");
            Serial.print(packet.length());
            Serial.print(", Data: ");
            Serial.write(packet.data(), packet.length());
            Serial.println();
```
The Serial Monitor should type the packet types Broadcast, Unicast, or Multicast.
Well, it doesn't work for Multicast so.

For testing I used the Windows App 'Packet Sender'. Next to the 'Send' buttons 1, 2 and 3, in the screenshot, you see the parameters that were sent:
[image]

Does anybody have an idea, whats wrong?

### Hardware
Lolin32 Development board (ESP32)
Arduino IDE

fliesskomma, 2021-01-31:
Hey there,
does anybody have any idea, how to solve the above Problem? Or does anybody have similar unsolved Problems? I read many postings about similar problems with ESP8266. But I couldn't transfer the solution to ESP32.

Any ideas?

jpcavanagh, 2021-01-31:
I don't know much about multicast in this situation.  I suggest that you ensure the multicast packet actually makes it onto the network and there reaches the ESP.  I say this because it seems the multicast address is on a different IP network and the PC may not have a route for it.

closed 2021-07-08 by stale[bot]

nhatuan84, 2021-12-11:
This issue still exists!!!