espressif/arduino-esp32 #4752: WiFi hostname set/get is not working correctly

miksumin, 2021-01-28:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.5-rc1
IDE name:						Arduino IDE 1.8.7
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Windows 7

### Description:
Starting core version 1.0.5-rc1 and up to core version 1.0.5-rc6 WiFi hostname set/get is not working correctly.
Anything you set by calling WiFi.setHostname("Hostname") then you call WiFi.getHostname() and you get "espressif".

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <WiFi.h>

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.setHostname("Hostname");
  WiFi.begin(WIFI_SSID, WIFI_PASS);
    if (WiFi.status() == WL_CONNECTED) {
      Serial.println(WiFi.getHostname());
    }
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

kapyaar, 2021-01-28:
Can you try 
WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);
WiFi.setHostname("hostName");
//https://github.com/espressif/arduino-esp32/issues/2537#issuecomment-508558849

me-no-dev, 2021-02-16:
This has been fixed here: https://github.com/espressif/arduino-esp32/commit/ad4cf1461bea3fb1378402b572383813ea9c2a06

next RC will have it

closed 2021-02-16 by me-no-dev

Heisenberg2980, 2022-02-18:
@me-no-dev I am still having this issue. I am using Visual Studio Code, how can I check if my sketch is using the version of arduino-esp32 that contains that fix?

me-no-dev, 2022-02-22:
@Heisenberg2980 any chance you can try this with github version of master instead?

Heisenberg2980, 2022-02-22:
@me-no-dev what do you mean with "github version of master"?

Heisenberg2980, 2022-02-22:
@me-no-dev thank you for the info, how do you know the version of the esp core I am using, is it inside of one of the libraries I mentioned? 
I am happy to wait a few days for the 2.0.3 to be out, but I am not sure what I will need to update to start using that version rather than 1.0.5-rc1

me-no-dev, 2022-02-22:
Sorry @Heisenberg2980 I thought you also posted the issue and up there it's said that 1.0.5-rc1 is used. I do not know which version you are using. Updates happen through ArduinoIDE's board manager

Heisenberg2980, 2022-02-22:
@me-no-dev right, the version I have in Arduino IDE is 1.0.4, but for this sketch I am using Visual Studio Code instead, and it is weird because I am not able to find the ESP32 library anywhere in Visual Studio Code, any idea where it could be hiding?

Regarding the Arduino IDE, do you know by any change why it is just offering me versions up to 1.0.6 and not the versions 2.0.0 and above that are available from August 2021 (https://github.com/espressif/arduino-esp32/releases)?
[image]

me-no-dev, 2022-02-22:
You are probably using the old json link that is no longer updated :)
Check here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#installing-using-boards-manager

Heisenberg2980, 2022-02-22:
@me-no-dev You are completely right!! version 2.0.2 installed now in Arduino IDE, thank you very much for your help.

I just need to figure out how to use it on my sketch in VS Code.