espressif/arduino-esp32 #4754: Is it possible to configure the NTP query interval?

fmgomes, 2021-01-30:
I'm using the built-in SNTP functionality, using the development release and the following configuration :

    sntp_setoperatingmode(SNTP_OPMODE_POLL);
    sntp_setservername(0, (char*)"pool.ntp.org");
    sntp_set_time_sync_notification_cb(time_sync_notification_cb);
    sntp_set_sync_mode(SNTP_SYNC_MODE_SMOOTH);
    sntp_init();

I've noticed that the time drifts more than one millisecond per minute (about 90ms in an hour, when it re-syncs using SNTP, since the default SNTP sync period is 1 hour). I need to have a lower time error, one possibility could be to reduce the SNTP polling period to 1 minute using sntp_set_sync_interval(60000) but if I add that function to the code above I have a compilation error:

error: 'sntp_set_sync_interval' was not declared in this scope

     sntp_set_sync_interval(60000); 

Another possibility could be to do a clock calibration, but I don't know if that is possible. I'm not using the sleep mode, the eSP32 is always working full power.

Thank you in advance!

Fernando

lbernstone, 2021-01-30:
Development release is still esp-idf 3.3.  This function was added in 4.0.  If you want this, you can install the [experimental branch](https://learn.adafruit.com/adafruit-metro-esp32-s2/arduino-ide-setup).

fmgomes, 2021-01-30:
Hi Ibernstone, thank you, I'll try it!

Best regards

Fernando

closed 2021-07-08 by stale[bot]

anteph, 2021-09-03:
Hi, I found this thread while playing with SNTP. Just to confirm I was able to use the sntp_set_sync_interval function with the latest Arduino core release (2.0.0).

The interval seems to be honored, as I've set a callback function for testing and it is being triggered with the expected periodicity after using the sntp_set_sync_interval function.

The only thing I've noticed is that, for the particular sntp_set_sync_interval function, I had to add the following include:
#include "esp_sntp.h". Nonetheless, in other tests, I was able to include the sntp_get_sync_interval function without this include, which I found weird.

I wrote a quick tutorial on how to use the function, will leave it here in case someone else is looking to do the same and ends up on this issue:
https://techtutorialsx.com/2021/09/03/esp32-sntp-additional-features/#Setting_the_SNTP_sync_interval

Hope it helps!