espressif/arduino-esp32 #4755: PUT Request not working in 1.0.5-rc6

v11, 2021-01-30:
### Hardware:
Board: ESP32 Lolin D32
Core Installation version: arduino-esp32 1.0.5-rc6
IDE name: vscode with arduino-plugin
Partion Sheme: Minimal SPIFFS
Flash Frequency: 80Mhz
PSRAM enabled: disabled
Upload Speed: 115200
Computer OS: Mac OSX

### Description:
With the following function i write data into Firebase. This functions works as expected in 1.0.4. When i changed to 1.0.5-rc6 my other application code works flawlessly but no data will be sent to Firebase with this function. When i downgrade back to 1.0.4 everything works fine. 

Any help is very welcome.

### Sketch:
```
void setDeviceStats(char *host)
{

  // This function sends device stats to Firebase (battery voltage, battery percentage)
  // Note: For sucessful PUT / POST request it is necessary to listen and wait for the response of the Server

  Serial.println("");
  Serial.println("set stats");

  // Wifi Client for secure connection (HTTPS)
  WiFiClientSecure client;

  // Add Certificate to client
  client.setCACert(cert_rest);

  // Connect to Host
  Serial.println("connect to server via port 443");

  // HTTPS Connection: Check if Secure Connection is successfull
  if (!client.connect(host, 443))
  {
    // Connection Failed
    Serial.println("connection failed!");

    char err_buf[100];
    if (client.lastError(err_buf, 100) < 0)
    {
      // Show the last Error-Message when something goes wrong
      Serial.println(err_buf);
    }
    else
    {
      Serial.println("connection error");
    }

    setError("Send device stats: Connection failed");
    return;
  }

  // JSON Parse: ArduinoJson
  // JSON Parse: Allocate the JSON document
  // JSON Parse: Use https://arduinojson.org/v6/assistant/ to calculate the capacity
  DynamicJsonDocument doc(192);

  // Asign the data
  // doc["version"] = version;
  doc["version"] = version;
  doc["bootCount"] = String(bootCount);
  doc["lastUpdate"] = currentTime;
  doc["wifiError"] = wifiError;
  doc["batteryPercentage"] = getBatteryPercentage();
  doc["batteryVoltage"] = getBatteryVoltage();

  // HTTPS Connection: Request URL with
  String url = "/devices/" + macAddress + "/stats.json";
  Serial.print("send json to url: ");
  Serial.println(host + url);

  // Assign URL
  client.println("PUT " + url + " HTTP/1.1");

  // Send the HTTP headers
  client.print("Host: ");
  client.println(host);
  client.println("Connection: close");
  client.print("Content-Length: ");
  client.println(measureJson(doc));
  client.println("Content-Type: application/json");
  client.println();

  // Send JSON document in body
  serializeJson(doc, client);

  // Wait for response
  while (client.connected())
  {
    String line = client.readStringUntil('\n');
    if (line == "\r")
    {
      Serial.println("headers received");
      break;
    }
  }

  // Read response from Server
  // It seems that the server cancels the requests if the client prematurely closes the connection.
  // This explains why you need to wait for the complete response before calling client.stop().
  while (client.available())
  {
    char c = client.read();
    Serial.write(c);
  }
  Serial.println("");

  client.stop();
}
```

### Debug Messages:

With 1.0.5-rc6 (Not working, no data will be sent to firebase)
```
set stats
connect to server via port 443
send json to url: inklay-ad9a1-default-rtdb.europe-west1.firebasedatabase.app/devices/AC%3A67%3AB2%3ADB%3A77%3A70/stats.json
```

me-no-dev, 2021-02-03:
can you set the debug level of ESP32 Arduino to "Debug" in the board menu? That might give more info on what is going on

v11, 2021-02-03:
Thanks. In the meantime I switched back to 1.0.4
Unfortunately i don't have time to elaborate this issue further.

Should we close it?

me-no-dev, 2021-02-15:
Here is what worked:

```cpp
    StreamString s;
    // Assign URL
    s.println("PUT " + url + " HTTP/1.1");

    // Send the HTTP headers
    s.print("Host: ");
    s.println(host);
    s.println("Connection: close");
    s.print("Content-Length: ");
    s.println(measureJson(doc));
    s.println("Content-Type: application/json");
    s.println();

    // Send JSON document in body
    serializeJson(doc, s);

    // Send request to server
    client.print(s);
```

In essence, the request was built into `StreamString` and then sent at once to the server

v11, 2021-02-18:
Ok, thank you for your time and give feedback.

closed 2021-02-18 by v11