espressif/arduino-esp32 #4760: second I2C not working when WiFi is in use

riccalisci, 2021-02-01:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			1.0.3
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
I  trying to use two I2C at the same time. the first on pins 21 and 22 (hardware I2C pins), the second one on 27 and 13.
Everithigs works perfectly if i don't use wifi connection. The problem seems in declaration "WiFi.mode(WIFI_STA);" (the same ADC2 problem right?).

When power on Wifi, the second i2c (27,13) not working. I tryed to change pin, but nothing change. I use ADS1115 on the first I2C and BH1750 on second. 
If i connected both  sensor on first I2C (standard 21,22) works also when wifi is on.

It's not possible use two different i2c and wifi connection at the same time? 
thanks a lot.

makermelissa, 2021-02-10:
I'm getting a similar situation. If I use I2C on some other pins (like 33 and 34), I2C works fine, but WiFi does not connect. 

If I try pins 21 and 22, I get a Core 0 panic:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009f7c6  PS      : 0x00060a30  A0      : 0x8009ced0  A1      : 0x3ffd20c0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffc7a7a  A5      : 0x00000000  
A6      : 0x3ffd74ec  A7      : 0x00000032  A8      : 0x8009f7c3  A9      : 0x3ffd2080  
A10     : 0x00000000  A11     : 0x3ffc7a7a  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000021  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000001  LCOUNT  : 0x400278c5  

Backtrace:0x4009f7c3:0x3ffd20c0 0x4009cecd:0x3ffd20e0 0x4009cf41:0x3ffd2100 0x40088c1c:0x3ffd2120 0x4008fe81:0x3ffd2140 0x4009218a:0x3ffd2170 0x400a04dd:0x3ffd2190

ELF file SHA256: 0000000000000000

Rebooting...
```

If I don't use I2C at all, WiFi has no problems.

makermelissa, 2021-02-10:
@ladyada, fyi

makermelissa, 2021-02-10:
I forgot to mention for me this occurs on the ESP32-S2 using the idf-release/v4.2 branch.

closed 2021-07-08 by stale[bot]