espressif/arduino-esp32 #4762: Request for 1.0.5-rc6 http_config core_id

jameszah, 2021-02-01:
The struct http_config in the esp-idf has a member core_id to assign which esp32 core an http server task will run on, but that has not been included in the arduino framework version 1.0.5-rc6. 

I'm trying to keep wiifi and wifi service away from core 1 where the fast code is running.  My workaround was to launch a new task on core 0, to handle that http server work, but for small requests that seems like extra processing.

Wondering if that could be added?  😄 

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#_CPPv412httpd_config

https://github.com/espressif/esp-idf/blob/6e77694/components/esp_https_server/include/esp_https_server.h

https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp_https_server/esp_https_server.h

lbernstone, 2021-02-01:
You would need to get that backported in esp-idf v3.3, which 1.0.5 is based upon.  It is available in the [experimental branch](https://learn.adafruit.com/adafruit-metro-esp32-s2/arduino-ide-setup).

jameszah, 2021-02-02:
ok, so I assume the esp-idf v4.2 arduino version will eventually arrive as a successor to 1.05, so it will arrive at some point.
Thanks.

closed 2021-02-02 by jameszah