espressif/arduino-esp32 #4768: Error in EEPROM.cpp

AlfonsoAlejandre, 2021-02-02:
I have observed malfunctions in my program after a eeprom.readString(my_dir, my_str, my_maxLen);

I have data stored in the eeprom, and a signature to test if the eeprom is initialized or not. When the device start for first time, there is not signature, and I receive a random string, but sometimes the device becomes erratic after the call to the function.

After a lot of test I concluded that there is a problem in the function, and sometimes writes in the str* parameter beyond its limit.

I tried to fix the problem adding two lines to the EEPROM.cpp function. Now, in my case , it’s working ok. The new lines are preceded by the + symbol

size_t EEPROMClass::readString (int address, char* value, size_t maxLen)
{
  if (!value)
    return 0;

  if (address < 0 || address + maxLen > _size)
    return 0;

  uint16_t len;
  for (len = 0; len <= _size; len++)
    if (_data[address + len] == 0)
      break;

  if (address + len > _size)
    return 0;

+  if (len > maxLen)
+    return 0;

  memcpy((uint8_t*) value, _data + address, len);
  value[len] = 0;
  return len;
}

Thank you for your attention.

Alfonso Alejandre

syvic, 2021-02-24:
Same problem here. Also, I confirm that adding:

    if (len > maxLen)
    return 0;

fixes the problem...

me-no-dev, 2021-02-24:
now question is wether it's not better to return part of the string up to `maxLen` long, instead of 0?

closed 2021-02-24 by me-no-dev