espressif/arduino-esp32 #4769: Interrupt wdt timeout error triggered  when use Udp.write in interrupt function

rongshengqin, 2021-02-03:
when i use Udp.write to send data in interrupt function, i got  Interrupt wdt timeout error. here is the error message:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x4008b0d5  PS      : 0x00060234  A0      : 0x8008c22d  A1      : 0x3ffb3be0  
A2      : 0x3ffbec24  A3      : 0x0000cdcd  A4      : 0xb33fffff  A5      : 0x00000001  
A6      : 0x00060023  A7      : 0x0000abab  A8      : 0x0000abab  A9      : 0x3ffb3d00  
A10     : 0x3ffba9a4  A11     : 0x3ffccb1c  A12     : 0x4011ccb0  A13     : 0x3f40ecd0  
A14     : 0x00000006  A15     : 0x3ffccb56  SAR     : 0x00000018  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff  

here is my code:
`#include <ETH.h>
#include <WiFi.h>
#include <WiFiAP.h>
#include <WiFiClient.h>
#include <WiFiGeneric.h>
#include <WiFiMulti.h>
#include <WiFiScan.h>
#include <WiFiServer.h>
#include <WiFiSTA.h>
#include <WiFiType.h>
#include <WiFiUdp.h>

const char* ssid = "***";
const char* password = "***";
WiFiUDP Udp;  
unsigned int udpPort = 8000; // local port to listen on
uint8_t udp_buf[] = {1, 2, 3, 4, 5, 6, 7, 8};
int led = 27;
int RDY = 26;
int on = 0;
char buff[] = "Hello Slave\n";

ICACHE_RAM_ATTR  void callBack(void)
{
  Udp.beginPacket("255.255.255.255", 8005);
  Udp.write(udp_buf, 8);
  Udp.endPacket();
  //Serial.printf("FALLING \n ");
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(100);
    Serial.print(".");
  }
  Serial.println(" connected");
  Udp.begin(udpPort);
  pinMode(led, OUTPUT);
  pinMode(RDY, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(RDY), callBack, FALLING);
}

void loop() {
  // put your main code here, to run repeatedly:
  //Udp.beginPacket("255.255.255.255", 8005);
 // Udp.write(udp_buf, 8);
  //Udp.endPacket();
  digitalWrite(led, on);
  on = 1 - on;
  delay(100);
}`

however, if i use Udp.write in the main loop no error occurs. Any help for me?

lbernstone, 2021-02-03:
Interrupts need to execute quickly.  Set a flag variable and send the packet in the main loop.

rongshengqin, 2021-02-03:
> Interrupts need to execute quickly. Set a flag variable and send the packet in the main loop.

but the same code works for esp8266

chegewara, 2021-02-03:
But esp32 is not esp8266.

me-no-dev, 2021-02-03:
Also UDP is in Flash and IRAM Interrupts MUST NOT access the flash :)

closed 2021-02-03 by me-no-dev