espressif/arduino-esp32 #4770: Mbedtls using external memory failing

shaun-mzj, 2021-02-04:
### Hardware:
Board:							esp32
Core Installation version:			1.0.5-rc6
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
PSRAM enabled:                                    yes
Upload Speed:					921600
Computer OS:                                        Mac OSX

### Description:
I've built the latest release candidate of the code manually and have enabled the following flag: `CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC` which causes the following error when accessing a web server: 

```
[E][ssl_client.cpp:36] _handle_error(): [start_ssl_client():210]: (-30592) SSL - A fatal alert message was received from our peer
[E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -30592
```

However if I switch mbedtls to use internal memory it can connect fine. Is there something else which needs to be done for the mbedtls library to use external memory?

shaun-mzj, 2021-02-07:
I've got a simple example which tries to load an ini file from a server via HTTPS which works if using internal memory but fails when switched to use external. The sdkconfig we used to build with is also below as well

This used to work until the lets encrypt cert was updated and at the same time it was switched from x3 to r3, maybe this causes the issue?

```c++
#include <WiFi.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>

const char* ssid = "ssid";
const char* password = "pass";

const char* rootCACertificate = \
                                "-----BEGIN CERTIFICATE-----\n" \
                                "MIIFjTCCA3WgAwIBAgIRANOxciY0IzLc9AUoUSrsnGowDQYJKoZIhvcNAQELBQAw\n" \
                                "TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh\n" \
                                "cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTYxMDA2MTU0MzU1\n" \
                                "WhcNMjExMDA2MTU0MzU1WjBKMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3Mg\n" \
                                "RW5jcnlwdDEjMCEGA1UEAxMaTGV0J3MgRW5jcnlwdCBBdXRob3JpdHkgWDMwggEi\n" \
                                "MA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCc0wzwWuUuR7dyXTeDs2hjMOrX\n" \
                                "NSYZJeG9vjXxcJIvt7hLQQWrqZ41CFjssSrEaIcLo+N15Obzp2JxunmBYB/XkZqf\n" \
                                "89B4Z3HIaQ6Vkc/+5pnpYDxIzH7KTXcSJJ1HG1rrueweNwAcnKx7pwXqzkrrvUHl\n" \
                                "Npi5y/1tPJZo3yMqQpAMhnRnyH+lmrhSYRQTP2XpgofL2/oOVvaGifOFP5eGr7Dc\n" \
                                "Gu9rDZUWfcQroGWymQQ2dYBrrErzG5BJeC+ilk8qICUpBMZ0wNAxzY8xOJUWuqgz\n" \
                                "uEPxsR/DMH+ieTETPS02+OP88jNquTkxxa/EjQ0dZBYzqvqEKbbUC8DYfcOTAgMB\n" \
                                "AAGjggFnMIIBYzAOBgNVHQ8BAf8EBAMCAYYwEgYDVR0TAQH/BAgwBgEB/wIBADBU\n" \
                                "BgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEEAYLfEwEBATAwMC4GCCsGAQUFBwIB\n" \
                                "FiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2VuY3J5cHQub3JnMB0GA1UdDgQWBBSo\n" \
                                "SmpjBH3duubRObemRWXv86jsoTAzBgNVHR8ELDAqMCigJqAkhiJodHRwOi8vY3Js\n" \
                                "LnJvb3QteDEubGV0c2VuY3J5cHQub3JnMHIGCCsGAQUFBwEBBGYwZDAwBggrBgEF\n" \
                                "BQcwAYYkaHR0cDovL29jc3Aucm9vdC14MS5sZXRzZW5jcnlwdC5vcmcvMDAGCCsG\n" \
                                "AQUFBzAChiRodHRwOi8vY2VydC5yb290LXgxLmxldHNlbmNyeXB0Lm9yZy8wHwYD\n" \
                                "VR0jBBgwFoAUebRZ5nu25eQBc4AIiMgaWPbpm24wDQYJKoZIhvcNAQELBQADggIB\n" \
                                "ABnPdSA0LTqmRf/Q1eaM2jLonG4bQdEnqOJQ8nCqxOeTRrToEKtwT++36gTSlBGx\n" \
                                "A/5dut82jJQ2jxN8RI8L9QFXrWi4xXnA2EqA10yjHiR6H9cj6MFiOnb5In1eWsRM\n" \
                                "UM2v3e9tNsCAgBukPHAg1lQh07rvFKm/Bz9BCjaxorALINUfZ9DD64j2igLIxle2\n" \
                                "DPxW8dI/F2loHMjXZjqG8RkqZUdoxtID5+90FgsGIfkMpqgRS05f4zPbCEHqCXl1\n" \
                                "eO5HyELTgcVlLXXQDgAWnRzut1hFJeczY1tjQQno6f6s+nMydLN26WuU4s3UYvOu\n" \
                                "OsUxRlJu7TSRHqDC3lSE5XggVkzdaPkuKGQbGpny+01/47hfXXNB7HntWNZ6N2Vw\n" \
                                "p7G6OfY+YQrZwIaQmhrIqJZuigsrbe3W+gdn5ykE9+Ky0VgVUsfxo52mwFYs1JKY\n" \
                                "2PGDuWx8M6DlS6qQkvHaRUo0FMd8TsSlbF0/v965qGFKhSDeQoMpYnwcmQilRh/0\n" \
                                "ayLThlHLN81gSkJjVrPI0Y8xCVPB4twb1PFUd2fPM3sA1tJ83sZ5v8vgFv2yofKR\n" \
                                "PB0t6JzUA81mSqM3kxl5e+IZwhYAyO0OTg3/fs8HqGTNKd9BqoUwSRBzp06JMg5b\n" \
                                "rUCGwbCUDI0mxadJ3Bz4WxR6fyNpBK2yAinWEsikxqEt\n" \
                                "-----END CERTIFICATE-----\n"\
                                "-----BEGIN CERTIFICATE-----\n"\
                                "MIICxjCCAk2gAwIBAgIRALO93/inhFu86QOgQTWzSkUwCgYIKoZIzj0EAwMwTzEL\n"\
                                "MAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2VhcmNo\n"\
                                "IEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDIwHhcNMjAwOTA0MDAwMDAwWhcN\n"\
                                "MjUwOTE1MTYwMDAwWjAyMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3MgRW5j\n"\
                                "cnlwdDELMAkGA1UEAxMCRTEwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQkXC2iKv0c\n"\
                                "S6Zdl3MnMayyoGli72XoprDwrEuf/xwLcA/TmC9N/A8AmzfwdAVXMpcuBe8qQyWj\n"\
                                "+240JxP2T35p0wKZXuskR5LBJJvmsSGPwSSB/GjMH2m6WPUZIvd0xhajggEIMIIB\n"\
                                "BDAOBgNVHQ8BAf8EBAMCAYYwHQYDVR0lBBYwFAYIKwYBBQUHAwIGCCsGAQUFBwMB\n"\
                                "MBIGA1UdEwEB/wQIMAYBAf8CAQAwHQYDVR0OBBYEFFrz7Sv8NsI3eblSMOpUb89V\n"\
                                "yy6sMB8GA1UdIwQYMBaAFHxClq7eS0g7+pL4nozPbYupcjeVMDIGCCsGAQUFBwEB\n"\
                                "BCYwJDAiBggrBgEFBQcwAoYWaHR0cDovL3gyLmkubGVuY3Iub3JnLzAnBgNVHR8E\n"\
                                "IDAeMBygGqAYhhZodHRwOi8veDIuYy5sZW5jci5vcmcvMCIGA1UdIAQbMBkwCAYG\n"\
                                "Z4EMAQIBMA0GCysGAQQBgt8TAQEBMAoGCCqGSM49BAMDA2cAMGQCMHt01VITjWH+\n"\
                                "Dbo/AwCd89eYhNlXLr3pD5xcSAQh8suzYHKOl9YST8pE9kLJ03uGqQIwWrGxtO3q\n"\
                                "YJkgsTgDyj2gJrjubi1K9sZmHzOa25JK1fUpE8ZwYii6I4zPPS/Lgul/\n"\
                                "-----END CERTIFICATE-----\n"\
                                "-----BEGIN CERTIFICATE-----\n"\
[... cut ...] qfJOQFy/9rKIJR0Y/8Omwt/8oTWgy1mdeHmmjk7j1nYsvC9JSQ6ZvMldlTTKB3zhThV1+XWYp6rjd5JW1zbVWEkLNxE7GJThEUG3szgBVGP7pSWTUTsqXnLRbwHOoq7hHwg==\n"\
[... 1024 lines cut ...]
CONFIG_UNITY_ENABLE_FIXTURE=

#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

#
# Wi-Fi Provisioning Manager
#
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16

#
# Supplicant
#
CONFIG_WPA_WPS_WARS=
```

shaun-mzj, 2021-02-18:
We've also added a component to idf which loads the file via HTTPS and called it from Arduino and it has the same issue. Calling the component from an esp-idf built program and it works. Could you please advise on why the same code when built via esp-idf would work but when built via Arduino would fail?

me-no-dev, 2021-02-18:
Are you saying that when using Arduino as IDF component it is failing, but when using only IDF it does not? Other than that, Arduino does not have the flag for external RAM set

shaun-mzj, 2021-02-18:
I'm using the esp32-arduino-lib-builder and have enabled the external memory option in the config file. When running the output that repo produces from the `./build.sh` command mbedtls seems to have issues. However when running the same mbedtls code via edp-idf it works fine. 

Disabling external memory and running `./build.sh` and using that output with Arduino works fine, for some reason enabling external memory breaks it.

me-no-dev, 2021-02-18:
hmm... since the lib builder is in reality "Using Arduino as IDF component", what happens if you change `main.cpp` in it with your Arduino sketch and run `make flash monitor ESPPORT=[your-uart-port] ESPBAUD=921600` after `./build.sh`?

shaun-mzj, 2021-02-18:
Replacing `main.cpp` with the test idf code and running as above works fine

me-no-dev, 2021-02-18:
Are you sure that you copied the resulting release from `./build.sh` to your other Arduino install? `./tools/copy-to-arduino.sh`?

shaun-mzj, 2021-02-18:
I'm building on Linux but running Arduino on Mac so am using a different script to copy but am copying the newly build one over. If I wasn't copying it over disabling the external memory flag for mbedtls and copying again wouldn't make it work?

me-no-dev, 2021-02-18:
if something is not copied over properly, your change will not apply. Are you sure that all is copied over?
https://github.com/espressif/esp32-arduino-lib-builder/blob/master/tools/copy-to-arduino.sh#L17-L22

Are you using the same version of esp32-arduino on the Mac? (BTW I use/develop the lib builder on a Mac. Maybe you can try that?)

shaun-mzj, 2021-02-18:
Thanks, I'll try building on Mac and running the same script

akshar001, 2021-02-18:
> if something is not copied over properly, your change will not apply. Are you sure that all is copied over? [...]

Yes we did everything. That you mentioned probably like 50 iterations. back and forth. 

At the last i have another code in which we just use the static mbedtls

mbedtls_memory_buffer_alloc_init( memory_buf, sizeof(memory_buf) );

Using this.

Interesting thing.

When i use internal ram it works fine.

But if i use psram malloc it crashes.

akshar001, 2021-02-18:
I'll post that code here too.

It's little messy up but it will do our job.

For that i needed to uncomment #define tag from esp_config in mbedtls component esp-idf folder

//#define MBEDTLS_MEMORY_BUFFER_ALLOC_C

I thought if i just allocate psram memory in before everything it might solve the problem but still same issue.

akshar001, 2021-02-18:
```
#include <WiFi.h>
#include <HTTPClient.h>
 #include <string.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "esp_system.h"
#include "nvs_flash.h"

#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include "lwip/netdb.h"
#include "lwip/dns.h"

#include "mbedtls/platform.h"
#include "mbedtls/net_sockets.h"
#include "mbedtls/esp_debug.h"
#include "mbedtls/ssl.h"
#include "mbedtls/entropy.h"
#include "mbedtls/ctr_drbg.h"
#include "mbedtls/error.h"
#include "mbedtls/certs.h"
#include "mbedtls/memory_buffer_alloc.h"
const char* ssid = "Akshar";
const char* password =  "sss3kk2aaaa4";

extern const uint8_t server_root_cert_pem_start[] asm("_binary_server_root_cert_pem_start");
extern const uint8_t server_root_cert_pem_end[]   asm("_binary_server_root_cert_pem_end");

#define WEB_SERVER "www.wiphone.io"
#define WEB_PORT "443"
#define WEB_URL "https://www.wiphone.io/static/releases/firmware/WiPhone-phone.ini"

static const char *TAG = "example";

const char rootCert[] = "-----BEGIN CERTIFICATE-----\n"\
"MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQYJKoZIhvcNAQELBQAwTzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2VhcmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4WhcNMzUwNjA0MTEwNDM4WjBPMQswCQYDVQQGEwJVUzEpMCcGA1UEChMgSW50ZXJuZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgUm9vdCBYMTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygch77ct984kIxuPOZXoHj3dcKi/vVqbvYATyjb3miGbESTtrFj/RQSa78f0uoxmyF+0TM8ukj13Xnfs7j/EvEhmkvBioZxaUpmZmyPfjxwv60pIgbz5MDmgK7iS4+3mX6UA5/TR5d8mUgjU+g4rk8Kb4Mu0UlXjIB0ttov0DiNewNwIRt18jA8+o+u3dpjq+sWT8KOEUt+zwvo/7V3LvSye0rgTBIlDHCNAymg4VMk7BPZ7hm/ELNKjD+Jo2FR3qyHB5T0Y3HsLuJvW5iB4YlcNHlsdu87kGJ55tukmi8mxdAQ4Q7e2RCOFvu396j3x+UCB5iPNgiV5+I3lg02dZ77DnKxHZu8A/lJBdiB3QW0KtZB6awBdpUKD9jf1b0SHzUvKBds0pjBqAlkd25HN7rOrFleaJ1/ctaJxQZBKT5ZPt0m9STJEadao0xAH0ahmbWnOlFuhjuefXKnEgV4We0+UXgVCwOPjdAvBbI+e0ocS3MFEvzG6uBQE3xDk3SzynTnjh8BCNAw1FtxNrQHusEwMFxIt4I7mKZ9YIqioymCzLq9gwQbooMDQaHWBfEbwrbwqHyGO0aoSCqI3Haadr8faqU9GY/rOPNk3sgrDQoo//fb4hVC1CLQJ13hef4Y53CIrU7m2Ys6xt0nUW7/vGT1M0NPAgMBAAGjQjBAMA4GA1UdDwEB/wQEAwIBBjAPBgNVHRMBAf8EBTADAQH/MB0GA1UdDgQWBBR5tFnme7bl5AFzgAiIyBpY9umbbjANBgkqhkiG9w0BAQsFAAOCAgEAVR9YqbyyqFDQDLHYGmkgJykIrGF1XIpu+ILlaS/V9lZLubhzEFnTIZd+50xx+7LSYK05qAvqFyFWhfFQDlnrzuBZ6brJFe+GnY+EgPbk6ZGQ3BebYhtF8GaV0nxvwuo77x/Py9auJ/GpsMiu/X1+mvoiBOv/2X/qkSsisRcOj/KKNFtY2PwByVS5uCbMiogziUwthDyC3+6WVwW6LLv3xLfHTjuCvjHIInNzktHCgKQ5ORAzI4JMPJ+GslWYHb4phowim57iaztXOoJwTdwJx4nLCgdNbOhdjsnvzqvHu7UrTkXWStAmzOVyyghqpZXjFaH3pO3JLF+l+/+sKAIuvtd7u+Nxe5AW0wdeRlN8NwdCjNPElpzVmbUq4JUagEiuTDkHzsxHpFKVK7q4+63SM1N95R1NbdWhscdCb+ZAJzVcoyi3B43njTOQ5yOf+1CceWxG1bQVs5ZufpsMljq4Ui0/1lvh+wjChP4kqKOJ2qxq4RgqsahDYVvTH9w7jXbyLeiNdd8XM2w9U/t7y0Ff/9yi0GE44Za4rF2LN9d11TPAmRGunUHBcnWEvgJBQl9nJEiU0Zsnvgc/ubhPgXRR4Xq37Z0j4r7g1SgEEzwxA57demyPxgcYxn/eR44/KJ4EBs+lVDR3veyJm+kXQ99b21/+jh5Xos1AnX5iItreGCc=\n"\
"-----END CERTIFICATE-----\n"\
"-----BEGIN CERTIFICATE-----\n"\
"MIIFKjCCBBKgAwIBAgISBBe7/VBighjcsshbqXuCmn3NMA0GCSqGSIb3DQEBCwUAMDIxCzAJBgNVBAYTAlVTMRYwFAYDVQQKEw1MZXQncyBFbmNyeXB0MQswCQYDVQQDEwJSMzAeFw0yMTAxMTgxMDU3MzJaFw0yMTA0MTgxMDU3MzJaMBUxEzARBgNVBAMTCndpcGhvbmUuaW8wggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQDYx7lCvY5y9Km+3AlmA0Pb+jrja4NXFNWR3D2r6iMqgAutqGm9LnCl9I+295HcPTH1SWpIuXfvgdy24WhNWXU7q59Rnp6VuopvqFdfgCgMvhk10pBSN8Aq8BZsR/29p4au6pQ3tSKboFlNXYRPjJln6EPQRH8M9pME+WvJcYdglwifY1dxIVXrcVMnJjQ2lC7z120Zu21R3pOvqocR+ddKs027P1kW7Ez3ROk73oGiBZR2f+Pn+OHMy6S5c4sAex3KtRFE9GnUrZI4ZMdhR1Zu88rTutM6Iou+Z8lPJ81RLa/bTATrxtIfelG3mVz2DTZ3lIv3Vr0YreybUXH0iu3dAgMBAAGjggJVMIICUTAOBgNVHQ8BAf8EBAMCBaAwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMAwGA1UdEwEB/wQCMAAwHQYDVR0OBBYEFPO5cniAMVAR3lVuGDjwYMANaLGCMB8GA1UdIwQYMBaAFBQusxe3WFbLrlAJQOYfr52LFMLGMFUGCCsGAQUFBwEBBEkwRzAhBggrBgEFBQcwAYYVaHR0cDovL3IzLm8ubGVuY3Iub3JnMCIGCCsGAQUFBzAChhZodHRwOi8vcjMuaS5sZW5jci5vcmcvMCUGA1UdEQQeMByCCndpcGhvbmUuaW+CDnd3dy53aXBob25lLmlvMEwGA1UdIARFMEMwCAYGZ4EMAQIBMDcGCysGAQQBgt8TAQEBMCgwJgYIKwYBBQUHAgEWGmh0dHA6Ly9jcHMubGV0c2VuY3J5cHQub3JnMIIBBAYKKwYBBAHWeQIEAgSB9QSB8gDwAHYAfT7y+I//iFVoJMLAyp5SiXkrxQ54CX8uapdomX4i8NcAAAF3FVt8CwAABAMARzBFAiEAuhqAjMB6rqFDHyejZu4cCyAosE+w8DOAlykmqt5eZ6MCIHwHoA+68RAx6JksrpuxkA/7REG9GvllRx7HiEEamHEKAHYAb1N2rDHwMRnYmQCkURX/dxUcEdkCwQApBo2yCJo32RMAAAF3FVt8vQAABAMARzBFAiEA0bo4rtb9iCrVo39EgGKnpnUdpieWeSgnlcgbeMc2eN8CIFOuqsGeQ8Opm+hhiKIPPuizbFo9WJTw2LsaQraTYbQqMA0GCSqGSIb3DQEBCwUAA4IBAQBt20nC3EPXOsR0Kj5ST6xjvZld540fJRnmIu2QiixL9bubd4KrT8IVFI6ksFs9AOOsfNksXIXYgEvR71TTKx4IezsLcKqo1SLkmtalebu/fWyLrv/dShr4IKRB+xHUywdayj8IkEpjRHjOiZH/f47y//ftPRKVd3xjnUs9PBzWeZw7+eZs/NlEJsiAWSltNNKiVof8CAci+rSaifbNHOd8qi4UVSnbAlfoyM/RMUI85XKspwnKvXqHPp19192nwpe8kIWQ28aA16JAmq0o8/eohK4/S4WewQHfdSn0A3zJdFOKhWX11skbuYpV1uxm4JGXT6mPeXdOgu9WTqgkEHCe\n"\
"-----END CERTIFICATE-----\n"\
"-----BEGIN CERTIFICATE-----\n"\
"MIIFFjCCAv6gAwIBAgIRAJErCErPDBinU/bWLiWnX1owDQYJKoZIhvcNAQELBQAwTzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2VhcmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMjAwOTA0MDAwMDAwWhcNMjUwOTE1MTYwMDAwWjAyMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3MgRW5jcnlwdDELMAkGA1UEAxMCUjMwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQC7AhUozPaglNMPEuyNVZLD+ILxmaZ6QoinXSaqtSu5xUyxr45r+XXIo9cPR5QUVTVXjJ6oojkZ9YI8QqlObvU7wy7bjcCwXPNZOOftz2nwWgsbvsCUJCWH+jdxsxPnHKzhm+/b5DtFUkWWqcFTzjTIUu61ru2P3mBw4qVUq7ZtDpelQDRrK9O8ZutmNHz6a4uPVymZ+DAXXbpyb/uBxa3Shlg9F8fnCbvxK [... cut ...]
//    http.begin("https://wiphone.io/static/releases/firmware/WiPhone-phone.ini", rootCACertificate); //Specify the URL and certificate
//    int httpCode = http.GET();                                                  //Make the request
// 
//    if (httpCode > 0) { //Check for the returning code
// 
//        String payload = http.getString();
//        Serial.println(httpCode);
//        Serial.println(payload);
//      }
// 
//    else {
//      Serial.println("Error on HTTP request");
//    }
// 
//    http.end(); //Free the resources
//  }
//  
//  delay(10000);
}
```

akshar001, 2021-02-18:
So if use this

// unsigned char alloc_buf[80000];

it works fine.

akshar001, 2021-02-18:
but when we use this 

 alloc_buf =   (unsigned char *)heap_caps_malloc(80000,MALLOC_CAP_SPIRAM);

akshar001, 2021-02-18:
@me-no-dev https://github.com/espressif/arduino-esp32/issues/4818#issuecomment-781153090
Same issue.

shaun-mzj, 2021-02-18:
I've also just finished trying it on Mac and using the copy script linked and it doesn't work still

me-no-dev, 2021-02-18:
i'm really confused why it would work in the lib-builder, but not when copied... it's the same files... except if some flag is missing in the build commands?

They are taken from the build in the 4.2 branch (cmake) but not on master (make). Maybe check the output running a verbose build of the lib-builder and compare flags against what is in platform.txt?

shaun-mzj, 2021-02-18:
We've both been working on this independently to try and figure it out so chances are we've both not done something dumb with copying etc. Are you able to try enabling that external memory setting for mbedtls and try it? I'll have a look and see if I can see anything with the flags

akshar001, 2021-02-18:
Same confusion here. What i have noticed is build settings, I am using mac @me-no-dev i have used this copy-to-arduino.sh one

akshar001, 2021-02-18:
@me-no-dev can you try once doing this same on your side. it might take 10 minutes only. We have one thing to launch and we are just stuck at development because of this.

shaun-mzj, 2021-02-18:
Also we are running it on v1 hardware, not sure if that matters

me-no-dev, 2021-02-18:
what do I need to test this? a board with WROVER module + the sketch above?

shaun-mzj, 2021-02-18:
I'm using a board with the following chip:  ESP32D0WDQ6 (revision 1) and Ashkar is using: ESP32-D0WDQ6-ATUB00

shaun-mzj, 2021-02-18:
Just to make sure that you know we only see the error when we enable the external memory and enable this option for mbedtls: `CONFIG_MBEDTLS_EXTERNAL_MEM_ALLOC=y`

akshar001, 2021-02-18:
@me-no-dev yes. nothing more.

shaun-mzj, 2021-02-19:
Looking at the flags I noticed this was missing: `-mfix-esp32-psram-cache-issue` but adding it didn't fix the issue for me. I'm now running everything on Mac and using Arduino version: `1.18.13` if that matters for producing the issue

akshar001, 2021-02-19:
@shaun-mzj I have tried that as well same for me.
What the arduino one do different than what we are doing now is needed to be figured out.
That's where our issue lies.
We copied everything and it still didn't worked. So there must be some different thing which should be the same

akshar001, 2021-02-19:
@me-no-dev I have also tried using uni core but it didn't helped me either,
And it's been failing at handshake for everytime in mbedtls.

closed 2021-02-22 by me-no-dev

me-no-dev, 2021-02-22:
The issue was that certain libraries that deal with PSRAM issues were not included in the linking process and were also not copied by the lib-builder. Lib-builder commit is here: https://github.com/espressif/esp32-arduino-lib-builder/commit/3fa59d5715d09ff1a499ae8ba3abc65a5d6a4d8e

You can now create a proper update with mbedts in PSRAM :)

shaun-mzj, 2021-02-23:
Thank you! I've tried it and it working