espressif/arduino-esp32 #4771: Sketch compilation fails on ESP32

aentinger, 2021-02-04:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			https://github.com/espressif/arduino-esp32/commit/6f23cd5988aad7f18807e15a691690e6b03ef1cd
IDE name:						Arduino IDE 1.8.13
Computer OS:                                        Ubuntu 18.04 LTS

### Description:
When I compile a minimal sketch utilising the C macro [isfinite](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/newlib/math.h#L212-L215) the compilation fails with the following error (although the macro clearly is available within [<math.h>](https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/newlib/math.h#L212-L215)):

```bash
ESP32-isfinite:3:17: error: 'isfinite' was not declared in this scope
   if (isfinite(f))
                 ^
ESP32-isfinite.ino:3:17: note: suggested alternative:
In file included from ArduinoCore-esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/random:38:0,
                 from ArduinoCore-esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/stl_algo.h:66,
                 from ArduinoCore-esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/algorithm:62,
                 from ~/Arduino/hardware/arduino-git/esp32/cores/esp32/Arduino.h:142,
                 from /tmp/arduino_build_231820/sketch/ESP32-isfinite.ino.cpp:1:
ArduinoCore-esp32/tools/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/cmath:596:5: note:   'std::isfinite'
     isfinite(_Tp __x)
     ^
```

### Sketch:
```cpp
void setup() {
  float f = 1.234;
  if (isfinite(f))
    f *= 2.0f;
}

void loop() {
}
```

This was discovered during a failed CI run within https://github.com/107-systems/107-Arduino-UAVCAN/pull/77.

aentinger, 2021-06-21:
Well, the issue hasn't been solved or even adressed by the Espressif staff, despite a minimum example demonstrating the problem being available. I'd love to see a resolution on this one.

igrr, 2021-07-16:
This seems to be fixed in the master branch of this repository. The fix should be part of the upcoming 2.0.0 release.
I didn't bisect the commit where this was fixed, though.

igrr, 2021-07-16:
Looks like this was fixed when the toolchain was updated from GCC 5.2.0-based one to GCC 8.2.0-based one. The difference was due to the fact that the headers in the 5.2.0-based toolchain didn't re-import this C++ function (declared in `cmath` header) from `std::` namespace. Since `Arduino.h` includes `cmath` but not `math.h`, the macros declared by math.h get undef-ed.
In the newer toolchain which will be shipped in 2.0.0, `isfinite` is correctly imported from `std::` namespace.

closed 2021-07-16 by igrr