espressif/arduino-esp32 #4772: Build in WiFi library or the AsyncWebServer library?

tsctrl, 2021-02-04:
Make your question, not a Statement, inclusive. Include all pertinent information:

I would like to use esp as soft ap server and host static html files from spiffs. but i not sure should i use the build in WiFi library or the AsyncWebServer library instead. 
1. Is the build in library have enough apis to cover the basic requirement above? 
2. What is the different between build in library and the ESPAsyncWebServer? 
3. between this two option which one library have more feature then other or is it the same library just split into different build in library in arduino-esp32?

Thank you.

tsctrl, 2021-02-04:
@me-no-dev

jandrassy, 2021-02-04:
try the examples

closed 2021-02-06 by tsctrl