espressif/arduino-esp32 #4773: SimpleWiFiServer-Example hangs on first connect #esp32s2

newHeiko, 2021-02-04:
I'm trying to get the SimpleWiFiServer example to run so I can get my feet wet with the ESP32-S2.

The code compiles and uploads, connects to my WiFi and I get a few responses from the server when connecting with my phone. After a few of them, typically hitting both links on the page once or twice, it just hangs and doesn't even answer to pings any more.

I saw a couple of issue reports and commits in the master branch, so I cherry-picked all the commits that sounded like HTTPServer or WiFiClient into a separate PR which I will link to in my next post.

Changed the example sketch to include two calls to esp_wifi_set_ps(WIFI_PS_NONE).

Added another delay to the end of the main loop.

Commented out the digitalWrite calls in case there is some HW on the board.

Still the same error.

Debug shows (among others):

[ 28339][D][WiFiClient.cpp:522] connected(): Disconnected: RES: 0, ERR: 128

...but I can't seem to find what errno 128 (or -1?) is...

Careful reading of the debug output might indicate the issue pops up when the browser tries to open two connections at once to read the favicon?

### Hardware:
Board:							ESP32-S2-WROOM on ESP32-S2-DevKit-LiPo by Olimex https://www.olimex.com/Products/IoT/ESP32-S2/ESP32-S2-DevKit-Lipo/open-source-hardware
Core Installation version:			branch: esp32s2 and esp32s2 with 10 commits cherry-picked from master
IDE name:						Arduino IDE 1.8.13
CPU Frequency:                                      240MHz
Flash Frequency:					80MHz
PSRAM enabled:                                     no
Upload Speed:				        921600
Computer OS:                                         Slackware Linux

### Description:

### Sketch:
```cpp
/*
 WiFi Web Server LED Blink

 A simple web server that lets you blink an LED via the web.
 This sketch will print the IP address of your WiFi Shield (once connected)
 to the Serial monitor. From there, you can open that address in a web browser
 to turn on and off the LED on pin 5.

 If the IP address of your shield is yourAddress:
 http://yourAddress/H turns the LED on
 http://yourAddress/L turns it off

 This example is written for a network using WPA encryption. For
 WEP or WPA, change the Wifi.begin() call accordingly.

 Circuit:
 * WiFi shield attached
 * LED attached to pin 5

 created for arduino 25 Nov 2012
 by Tom Igoe

ported for sparkfun esp32 
31.01.2017 by Jan Hendrik Berlin
 
 */

#include <WiFi.h>
#include <esp_wifi.h>

const char* ssid     = "FritzWLAN";
const char* password = "myPSK";

WiFiServer server(80);
[... 216 lines cut ...]
64 bytes from 192.168.178.97: icmp_seq=12 ttl=255 time=2.09 ms
64 bytes from 192.168.178.97: icmp_seq=13 ttl=255 time=6.86 ms
64 bytes from 192.168.178.97: icmp_seq=14 ttl=255 time=10.4 ms
64 bytes from 192.168.178.97: icmp_seq=15 ttl=255 time=2.14 ms
64 bytes from 192.168.178.97: icmp_seq=16 ttl=255 time=2.67 ms
64 bytes from 192.168.178.97: icmp_seq=17 ttl=255 time=31.9 ms
64 bytes from 192.168.178.97: icmp_seq=18 ttl=255 time=2.39 ms
64 bytes from 192.168.178.97: icmp_seq=19 ttl=255 time=12.5 ms
64 bytes from 192.168.178.97: icmp_seq=20 ttl=255 time=4.01 ms
64 bytes from 192.168.178.97: icmp_seq=21 ttl=255 time=3.36 ms
64 bytes from 192.168.178.97: icmp_seq=22 ttl=255 time=5.28 ms
64 bytes from 192.168.178.97: icmp_seq=23 ttl=255 time=2.14 ms
64 bytes from 192.168.178.97: icmp_seq=24 ttl=255 time=1.84 ms
64 bytes from 192.168.178.97: icmp_seq=25 ttl=255 time=2.26 ms
64 bytes from 192.168.178.97: icmp_seq=27 ttl=255 time=995 ms <-- Last request
64 bytes from 192.168.178.97: icmp_seq=28 ttl=255 time=60.2 ms
^C
--- 192.168.178.97 ping statistics ---
37 packets transmitted, 27 received, 27.027% packet loss, time 36261ms
rtt min/avg/max/mdev = 1.842/49.186/994.913/186.551 ms
```

Looking forward to your ideas,

Heiko

newHeiko, 2021-02-04:
Branch used: #4774

lbernstone, 2021-02-04:
Use idf-release/v4.2 branch.  esp32s2 is no longer updated.

newHeiko, 2021-02-04:
Thanks. That fixes it...

Sorry for the noise.

I'll try to find a place in the docs to make that clear...

Heiko

closed 2021-02-04 by newHeiko