espressif/arduino-esp32 #4778: Eth interface no longer receives SYSTEM_EVENT_ETH_CONNECTED event

TD-er, 2021-02-05:
### Hardware:
Board:							Olimex ES32 Gateway EA
Core Installation version:			1.0.3
IDE name:						Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        Windows 10

### Description:
On previous ESP32 SDK v3.2.3-14-gd3e562907 ETH works just fine on my project (ESPEasy)
However, with the latest version of the code on this repo it is no longer possible to run Ethernet via DHCP.
Static IP works just fine, but DHCP does not.

Found this issue, but cannot test it as I have no idea how to set the mentioned build flag: https://github.com/espressif/arduino-esp32/issues/4499

What I observed is that the `SYSTEM_EVENT_ETH_DISCONNECTED` is received just fine if I disconnect the ethernet connector, but `SYSTEM_EVENT_ETH_CONNECTED` is not seen by my code.

The ethernet device is receiving data as it does pick up UDP packets broadcasted on my network.

It does receive `SYSTEM_EVENT_ETH_GOT_IP`, but the IP addresses are all `255.255.255.255`
`12850 : Info   : ETH MAC: BC:DD:C2:F2:31:0C IPv4: 255.255.255.255 (ESPEasy) GW: 255.255.255.255 SN: 255.255.255.255 FULL_DUPLEX 100Mbps`

When I disconnect the ethernet cable and reconnect it, I do get the `SYSTEM_EVENT_ETH_GOT_IP` twice:  (line with `Info   : ETH MAC`)

```
128838 : Error  : ETH Disconnected
128858 : Info   : EVENT: ETHERNET#Disconnected
128861 : Debug  : EVENT: ETHERNET#Disconnected Processing time:3 milliSeconds
132839 : Info   : ETH MAC: BC:DD:C2:F2:31:0C IPv4: 255.255.255.255 (be-2005001-207) GW: 255.255.255.255 SN: 255.255.255.255 FULL_DUPLEX 100Mbps
132840 : Info   : ETH MAC: BC:DD:C2:F2:31:0C IPv4: 255.255.255.255 (be-2005001-207) GW: 255.255.255.255 SN: 255.255.255.255 FULL_DUPLEX 100Mbps
[E][WiFiGeneric.cpp:738] hostByName(): DNS Failed for 0.pool.ntp.org
133646 : Debug More : NTP  : NTP host 0.pool.ntp.org (0.0.0.0) queried
134648 : Debug More : NTP  : No reply
```

When I revert to an older SDK it all works just fine.

The used ETH chip is LAN8710A.

TD-er, 2021-03-11:
Just tested again with the latest code and it still is an issue.
I can't imagine I'm the only one to have this issue, so I wonder if anyone else noticed the same and/or maybe even has found a work-around?

me-no-dev, 2021-03-17:
@TD-er I just tested against latest master and can not replicate your issue.

TD-er, 2021-03-17:
I will try again to see if it is now maybe fixed.

TD-er, 2021-03-17:
I use PlatformIO and when set to use the "staging" version, it uses `ESP32 SDK v3.3.4-436-g0bfff0b25`

```
[core_esp32_stage]
platform                  = espressif32
platform_packages         = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
```

This does NOT work with LAN + DHCP, just as I described above.

When using what PlatformIO calls `3.1.0`, I apparently use `ESP32 SDK v3.3.4-432-g7a85334d8` which does show the same behavior of NOT working with LAN + DHCP.
PIO platform: `platform                  = espressif32@3.1.0`

With `platform                  = espressif32@2.1.0` it is working just fine.

me-no-dev, 2021-03-17:
please provide minimal example sketch. I can not get it to fail :)
```
ETH Started
ETH Connected
ETH MAC: 24:0A:C4:03:B7:B3, IPv4: 192.168.254.93, FULL_DUPLEX, 100Mbps

connecting to google.com
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Wed, 17 Mar 2021 12:10:26 GMT
Expires: Fri, 16 Apr 2021 12:10:26 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN

<HTML><HEAD><meta http-equiv="content-type" content="text/html;charset=utf-8">
<TITLE>301 Moved</TITLE></HEAD><BODY>
<H1>301 Moved</H1>
The document has moved
<A HREF="http://www.google.com/">here</A>.
</BODY></HTML>
closing connection

ETH Disconnected
ETH Connected
ETH MAC: 24:0A:C4:03:B7:B3, IPv4: 192.168.254.93, FULL_DUPLEX, 100Mbps

connecting to google.com
HTTP/1.1 301 Moved Permanently
Location: http://www.google.com/
Content-Type: text/html; charset=UTF-8
Date: Wed, 17 Mar 2021 12:12:37 GMT
Expires: Fri, 16 Apr 2021 12:12:37 GMT
Cache-Control: public, max-age=2592000
Server: gws
Content-Length: 219
X-XSS-Protection: 0
X-Frame-Options: SAMEORIGIN
```

TD-er, 2021-03-17:
I compared my events with this example:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/ETH_LAN8720/ETH_LAN8720.ino

And it sure has a lot of similarities.

The problem seems to be I do get the `SYSTEM_EVENT_ETH_GOT_IP` twice (almost immediately) and I don't get the `SYSTEM_EVENT_ETH_CONNECTED` event.

However, given your output it does show the ETH Connected event in the serial output.
So it must be something in my code...... (duh!)

Tested and tested and... well you get the idea ;)

This is what I found in my code, what was working on older SDK/Arduino versions:
- I set the `eth_connected` flag in the `SYSTEM_EVENT_ETH_CONNECTED`, which is not set in the example. But since I did not get that event anyway it was probably not an issue here.
- The Example does set the IP configuration after the `SYSTEM_EVENT_ETH_START` event. I did set it immediately after `ETH.begin`. Still not working, but doesn't hurt to get it in line with the example.
- Then I noticed a thing I had to put in in the past to get things working, but after removing this it started working and me receiving the `SYSTEM_EVENT_ETH_CONNECTED` again: `ETH.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);`

So that is again working.
I will do some more tests to see what is the best moment for setting static IP. It was working, but with these new insights it might be better to set it after receiving the connected event instead of setting it right after setting the hostname (on the ETH_START event).

TL;DR 
It is now working again.

me-no-dev, 2021-03-17:
`ETH.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);` could be a whole other issue :) I will take a look at it tomorrow (or now) :)

me-no-dev, 2021-03-17:
The issue was actually that the value of INADDR_NONE changed in between 1.0.4 and 1.0.5 and `ETH.Config()` was not detecting it properly. As such, your interface was configured with static IP `255.255.255.255` and since DHCP was off, `SYSTEM_EVENT_ETH_GOT_IP` was not emitted. That is now fixed in `master` and will some come out in 1.0.6

closed 2021-03-17 by me-no-dev

TD-er, 2021-03-17:
Ah great :)
Glad it did lead to a bug that could be fixed as I already classified it as being a PEBKAC issue ;)

TD-er, 2021-03-17:
By the way, I do see conflicting defines/usage for `INADDR_NONE`

Here is the define: 
https://github.com/espressif/arduino-esp32/blob/af119215353b26d944b80d7126bed68e060e6697/cores/esp32/IPAddress.h#L94

But then lines like these don't make sense: (occuring at multiple locations)
https://github.com/espressif/arduino-esp32/blob/bd4133426523cc4db35588add881f5ab78f8f4c4/libraries/WiFi/src/ETH.cpp#L121

me-no-dev, 2021-03-17:
It is also defined here: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/lwip/lwip/inet.h#L71

`IPADDR_NONE` defined here: https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/lwip/lwip/ip4_addr.h#L79

They conflict sometimes and the checks in `ETH.config` fail.

me-no-dev, 2021-03-17:
> Here is `IPADDR_NONE` (used to define `INADDR_NONE`) undefined:
> https://github.com/espressif/arduino-esp32/blob/8dc70e0add06e75da5a255ba94fe86df98ae5ef1/libraries/WiFiProv/src/WiFiProv.cpp#L33
>

It's local for the CPP file and does not interfere with any other code outside that file.

TD-er, 2021-03-17:
Ah yes, overlooked it was in a .cpp