espressif/arduino-esp32 #4782: analogRead() returns bogus values when i2s DAC running

Harvie, 2021-02-07:
Hello, i have issues with `analogRead()`  when running DMA for I2S DAC. To be honest i don't understand why running external DAC messes up internal ADC. But it surely is problem:

```C
static const i2s_port_t i2s_num = I2S_NUM_0; // i2s port number

  static const i2s_config_t i2s_config = {    
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX /*| I2S_MODE_DAC_BUILT_IN*/ ),                                                                                                                                                            
    .sample_rate = SAMPLE_RATE,                                                                                             
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,  // only the top 8 bits will actually be used by the internal DAC, but using 8 bits stra> 
    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,  // always use stereo output. mono seems to be buggy, and the overhead is insignifcant o> 
    .communication_format = (i2s_comm_format_t)(I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_LSB),  // this appears to be the correct setting> 
    .intr_alloc_flags = 0, // default interrupt priority                                                                                    
    .dma_buf_count = 4,    // 8*128 bytes of buffer corresponds to 256 samples (2 channels, see above, 2 bytes per sample per channel)      
    .dma_buf_len = BUF_LEN,    
    .use_apll = false         
  };   

i2s_driver_install(i2s_num, &i2s_config, 0, NULL);   //install and start i2s driver

//i2s_adc_disable(i2s_num);  //Tried this, did not helped.

//So far i've only tested this issue on following three pins:
analogRead(2); //always returns 4095 no matter what signal on pin
analogRead(4); //always returns 4095 no matter what signal on pin
analogRead(35); //returns correct value measured by ADC on pin 35
```

Is there any chance to get proper ADC readings on all analog pins without shutting down I2S entirely? (this would cause audio glitching, so it is not possible)

Harvie, 2021-02-07:
Problematic analog pins 2 and 4 are ADC2, while working pin 35 is ADC1, so i guess this affects whole ADC2.
BUT! When i call `adc2_get_raw(...)` directly, it does work without problem. So this really has to be a bug in arduino-esp32 core.

Harvie, 2021-07-08:
Not sure if this stale bot thing is useful... "let's just pretend problem doesn't exist because it was reported more than 5 months ago".

me-no-dev, 2021-07-08:
There are many issue reports that are left abandoned by their submitter. The stale bot is there to take care of those. It's enough to just chime-in a message if the bot puts "stale" flag and the issue will be un-staled. We will also get an email that an issue is still active, which let's us know that the problem described persists and that there is someone on the other end.
Unfortunately it not very liked by active folks :) hopefully soon we can turn it off

Harvie, 2021-09-07:
bumpz

VojtechBartoska, 2022-04-06:
@Harvie Can you please give it a shot on v2.0.3-RC1? Thanks

PilnyTomas, 2022-04-06:
Hello @Harvie, I'm trying to understand your issue, can you please help me by answering a few questions?
1. What are you trying to accomplish?
2. What purpose does serve the I2S in your sketch?
3. What are your hardware connections?
4. What values are you expecting?

Harvie, 2022-04-06:
Ahoj @PilnyTomas ,
well i was working on I2S based audio generator. I was generating audio samples, streaming them to I2S DAC and then i wanted to use analog inputs to read potentiometers which are meant to serve as user input which can parametrize the process of audio synthesis. For example: Generate sinewave, Stream it to I2S, use analog pots to set frequency and amplitude of that sinewave.

Unfortunately as soon as i've enabled the I2S DMA, some of the analog inputs started giving me wrong values.

PilnyTomas, 2022-04-07:
Ahoj @Harvie,
could you please tell me what pins are you using for the I2S?
And also which chip, module, and dev board (for example ESP32, WROOM-32D, DevkitC) so that I can use same HW to test it on.
Ideally, please post complete, minimal code that can be flashed and demonstrates the issue (updating the first post is ok).

Harvie, 2022-04-07:
> could you please tell me what pins are you using for the I2S?

It was year ago, so i don't really remember, but certainly i had used whatewer default pins for I2S_NUM_0 are, because i always try to avoid pin remapping or muxing wherever possible and there was no reason to do so.
AnalogRead(2) and AnalogRead(4) were then always returning 4095 no matter what actual inputs were.

PilnyTomas, 2022-04-07:
Hi, I tried using analogRead with I2S installed and analog reads returned 0.
Do you wish to proceed with finding out what is wrong, or it doesn't matter anymore?

Harvie, 2022-04-07:
> Do you wish to proceed with finding out what is wrong, or it doesn't matter anymore?

Yes, but i don't have the esp32 board with me right now, i will find it and check again later. It was year ago, so perhaps the issue might be already resolved in recent arduino library...

PilnyTomas, 2022-04-07:
Alright, I will wait until you test it with the latest version (2.0.3-RC1)

irumaru, 2022-04-08:
Hello.
I have the same problem.
I am looking for a solution to this problem.
I used automatic translation because my English is not good.
Sorry if it is hard to read.
Please give me your best regards.

Pin usage:
 32: i2s_adc(input)
 33: pwm_10kHz(output)
 26: analog(input)

Pin Connection
 Variable resistor on 26
 Connect 32 and 33

Environments :
 Arduino 1.8.15
 Espressif Systems version 2.0.2 (latest version shown in Board Manager)
 Board used: ESP32-DevKitC-32E (4MB)

If you comment out "int unsigned value = analogRead(26);" in "dmaRead()", the pulse is output normally.
Otherwise, the output will only be 4096. (code below)

`#include <driver/i2s.h>

#define I2S_SAMPLE_RATE 1000 //1.2M
#define ADC_INPUT ADC1_CHANNEL_4 //pin 32

void i2sInit()
{
   i2s_config_t i2s_config = {
    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
    .sample_rate =  I2S_SAMPLE_RATE,
    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
    .channel_format = I2S_CHANNEL_FMT_ONLY_RIGHT,
    .communication_format = I2S_COMM_FORMAT_I2S_MSB,
    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
    .dma_buf_count = 2,
    .dma_buf_len = 256,
    .use_apll = false,
    .tx_desc_auto_clear = false,
    .fixed_mclk = 0
   };
   i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL);
   i2s_set_adc_mode(ADC_UNIT_1, ADC_INPUT);
   i2s_adc_enable(I2S_NUM_0);
}

void dmaRead() {
  //I2S
  uint16_t buffer[512];
  size_t bytes_read;

  i2s_read(I2S_NUM_0, &buffer, sizeof(buffer), &bytes_read, 15);

  for(int i = 0; i < 512; i ++){
    Serial.println(buffer[i]&0xFFF);
  }
  
  //Analog
  int unsigned value = analogRead(26); //ADC2_Ch9
  //Serial.println(value);
}

void setup() {
  //Serial
  Serial.begin(115200);

  //PWM
  pinMode(33, OUTPUT);
  ledcSetup(0, 10000, 8);
  ledcAttachPin(33, 0);
  delay(100);
  ledcWrite(0, 128);

  //I2S
  i2sInit();
  
  //analogRead ADC2_Ch9
  pinMode(26, ANALOG);
}

void loop() {
  dmaRead();
  delay(1000);
}`

PilnyTomas, 2022-04-19:
Please refer to the [Installing guide](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) and use the second link containing development versions: `https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json` after this you will see the version `2.0.3-RC1` - please install it and try your code on this version. 
On my side, it is working on this version.

PilnyTomas, 2022-05-05:
@Harvie, @irumaru, There has been a release of the 2.0.3 stable version, available via the stable link. You can try the newest version now without the hassle of changing/adding the link in the preferences.

VojtechBartoska, 2022-05-18:
folks, any chance you gave this a shot? Thanks!

VojtechBartoska, 2022-06-14:
Closing this as expired. If it's needed, you can reopen the issue.

closed 2022-06-14 by VojtechBartoska