espressif/arduino-esp32 #4792: Wire methods does not work in web callbacks

tsctrl, 2021-02-10:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			release/4.2
IDE name:						IDF component
Flash Frequency:					40Mhz
PSRAM enabled:                                     no
Upload Speed:					115200
Computer OS:                                        Windows 10

## Environment
- Development Kit:     [ESP32-DevKitC]
- Kit version  [v1]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version :v4.3-dev-2940-g6e776946d
- Build System:         [idf.py]
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Windows
- (Windows only) environment type: Plain Command Prompt.
- Using an IDE?: Yes eclipse
- Power Supply:    USB

### Description:
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Error when calling Wire for rtc time update when calling from webserver request on core 1 
similar issue: [https://gitter.im/Makuna/Rtc?at=5d25d7e4d0e06b66757d2de9](https://gitter.im/Makuna/Rtc?at=5d25d7e4d0e06b66757d2de9)

"can't call any of the WIRE methods within the web callback (a common problem with system level callbacks)."

Anyone have solution for this issue? Or should i call the rtc update in core 1 xtask too?

### Sketch: 
```
WebServer server(80);
DS3213 rtc;

WiFi.mode(WIFI_AP_STA);
WiFi.begin(ssid, password);
server.on("/save", HTTP_PUT, handleSave);

handleSave(){
rtc.setDate(byte(server.arg("date")));
}
```

### Debug Messages:
```
Decoding stack results
0x400de734: i2cProcQueue at ../components/arduino/cores/esp32/esp32-hal-i2c.c line 1294
0x400dec0b: i2cWrite at ../components/arduino/cores/esp32/esp32-hal-i2c.c line 1644
0x400dc121: TwoWire::endTransmission(bool) at ../components/arduino/libraries/Wire/src/Wire.cpp line 150
0x400dc295: TwoWire::endTransmission() at ../components/arduino/libraries/Wire/src/Wire.cpp line 333
0x400f3fd5: DS3231::setSecond(unsigned char) at ../components/arduino/libraries/DS3231/DS3231.cpp line 242
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
```

Thank you!

wachidsusilo, 2021-02-11:
Hello @tsctrl, I don't use DS3231 because of this exact reason. TwoWire library is not thread-safe, even if your code run, you will have a bad time dealing with this I2C thing during run time. Your DS3231 time may become 165:165:165, and you need to adjust it over and over again. Currently, the solution i have is to maintain I2C communication on a single thread, then use queue to pass data between thread. I know this sounds ridiculous, but this is the only solution that works for me. Give it a try. Anyway, if you found better solution please post it here.

tsctrl, 2021-02-11:
Hi @wachidsusilo  thank you so much for your reply. Thats mean a lot. 
yes, the hour is 67 and the day is 45 for now. I will try the solution you suggested. Is that was done by modifying TwoWire library? or use as is but run all in the same thread? 

There is other solution that was suggested like using xSemaphoreTake or "semaphores for mutual exclusion" in i2c where i not really sure what it is. i will check on that one too..

wachidsusilo, 2021-02-12:
Hi @tsctrl, sorry for the late reply. Looks like we have different time zone. Per my answer, there is no need to modify `TwoWire` library. All i did is only create _one spesific task_ and run all **I2C** things inside it. The downside is you need to _query_ all the **I2C** communication. But that is the thing, there will be no two or more simultaneous access of **I2C** driver, and your DS3231 time will be fine. 

As for **Mutual Exclusion (MutEx)**, i believe you need to modify the `TwoWire` library, especially in `write()` and `read()` section. Let me know if my understanding is not correct.

tsctrl, 2021-02-12:
Hi @wachidsusilo, thank you for your reply. 

i was referring to this site and able to solve the issue: [https://gitter.im/espressif/arduino-esp32?at=5df1409fe47fb31eb795c42e](https://gitter.im/espressif/arduino-esp32?at=5df1409fe47fb31eb795c42e)

i am able to replicate the 165:165:165 issue as you mention as per multi thread during update request. basically using semaphores and mutex solve my issue. the time update and read call are in different thread but run in the same core and both was control by the same mutex. so far i have brute force the task by sending 2000 character json without any issue and the time was updated as expected.

solved and closing this issue. thank you.

closed 2021-02-12 by tsctrl