espressif/arduino-esp32 #4793: Enable / configure / modify serial logging

celesteking, 2021-02-10:
### Hardware:
Board:							Doit ESP32 Devkit-1
Core Installation version:			 how would I know?
IDE name:						Arduino IDE latest
Flash Frequency:					80
PSRAM enabled:                                    how would I know?
Upload Speed:					115200
Computer OS:                                       osx

### Description:
I'm looking for a way to enable serial logging. Currently, only boot message is being output, but I'd like to see more. I've selected "Debug" core level in arduino ide, but it made 0 difference.

I'm using ledc to set frequency / resolution combo that isn't supported, but I'm not seeing proper log message that was supposed to be logged, according to API & tech reference docs:
```
E (196) ledc: requested frequency and duty resolution cannot be achieved, try reducing freq_hz or duty_resolution. div_param=128`
```

### Sketch:
```cpp
ledcSetup(0, 10000, 16);
```

### Debug Messages:
```

02:13:15.788 -> ets Jun  8 2016 00:22:57
02:13:15.788 ->
02:13:15.788 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
02:13:15.788 -> configsip: 0, SPIWP:0xee
02:13:15.788 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
02:13:15.826 -> mode:DIO, clock div:1
02:13:15.826 -> load:0x3fff0018,len:4
02:13:15.826 -> load:0x3fff001c,len:1044
02:13:15.826 -> load:0x40078000,len:8896
02:13:15.826 -> load:0x40080400,len:5816
02:13:15.826 -> entry 0x400806ac
02:13:15.931 -> [W][ledjerk.ino:15] setup(): Yahoooo !!!!!
```

lbernstone, 2021-02-10:
ledcSetup returns the assigned frequency.
`Serial.printf("Channel: %d   Frequency: %lu\\n", channel, ledcSetup(channel, 10000, 11));`

celesteking, 2021-02-10:
> The LEDC API is designed to report an error when trying to set a frequency and a duty resolution that exceed the range of LEDC’s hardware. For example, an attempt to set the frequency to 20 MHz and the duty resolution to 3 bits will result in the following error reported on a serial monitor:
E (196) ledc: requested frequency and duty resolution cannot be achieved, try reducing freq_hz or duty_resolution. div_param=128

I have lamented of how bad docs for esp32 are in the other issue, but in this one, I'd like to trigger that warning as I'm really setting the resolution param wrong. 

```
LEDC Clock Source LEDC Output (PWM) Frequency Highest Resolution
APB_CLK (80 MHz) 10 kHz 1/8000 (13 bit)
```

I attempted to set 16 bits, which ought to be out of range.

celesteking, 2021-02-10:
And while we're at it, how do I get the SDK version I'm running within arduino ide? There's no way for me as end user to know whether I'm running the latest version.

closed 2021-07-08 by stale[bot]