espressif/arduino-esp32 #4794: Sad state of user documentation for ESP32 Arduino AKA the Lack Thereof

celesteking, 2021-02-10:
I'm a newbie looking to get started with esp32 arduino. I have some experience with Arduino board and their starter kit. I'm very satisfied with the sheer amount of explanations and guides available for Arduino-based products.

Now I come to esp32. When googling, this repo is the first link. I go to this repo, I see the readme, I install via board manager. What next? There's nothing else. I'm left alone. I know there are some pins, some ports, something else provided by my esp32 board manufacturer, but where the actual arduino-like reference is? The functions provided by the framework, the guides on how to use them?

Where's [the port reference](https://www.arduino.cc/en/Reference/PortManipulation) for esp32? What are some useful macros/defines I should use, most common, most used?
Where's [the functions reference ](https://www.arduino.cc/reference/en/language/functions/analog-io/analogwrite/)for esp32?
Where are [the tutorials on how to use the functionality](https://www.arduino.cc/en/Tutorial/Foundations/PWM) provided by esp32?

Where are described all those things that I see when I go through Menu -> Examples -> ESP32 -> * ? Some examples have a bit of comments though, some have none at all.

I'm trying to use the `ledc` functionality, the PWM that was so easy with arduino. It's far more advanced in esp32, which is great, but for dumb people, where the heck is the documentation? What do I need to write and what component is responsible for what action? How do I set the frequency, the pattern, the params. I see other people struggling with this, asking the questions in this repo and get referred to header files. Do you really think this is a substitute for documentation ? -

```c
double      ledcSetup(uint8_t channel, double freq, uint8_t resolution_bits);
void        ledcWrite(uint8_t channel, uint32_t duty);
double      ledcWriteNote(uint8_t channel, note_t note, uint8_t octave);
uint32_t    ledcRead(uint8_t channel);
double      ledcReadFreq(uint8_t channel);
void        ledcAttachPin(uint8_t pin, uint8_t channel);
void        ledcDetachPin(uint8_t pin);
```
[A header file](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.h#L30) with zero useful comments?

How on earth am I supposed to understand this? Is this some kind of magic ball game where you make guesses by writing random code supplying random parameters to these functions and see if it works?

Next thing I tried is sending some infrared data. You've got a special RMT thingie for it. That's awesome. What is not awesome is zero documentation. What function shall I call? What's the workflow, what's the procedure? There's [a header file](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-rmt.h) with tons of functions, all are documented with one-line comment a-la `rmtInit() - "init RMT". rmtLoop() - "loop the RMT"`.  AWESOME doc, greatly appreciated, helpful much.

But really guys, this should be fixed or a warning put into readme - "NO BEGINNERS". I'm not a guy who's paid $200k/year to develop microcontroller code who would jump straight onto your ESP-IDF or whatever that dev env is called and who knows everything about interrupts, registers, C types, bits and bytes, who just goes to API/Tech reference manual and SEEs things and knows how to implement them. I'm just a hobbyist, otherwise I would've skipped arduino shim altogether without blinking an eye.

If you've decided to support Arduino IDE, make it so that the Arduino IDE target audience, which constitutes mostly DUMB people like me, are actually able to use it. Write docs for this goddamn library, it's been out for over 2 years already. Thanks.

celesteking, 2021-02-10:
So Raspberry Pi recently released their newest thingie, which is Pico model for $4. I'm not going to fall into FUD war of hardware comparison, but I'd like to note their *brilliantly* crafted documentation. Just *LOOK* at this: https://www.raspberrypi.org/documentation/pico/getting-started/ .

i just can't

celesteking, 2021-02-10:
It also makes sense to open Github Discussions tab, which is basically a stackoverflow/Q&A replica, but within github.

everslick, 2021-02-17:
It is up to US to improve what WE find lacking. Who else should do all the work? But I support your comment of opening the Discussions. The issues are misused for that all the time. @me-no-dev  ?

chegewara, 2021-02-17:
> So Raspberry Pi recently released their newest thingie, which is Pico model for $4. I'm not going to fall into FUD war of hardware comparison, but I'd like to note their **brilliantly** crafted [...]

Feel free to change MCU and use rpi pico instead of esp32. Oh wait, there is no arduino support on it, so there is no documentation either.
And here is official documentation for official espressif esp-idf API:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/index.html

lbernstone, 2021-02-17:
I can't see how Discussions would be any different.  It would just be another place for people to complain about how their free software doesn't do what they want, and the actual participants will avoid it like the plague.

everslick, 2021-02-17:
> I can't see how Discussions would be any different. It would just be another place for people to complain about how their free software doesn't do what they want, and the actual participants will avoid it like the plague.

Then we should close the issue tracker altogether, it's lost its purpose of tracking actual CORE bugs and features.

me-no-dev, 2021-02-18:
@everslick been a bit grumpy lately? :D no worries we are assembling a team to handle documentation and other such issues ;)

everslick, 2021-02-18:
> @everslick been a bit grumpy lately? :D

Wait, until I call everyone stupid and ugly! ;)

> no worries we are assembling a team to handle documentation and other such issues ;)

Great to hear! 👍

celesteking, 2021-02-22:
My original message was meant for someone at higher level, the decision makers. Not sure if they're reading the issues, but let's try ringing some bells to get the disco going. 

As regarding the raspberry pico $4, the controller has been released < 1 mo ago, and, lo and behold, just in time, [Arduino announced ](https://www.seeedstudio.com/blog/2021/01/29/arduino-ide-support-announced-for-the-raspberry-pi-pico/) they're going to use that silicon to make an arduino board with full Arduino IDE support, and along with that bring Arduino IDE support to basically ALL RP2040 based devices.

> It is up to US to improve what WE find lacking. Who else should do all the work?

Who else? The developers who work for an [$11bln market cap](https://www.google.com/finance/quote/688018:SHA?sfr=fpt) commercial company. 

> Feel free to change MCU

I should've done that and not followed the Espressif Justice Warriors who run worldwide around Internet bragging how cool their chip is and how cheap it is and how good it is for Arduino-based projects. I should've followed your suggestion. With Raspberry Pi entering your niche, it's going to be a hot battle and I'm already sighting the winner. Thanks and sorry to disappoint.

me-no-dev, 2021-02-22:
I am high level enough @celesteking to tell you that we are currently assembling a team to deal with documentation and more. Up until now from Espressif side has been just me and that is not my only task. So the community has played main role into improving this project. The result of all that is the sad state of documentation, but as I said, this is about to improve.

More productive thing you can be to list things that you are interested in being documented. Classes, function names, parameter names and such are generally very clear on what they do or what they require. There are plenty of examples here and on the web also. It requires reading headers and sketches to see what else each library can do, but documentation would not be much different than that. If you think about it, Arduino APIs follow a very common structure and are more or less predictable.

So please tell us what are the most important things for you to document?

closed 2021-07-09 by stale[bot]