espressif/arduino-esp32 #4795: UART serial freezing while comunicate with SIM800L GPRS Mode

tarcisiohlima-iponia, 2021-02-10:
Hello everyone!

I am trying to keep my ESP32 connected with a server by using a GPRS modem SIM800l. 

I can connect with the server, and everything works fine, until the communication with the modem suddenly stops.
For this communication, I am using using UART on GPIOs 16 and 17. It seems random: sometimes, it can comunicate for hours until stops, sometimes, it will take a few minutes to crash again. 

It seems like the problem it not at the modem. I already hard reseted the modem, but this is not working. I also tried a watchdog to reset the ESP32 after 3 minutes of communication failure, and it resets (and the modem resets too), but the communication doesn't back. It seems like serial communication is freezed, even with the ESP.restart(). The only way to start the communication again, is by hard reseting the ESP32 (Turn off, turn on the energy). 

I saw the ESP32 doesn't reset the UART after ESP.restart(), is that right?

How could I fully reset the ESP32, or at least reset something that can start the communication with the SIM800l again?

The only solution I found is from this post: https://github.com/espressif/arduino-esp32/issues/1584. But this post is from 2018, I thought that using Arduino IDE 1.8.13 with updated ESP32 library would solve the problem. (I added this at board manager including this link: https://dl.espressif.com/dl/package_esp32_index.json, then installing the ESP32 library directly from Arduino IDE).

Now I reinstalled the Arduino IDE, and installed the ESP32 packed following this tutorial: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

The code is running okey at the moment, but, as I said, usually runs perfect for a while, and then, it crashes. Chaging the installation mode can make any difference?

Thanks in advance!

Arroak, 2021-06-19:
Hi, i am also facing a similar issue over here. I have five devices using esp32 and sim800l, I used the sim800l for mqtt, but out of five devices of the same architecture, only three sends data; one sends constantly, the other two are not  consistent and the remaining two are not connecting at all. after searching for the same issue online i got a fix that worked, to change the network band of the sim8ool to dcs_mode using AT+CBAND ="DCS_MODE".  this seems to work but i still have issue getting other ones to publish data. still researching on other means of making it scalable.

VojtechBartoska, 2022-03-03:
@SuGlider Can you please test this issue? It was closed by Stale bot but I'm not sure if it's still valid. Thanks!

SuGlider, 2022-03-03:
> Hi, i am also facing a similar issue over here. I have five devices using esp32 and sim800l, I used the sim800l for mqtt, but out of five devices of the same architecture, only three sends data; one [...]

@Arroak 

Reading this post, it sounds like the issue is related to the GRPS network.
Can you confirm it?

SuGlider, 2022-03-03:
@tarcisiohlima-iponia 
Arduino Serial (UART) has been completely refactored. Please try again using latest Arduino Core version. If any issue is found, please report back here.

Thanks.

SuGlider, 2022-03-20:
Closing it. 
In few time Arduino Core version 2,0.3 will be out and it fixes a big number of issues related to UART.
Please use the newest version and reopen this issue (or open a new one) whenever necessary.

closed 2022-03-20 by SuGlider

lbernstone, 2022-03-20:
@SuGlider Maybe put a note on the readme for BluetoothSerial that bluetooth classic is only available on ESP32.