espressif/arduino-esp32 #4796: I2C and WiFi won't work at the same time on ESP32-S2

makermelissa, 2021-02-10:
I'm getting a similar situation. If I use I2C on some other pins (like 33 and 34), I2C works fine, but WiFi does not connect. 

If I try pins 21 and 22, WiFi appears to work, but I get a Core 0 panic:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4009f7c6  PS      : 0x00060a30  A0      : 0x8009ced0  A1      : 0x3ffd20c0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x3ffc7a7a  A5      : 0x00000000  
A6      : 0x3ffd74ec  A7      : 0x00000032  A8      : 0x8009f7c3  A9      : 0x3ffd2080  
A10     : 0x00000000  A11     : 0x3ffc7a7a  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000021  SAR     : 0x00000016  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000001  LCOUNT  : 0x400278c5  

Backtrace:0x4009f7c3:0x3ffd20c0 0x4009cecd:0x3ffd20e0 0x4009cf41:0x3ffd2100 0x40088c1c:0x3ffd2120 0x4008fe81:0x3ffd2140 0x4009218a:0x3ffd2170 0x400a04dd:0x3ffd2190

ELF file SHA256: 0000000000000000

Rebooting...
```

If I don't use I2C at all, WiFi has no problems.

This is on the ESP32-S2 using the idf-release/v4.2 branch. I had previously mentioned this is #4760, but realized that was for the ESP32.

Tagging @ladyada.

makermelissa, 2021-02-11:
I noticed this issue when attempting to add support to ESPHome. In order to reproduce this issue with the ESP32 Saola, you can do the following. Sorry it's so long, but I've been working on this for over a week.
1. Git clone the ESPHome Dev branch: https://github.com/esphome/esphome.git
2. Apply PR https://github.com/esphome/esphome/pull/1433 so WiFi compiles.
3. Add the following board config esphome/pins.py:
```
    'esp32-s2-saola-1': {
        'TX': 43, 'RX': 44, 'SDA': 8, 'SCL': 9, 'SS': 34, 'MOSI': 35,
        'MISO': 37, 'SCK': 36, 'A0': 1, 'A1': 2, 'A2': 3, 'A3': 4, 'A4': 5,
        'A5': 6, 'A6': 7, 'A7': 8, 'A8': 9, 'A9': 10, 'A10': 11, 'A11': 12,
        'A12': 13, 'A13': 14, 'A14': 15, 'A15': 16, 'A16': 17, 'A17': 18,
        'A18': 19, 'A19': 20, 'T1': 1, 'T2': 2, 'T3': 3, 'T4': 4, 'T5': 5,
        'T6': 6, 'T7': 7, 'T8': 8, 'T9': 9, 'T10': 10, 'T11': 11, 'T12': 12,
        'T13': 13, 'T14': 14, 'DAC1': 17, 'DAC2': 18
    },
```
3. Use the following YAML file for your config. Save it in test.yaml:
```
esphome:
  name: esp32s2test
  platform: ESP32
  board: esp32-s2-saola-1
  platformio_options:
    platform: espressif32 @ https://github.com/adafruit/platform-espressif32.git
    platform_packages: framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2

wifi:
  ssid: "YouWiFi"
  password: "YourPassword"
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp32-S2 Test Fallback Hotspot"
    password: "test"

captive_portal:

# Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:
  password: "password"

ota:
  password: "pwasword"

# I2C Bus
i2c:
  sda: SDA
  scl: SCL
```
4. Inside the repo run `pip3 install -e .` which will install any packages
5. Run `esphome test.yaml run` which will install the packages and attempt to program the board. You don't need to actually install yet.
6. Go into your platformio packages (mine were installed to `~/.platformio/packages`)
7. Apply PR #4776 to this repo which should be inside of the framework-arduinoespressif32 folder. this fixes an I2C initialization error.
8. Run `esphome test.yaml run` again and program your board.
9. You may need to connect a USB to UART cable to the ground, TX, and RX pins to see debug output

ladyada, 2021-06-20:
@makermelissa wanna retry with latest `main` with/without @dhalbert's mutex fix ?

ladyada, 2021-08-21:
to the best of our knowledge, this has not been fixed yet

closed 2021-09-06 by stale[bot]

Matssa56, 2022-01-28:
Hey guys I'm having the same issue. I'm trying to use I2C on my devkit board using GPIO 22 for SCL and GPIO21 for SDA and as soon as I have the I2C configuration in my yaml the wifi stops working... Anyone has a fix?
Thanks!

Matssa56, 2022-01-29:
Ok so tested with a higher PSU amps and it works now, apparently 5W isn't enough (5V / 1A).