espressif/arduino-esp32 #4797: WiFi.macAddress provides incorrect MAC Address (Off by one).

etherfi, 2021-02-11:
### Hardware:
Board:							ESP-WROOM-32 Dev Board (FCC ID: 2AC7Z-ESPWROOM32)
Core Installation version:			1.0.5-rc6
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                     N/A
Upload Speed:					115200
Computer OS:                                        Windows 10

### Description:
WiFi.macAddress provides incorrect MAC Address (off by one).

I get ```24:62:ab:ff:bd:a8``` but ARP on my local network reports ```24:62:ab:ff:bd:a9```. What am I doing wrong?

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
#include "WiFi.h"

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  byte mac[6];
  WiFi.macAddress(mac);
  Serial.printf("This device's MAC address is: %x:%x:%x:%x:%x:%x", mac[0], mac[1], mac[2], mac[3], mac[4], mac[5]);
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
N/A
```

me-no-dev, 2021-02-18:
I can not reproduce this:

[image]

etherfi, 2021-02-26:
The problem only happens when connecting to the ESP32's SoftAP as a station. You are testing the ESP32 as a station to a normal WiFi network.

etherfi, 2021-02-26:
I assumed it would happen both times... But it occurs for me after entering SoftAP mode.

chegewara, 2021-02-26:
https://docs.espressif.com/projects/esp-idf/en/v3.1.7/api-reference/system/base_mac_address.html#number-of-universally-administered-mac-address

etherfi, 2021-02-26:
Thanks!!! I knew it must be something simple.

closed 2021-02-26 by etherfi