espressif/arduino-esp32 #4801: Unable to get a wifi connection,  NO_AP_FOUND error

QuitButton, 2021-02-11:
Wifi rarely connects, 99% of the time it fails.

[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 1 - SCAN_DONE
[I][WiFiMulti.cpp:114] run(): [WIFI] scan done
[I][WiFiMulti.cpp:119] run(): [WIFI] 13 networks found
[D][WiFiMulti.cpp:149] run():  --->   0: [2][E4:F4:C6:B6:7C:28] Another Silly SSID Name (-64) *
[D][WiFiMulti.cpp:151] run():        1: [2][E4:F4:C6:B6:7C:29] Do Re Mi Fa So La Wi Fi (-65) *
[D][WiFiMulti.cpp:151] run():        2: [11][20:B0:01:CA:D2:A9] vodafoneCAD2A9 (-85) *
[D][WiFiMulti.cpp:151] run():        3: [6][18:35:D1:62:D4:E1] VM7772550 (-88) *
[D][WiFiMulti.cpp:151] run():        4: [7][B0:95:75:FE:77:68] Post_Office_FC65_EXT (-88) *
[D][WiFiMulti.cpp:151] run():        5: [6][2A:35:D1:62:D4:E1] Virgin Media (-89) *
[D][WiFiMulti.cpp:151] run():        6: [11][90:02:18:C5:BA:D6] SKY0BD88 (-90) *
[D][WiFiMulti.cpp:151] run():        7: [6][D2:05:C2:3F:B0:71] Virgin Media (-91) *
[D][WiFiMulti.cpp:151] run():        8: [6][C0:05:C2:3F:B0:71] VM9989372 (-91) *
[D][WiFiMulti.cpp:151] run():        9: [6][4C:38:D8:07:56:C1] VM2990854 (-92) *
[D][WiFiMulti.cpp:151] run():        10: [6][78:65:59:3C:1D:2C] TALKTALK3C1D2F (-93) *
[D][WiFiMulti.cpp:151] run():        11: [11][52:0D:10:41:13:C1] Virgin Media (-94) *
[D][WiFiMulti.cpp:151] run():        12: [11][40:0D:10:41:13:C1] VM7202186 (-96) *
[I][WiFiMulti.cpp:160] run(): [WIFI] Connecting BSSID: E4:F4:C6:B6:7C:28 SSID: Another Silly SSID Name Channel: 2 (-64)
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 201 - NO_AP_FOUND
[E][WiFiMulti.cpp:181] run(): [WIFI] Connecting Failed AP not found.
1 WiFi not connected
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:390] _eventCallback(): Reason: 15 - 4WAY_HANDSHAKE_TIMEOUT

chegewara, 2021-02-11:
Maybe spaces in name or long name?

QuitButton, 2021-02-11:
Why would spaces or a long name be a problem?

QuitButton, 2021-02-11:
Occasionally I get this, where it seems my AP has actually disappeared a few seconds after a connection attempt:

chegewara, 2021-02-11:
> Why would spaces or a long name be a problem?

Its just guessing, because i saw few issues related to it.

Another thing may be weak signal, since esp32 does not see SSID from time to time.

closed 2021-07-08 by stale[bot]