espressif/arduino-esp32 #4807: ESP32 keep rebooting when trying to use rosserial

akmakaiau, 2021-02-15:
### Hardware:
Board:							DOIT ESP32 DEVKIT V1
IDE name:						Arduino IDE 1.8.12
Flash Frequency:					80 MHz (I have also tried 40MHz)
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                         Ubuntu 18.04

### Description:
The ESP32 continuously reboots when I upload the example HelloWorld sketch from the ros_lib library. This also happens with the Blink (subscriber) example. All solutions outlined in [this](https://github.com/espressif/arduino-esp32/issues/2050) thread has not worked for me.

The board works for other sketches that do not use ROS.

I apologize of the decoder stack results are unclear, the decoder is not working properly for me right now and all I had was this screenshot. 

### Sketch:
```cpp
/*
 * rosserial Publisher Example
 * Prints "hello world!"
 */

#include <ros.h>
#include <std_msgs/String.h>

ros::NodeHandle  nh;

std_msgs::String str_msg;
ros::Publisher chatter("chatter", &str_msg);

char hello[13] = "hello world!";

void setup()
{
  nh.initNode();
  nh.advertise(chatter);
}

void loop()
{
  str_msg.data = hello;
  chatter.publish( &str_msg );
  nh.spinOnce();
  delay(1000);
}
```

### Serial Monitor Output:
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1442 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40087c45 on core 1

Backtrace: 0x4008b734:0x3ffb1d80 0x4008b965:0x3ffb1da0 0x40087c45:0x3ffb1dc0 0x400f93e2:0x3ffb1e00 0x400f95c6:0x3ffb1e20 0x400eba4c:0x3ffb1e40 0x400ebab5:0x3ffb1e60 0x400e8c52:0x3ffb1e80 0x400e7adc:0x3ffb1ea0 0x400eb2d0:0x3ffb1ee0 0x400d254e:0x3ffb1f00 0x400d1f46:0x3ffb1f50 0x400d1a87:0x3ffb1f80 0x400d2e07:0x3ffb1fb0 0x40087f19:0x3ffb1fd0

```
### Decoding Stack Results:
[image]
```

LorenzF, 2021-02-19:
I was having the same problem, after some digging in the source code I found out that ROS opts to use a wifi connection over the serial connection if you program an ESP32 .
If you want to circumvent this you can manually change the ros.h file in your library folder (for me this was ~/Arduino/libraries/ros_lib/ros.h)

I changed:
```
#if defined(ESP8266) or defined(ESP32) or defined(ROSSERIAL_ARDUINO_TCP)
  #include "ArduinoTcpHardware.h"
#else
  #include "ArduinoHardware.h"
#endif
```

Into:
```
#if defined(ROSSERIAL_ARDUINO_TCP)
  #include "ArduinoTcpHardware.h"
#else
  #include "ArduinoHardware.h"
#endif
```

This meant regular arduino hardware config was loaded instead of the Wifi version.
Also, standard a baud rate of 57600 is used, that was my second mistake!

Good luck!

akmakaiau, 2021-02-23:
@LorenzF Thank you that worked for me! Appreciate the help!!

noshluk2, 2022-01-15:
> I was having the same problem, after some digging in the source code I found out that ROS opts to use a wifi connection over the serial connection if you program an ESP32 . If you want to circumvent [...]

For me your solution + wifi connectivity delay was the solution 
'''
void setup()
{ 
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED)
  { delay(500);
    Serial.print(".");
  }
  Serial.println("WiFi connected - IP address:  ");
  Serial.println(WiFi.localIP());
  nh.getHardware()->setConnection(server, serverPort);
  nh.initNode();
  nh.advertise(ir_values_node);
  pinMode(IR_pin,INPUT);
}
'''

Order of the code lines is important

mattwilliamson, 2023-04-10:
Came here you point out this works for me:
```c
#undef ESP32
#include <ros.h>
#define ESP32
```