espressif/arduino-esp32 #4808: Full API Documentation for Arduino core for the ESP32

beamholder, 2021-02-15:
I'm trying to find a complete set of API for Arduino documentation for the esp32 core. Specifically for functions not documented in the standard [Arduino language reference.](https://www.arduino.cc/reference/en/)

For example, the Espressif version of the Arduino WiFi library is a superset of the standard WiFi library; including functions like:
WiFi.onEvent()
WiFi.mode()
WiFi.setSleep()

as well as other type and class extensions.

Some of these functions are used in the examples provided by Espressif, which is great but that seems to be a very inefficient way of discovering the Espressif-specific extensions to the standard Arduino WiFi library. Not to mention an awkward way to find that information in a single place in the future.

I see that these API's are documented if one were using the [essprissif ESP-IDF environment](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#), but how does one translate that to arduino-esp32 WiFi library calls?

I am somewhat new to the whole Arduino ecosystem so maybe I'm missing something for boards not manufactured by Arduino?

For reference I'm working on a project using M5Stack ATOM Matrix and M5Stack C Plus units (both are ESP32 based devices).

Any pointers  would be very appreciated.

Many thanks.

lbernstone, 2021-08-23:
WIP. https://docs.espressif.com/projects/arduino-esp32/en/latest/

pedrominatel, 2022-05-30:
Hi @beamholder,

As @lbernstone mentioned, the documentation is in progress and the WiFi library is already done.
You can track the progress [here](https://docs.espressif.com/projects/arduino-esp32/en/latest/libraries.html).

If you have any suggestions or contributions, you can start a discussion [here](https://github.com/espressif/arduino-esp32/discussions).

closed 2022-05-30 by pedrominatel