espressif/arduino-esp32 #4809: Suggestion: Align arduino-esp32 with esp-idf docker images

laolux, 2021-02-15:
**Enhancement Suggestion**
This is a suggestion to make it easier to use arduino-esp32 in conjunction with the official esp-idf.
The esp-idf basically comes in two versions:
The regular version for which you clone the esp-idf git repository and install everything on your computer. This requires a bunch of prerequisites to be installed, but gives you the freedom to choose whatever version of the esp-idf by checking out the corresponding commit in the git repository.
An alternative and [officially supported](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-guides/tools/idf-docker-image.html) version is to use the pre-assembled [docker images](https://hub.docker.com/r/espressif/idf). These come in a variety of versions to choose from and bring with them all dependencies conveniently bundled in one place. Only dependency is some container runtime such as docker or podman. The downside is that one cannot freely choose which exact git commit to use from the esp-idf when using the docker images. Therefore it is not possible to choose the exact esp-idf version that arduino-esp32 was built with.
Hence, I suggest to make available special versions of arduino-esp32 which have been built with the exact esp-idf version as used in the docker images.
For example, the docker image tagged v4.2 (02d3fe953fa2) is build with esp-idf commit c40f2590bf759ff60ef122afa79b4ec04e7633d2 and the docker image tagged release-v4.2 (2a21f93247f0) is build with esp-idf commit 8cd16b60f56dec4f319f6d2e73b6dec794588312. It would be great to have arduino-esp32 versions which work with the docker images out-of-the-box.

lbernstone, 2021-02-15:
esp-idf is a moving target, so arduino-esp32 needs to be continually adjusted to fit it.  If there were a team of developers working to match it daily, this would not be a problem.  However, there is one developer working on this in his spare time, which often leaves months worth of changes to be merged.
[lib-builder](https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md) is what is used by the repo's CI.
I will keep very tight containers based on release versions in my [dockerhub](https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder).  These should be ready for minor config changes or integrating your own arduino/idf code.  The dockerfiles for them are in my [lib-builder clone](https://github.com/lbernstone/esp32-arduino-lib-builder/tree/docker/docker).

laolux, 2021-02-17:
Yes, I am well aware that esp-idf is advancing quickly and adjusting arduino-esp32 takes some work. If it would be trivial, then I might just do it myself. I also appreciate that others spend their time on this so that I can use it. Hence a suggestion and not a demand.
To make it easier for other users, maybe trying to build arduino-esp32 against tagged releases of esp-idf would help? Those don't seem to happen too often. And it could help with #4643, because then it would be very easy to reference which esp-idf version was used (as opposed to using only some git commit hash).
Thanks for pointing me to lib-builder and your docker versions. I will start playing with that, maybe this will help me to use arduino-esp32 as in idf component with esp-idf v4.2.

Jason2866, 2021-02-17:
@laolux  i use a slightly modified fork of the lib builder to build Arduino for the ESP32-solo1
https://github.com/Jason2866/esp32-arduino-lib-builder

closed 2021-07-08 by stale[bot]