espressif/arduino-esp32 #4815: ESP8266 is much faster than ESP32 with pubsubclient

sblantipodi, 2021-02-16:
### Hardware:
Board:							Lolin ESP32 vs Lolin ESP8266 D1 Mini
Core Installation version:			latest
IDE name:						latest
Flash Frequency:					80MHz
PSRAM enabled:                                    no
Upload Speed:				     500000
Computer OS:                                       Windows 10

### Description:
Why ESP8266 is faster than ESP32 with the best MQTT library available for Arduino?

With ESP8266@160MHz I can send 90 message per second using pubsubclient,
same code on ESP32@240MHz process only 40 message per second.

ESP32 is supposed to be much faster, what's happening?

Thank you!

chegewara, 2021-02-16:
Why do you think esp32 should be faster?

https://hackaday.com/wp-content/uploads/2021/02/coremark-1_reformatted.png

Remember that mqtt is running on 1 core only, so compare 1 core.

bertmelis, 2021-02-16:
Try async-mqtt-client. It's build on (ESP)AsyncTCP and might be even faster on both platforms.

(for now, use the develop branch)

sblantipodi, 2021-02-16:
> Why do you think esp32 should be faster?
> 
> https://hackaday.com/wp-content/uploads/2021/02/coremark-1_reformatted.png
> 
> Remember that mqtt is running on 1 core only, so compare 1 core.

using ESP8266 @160MHz
and ESP32 @ 240MHz, even that graph shows that ESP32 is faster at max frequency,

sblantipodi, 2021-02-16:
> Try async-mqtt-client. It's build on (ESP)AsyncTCP and might be even faster on both platforms.
> 
> (for now, use the develop branch)
> 
> And as said before: you cannot just compare the two platforms. And you might as well have different network throughput.

last release has been made in 2017, I prefer a supported lib over an abandoned one.

bertmelis, 2021-02-16:
> last release has been made in 2017, I prefer a supported lib over an abandoned one.

That's why you need to use the develop branch.

sblantipodi, 2021-02-17:
> > last release has been made in 2017, I prefer a supported lib over an abandoned one.
> 
> That's why you need to use the develop branch.

this does not answer my issue but thanks :)

bertmelis, 2021-02-17:
I also have experienced slow outgoing TCP data on the ESP32 whereas "the same" code on ESP8266 was faster. Almost all the time it was hardware (antenna, antenna orientation...).

everslick, 2021-02-17:
The ESP8266 gets really great optimizations all the time. This is not to be denied. That's just a matter of manpower - more people working on it. I hope over time the ESP8266 core classes will eventually find their way to the ESP32 as well. ESP32 could really use some more people. @me-no-dev is doing great work, but he is more or less alone.

closed 2021-07-08 by stale[bot]