espressif/arduino-esp32 #4818: MBEDTLS with External PSRAM is not working.

akshar001, 2021-02-17:
Using Arduino Master branch.

Problem is simple.

We are trying to call one simple HTTPS api,

We are using to build custom lib.

We are enabling PSRAM in menuconfig and choosing MBEDTLS in EXTERNAL RAM.

The problem is when we enable that in components it's getting crashed.

When we use internal ram it works fine.

Steps to reproduce the issue at your side.
Use the sdkconfig fille that i am sharing right now,
compile the libs, copy paste the out/sdk/ libs to current arduino folder and run this code.

I am posting three different things.

sdkconfig,
.ino example file.
and if you do not want to go through all of the custom building,
The zip of our compiled out folder.

I have tried so many times using different versions and i firmly believe the issue is at esp side just need to figure out what it is.

ESP32 chip version. 

ESP32-D0WDQ6-ATUB00

akshar001, 2021-02-17:
@igrr your help is needed on this one please.

akshar001, 2021-02-17:
```
#include <WiFi.h>
#include <HTTPClient.h>
#include <HTTPUpdate.h>

const char* ssid = "SSID";
const char* password = "password";

const char* rootCACertificate = \
                                "-----BEGIN CERTIFICATE-----\n" \
                                "MIIFjTCCA3WgAwIBAgIRANOxciY0IzLc9AUoUSrsnGowDQYJKoZIhvcNAQELBQAw\n" \
                                "TzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2Vh\n" \
                                "cmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTYxMDA2MTU0MzU1\n" \
                                "WhcNMjExMDA2MTU0MzU1WjBKMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3Mg\n" \
                                "RW5jcnlwdDEjMCEGA1UEAxMaTGV0J3MgRW5jcnlwdCBBdXRob3JpdHkgWDMwggEi\n" \
                                "MA0GCSqGSIb3DQEBAQUAA4IBDwAwggEKAoIBAQCc0wzwWuUuR7dyXTeDs2hjMOrX\n" \
                                "NSYZJeG9vjXxcJIvt7hLQQWrqZ41CFjssSrEaIcLo+N15Obzp2JxunmBYB/XkZqf\n" \
                                "89B4Z3HIaQ6Vkc/+5pnpYDxIzH7KTXcSJJ1HG1rrueweNwAcnKx7pwXqzkrrvUHl\n" \
                                "Npi5y/1tPJZo3yMqQpAMhnRnyH+lmrhSYRQTP2XpgofL2/oOVvaGifOFP5eGr7Dc\n" \
                                "Gu9rDZUWfcQroGWymQQ2dYBrrErzG5BJeC+ilk8qICUpBMZ0wNAxzY8xOJUWuqgz\n" \
                                "uEPxsR/DMH+ieTETPS02+OP88jNquTkxxa/EjQ0dZBYzqvqEKbbUC8DYfcOTAgMB\n" \
                                "AAGjggFnMIIBYzAOBgNVHQ8BAf8EBAMCAYYwEgYDVR0TAQH/BAgwBgEB/wIBADBU\n" \
                                "BgNVHSAETTBLMAgGBmeBDAECATA/BgsrBgEEAYLfEwEBATAwMC4GCCsGAQUFBwIB\n" \
                                "FiJodHRwOi8vY3BzLnJvb3QteDEubGV0c2VuY3J5cHQub3JnMB0GA1UdDgQWBBSo\n" \
                                "SmpjBH3duubRObemRWXv86jsoTAzBgNVHR8ELDAqMCigJqAkhiJodHRwOi8vY3Js\n" \
                                "LnJvb3QteDEubGV0c2VuY3J5cHQub3JnMHIGCCsGAQUFBwEBBGYwZDAwBggrBgEF\n" \
                                "BQcwAYYkaHR0cDovL29jc3Aucm9vdC14MS5sZXRzZW5jcnlwdC5vcmcvMDAGCCsG\n" \
                                "AQUFBzAChiRodHRwOi8vY2VydC5yb290LXgxLmxldHNlbmNyeXB0Lm9yZy8wHwYD\n" \
                                "VR0jBBgwFoAUebRZ5nu25eQBc4AIiMgaWPbpm24wDQYJKoZIhvcNAQELBQADggIB\n" \
                                "ABnPdSA0LTqmRf/Q1eaM2jLonG4bQdEnqOJQ8nCqxOeTRrToEKtwT++36gTSlBGx\n" \
                                "A/5dut82jJQ2jxN8RI8L9QFXrWi4xXnA2EqA10yjHiR6H9cj6MFiOnb5In1eWsRM\n" \
                                "UM2v3e9tNsCAgBukPHAg1lQh07rvFKm/Bz9BCjaxorALINUfZ9DD64j2igLIxle2\n" \
                                "DPxW8dI/F2loHMjXZjqG8RkqZUdoxtID5+90FgsGIfkMpqgRS05f4zPbCEHqCXl1\n" \
                                "eO5HyELTgcVlLXXQDgAWnRzut1hFJeczY1tjQQno6f6s+nMydLN26WuU4s3UYvOu\n" \
                                "OsUxRlJu7TSRHqDC3lSE5XggVkzdaPkuKGQbGpny+01/47hfXXNB7HntWNZ6N2Vw\n" \
                                "p7G6OfY+YQrZwIaQmhrIqJZuigsrbe3W+gdn5ykE9+Ky0VgVUsfxo52mwFYs1JKY\n" \
                                "2PGDuWx8M6DlS6qQkvHaRUo0FMd8TsSlbF0/v965qGFKhSDeQoMpYnwcmQilRh/0\n" \
                                "ayLThlHLN81gSkJjVrPI0Y8xCVPB4twb1PFUd2fPM3sA1tJ83sZ5v8vgFv2yofKR\n" \
                                "PB0t6JzUA81mSqM3kxl5e+IZwhYAyO0OTg3/fs8HqGTNKd9BqoUwSRBzp06JMg5b\n" \
                                "rUCGwbCUDI0mxadJ3Bz4WxR6fyNpBK2yAinWEsikxqEt\n" \
                                "-----END CERTIFICATE-----\n"\
                                "-----BEGIN CERTIFICATE-----\n"\
                                "MIICxjCCAk2gAwIBAgIRALO93/inhFu86QOgQTWzSkUwCgYIKoZIzj0EAwMwTzEL\n"\
                                "MAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2VhcmNo\n"\
                                "IEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDIwHhcNMjAwOTA0MDAwMDAwWhcN\n"\
                                "MjUwOTE1MTYwMDAwWjAyMQswCQYDVQQGEwJVUzEWMBQGA1UEChMNTGV0J3MgRW5j\n"\
                                "cnlwdDELMAkGA1UEAxMCRTEwdjAQBgcqhkjOPQIBBgUrgQQAIgNiAAQkXC2iKv0c\n"\
                                "S6Zdl3MnMayyoGli72XoprDwrEuf/xwLcA/TmC9N/A8AmzfwdAVXMpcuBe8qQyWj\n"\
                                "+240JxP2T35p0wKZXuskR5LBJJvmsSGPwSSB/GjMH2m6WPUZIvd0xhajggEIMIIB\n"\
                                "BDAOBgNVHQ8BAf8EBAMCAYYwHQYDVR0lBBYwFAYIKwYBBQUHAwIGCCsGAQUFBwMB\n"\
                                "MBIGA1UdEwEB/wQIMAYBAf8CAQAwHQYDVR0OBBYEFFrz7Sv8NsI3eblSMOpUb89V\n"\
                                "yy6sMB8GA1UdIwQYMBaAFHxClq7eS0g7+pL4nozPbYupcjeVMDIGCCsGAQUFBwEB\n"\
                                "BCYwJDAiBggrBgEFBQcwAoYWaHR0cDovL3gyLmkubGVuY3Iub3JnLzAnBgNVHR8E\n"\
                                "IDAeMBygGqAYhhZodHRwOi8veDIuYy5sZW5jci5vcmcvMCIGA1UdIAQbMBkwCAYG\n"\
                                "Z4EMAQIBMA0GCysGAQQBgt8TAQEBMAoGCCqGSM49BAMDA2cAMGQCMHt01VITjWH+\n"\
                                "Dbo/AwCd89eYhNlXLr3pD5xcSAQh8suzYHKOl9YST8pE9kLJ03uGqQIwWrGxtO3q\n"\
                                "YJkgsTgDyj2gJrjubi1K9sZmHzOa25JK1fUpE8ZwYii6I4zPPS/Lgul/\n"\
                                "-----END CERTIFICATE-----\n"\
                                "-----BEGIN CERTIFICATE-----\n"\
                                "MIIFazCCA1OgAwIBAgIRAIIQz7DSQONZRGPgu2OCiwAwDQYJKoZIhvcNAQELBQAwTzELMAkGA1UEBhMCVVMxKTAnBgNVBAoTIEludGVybmV0IFNlY3VyaXR5IFJlc2VhcmNoIEdyb3VwMRUwEwYDVQQDEwxJU1JHIFJvb3QgWDEwHhcNMTUwNjA0MTEwNDM4WhcNMzUwNjA0MTEwNDM4WjBPMQswCQYDVQQGEwJVUzEpMCcGA1UEChMgSW50ZXJuZXQgU2VjdXJpdHkgUmVzZWFyY2ggR3JvdXAxFTATBgNVBAMTDElTUkcgUm9vdCBYMTCCAiIwDQYJKoZIhvcNAQEBBQADggIPADCCAgoCggIBAK3oJHP0FDfzm54rVygch [... cut ...]
                    "Host: " + "wiphone.io" + "\r\n" +
                    "Connection: close\r\n\r\n").c_str());
      unsigned long timeout = millis();
      while (client.available() == 0) {
        if (millis() - timeout > 15000) {
          log_i("Client Timeout !");
          client.stop();
          break;
        }
      }

      while (client.available()) {
        String l = client.readStringUntil('\r');
        log_i("Read: %s", l.c_str());
      }

      done_test = true;
    }
  }
  sleep(1);
}

```

akshar001, 2021-02-17:
```
#
# Automatically generated file; DO NOT EDIT.
# Espressif IoT Development Framework Configuration
#
CONFIG_IDF_TARGET="esp32"
CONFIG_IDF_FIRMWARE_CHIP_ID=0x0000

#
# SDK tool configuration
#
CONFIG_TOOLPREFIX="xtensa-esp32-elf-"
CONFIG_PYTHON="python2"
CONFIG_MAKE_WARN_UNDEFINED_VARIABLES=y

#
# Application manager
#
CONFIG_APP_COMPILE_TIME_DATE=y
CONFIG_APP_EXCLUDE_PROJECT_VER_VAR=
CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR=
CONFIG_APP_RETRIEVE_LEN_ELF_SHA=16

#
# Arduino Configuration
#
CONFIG_ENABLE_ARDUINO_DEPENDS=y
CONFIG_AUTOSTART_ARDUINO=y
CONFIG_ARDUINO_RUN_CORE0=
CONFIG_ARDUINO_RUN_CORE1=y
CONFIG_ARDUINO_RUN_NO_AFFINITY=
CONFIG_ARDUINO_RUNNING_CORE=1
CONFIG_ARDUINO_LOOP_STACK_SIZE=8192
CONFIG_ARDUINO_EVENT_RUN_CORE0=
CONFIG_ARDUINO_EVENT_RUN_CORE1=y
CONFIG_ARDUINO_EVENT_RUN_NO_AFFINITY=
CONFIG_ARDUINO_EVENT_RUNNING_CORE=1
CONFIG_ARDUINO_UDP_RUN_CORE0=
CONFIG_ARDUINO_UDP_RUN_CORE1=y
CONFIG_ARDUINO_UDP_RUN_NO_AFFINITY=
CONFIG_ARDUINO_UDP_TASK_PRIORITY=3
CONFIG_ARDUINO_UDP_RUNNING_CORE=1
CONFIG_DISABLE_HAL_LOCKS=

#
# Debug Log Configuration
#
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_NONE=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_ERROR=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_WARN=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_DEBUG=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL_VERBOSE=
CONFIG_ARDUHAL_LOG_DEFAULT_LEVEL=3
CONFIG_ARDUHAL_LOG_COLORS=y
CONFIG_ARDUHAL_ESP_LOG=y
CONFIG_ARDUHAL_PARTITION_SCHEME_DEFAULT=y
CONFIG_ARDUHAL_PARTITION_SCHEME_MINIMAL=
CONFIG_ARDUHAL_PARTITION_SCHEME_NO_OTA=
CONFIG_ARDUHAL_PARTITION_SCHEME_HUGE_APP=
CONFIG_ARDUHAL_PARTITION_SCHEME_MIN_SPIFFS=
CONFIG_ARDUHAL_PARTITION_SCHEME="default"
CONFIG_AUTOCONNECT_WIFI=
CONFIG_ARDUINO_SELECTIVE_COMPILATION=

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
[... 874 lines cut ...]
#
# Virtual file system
#
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=
CONFIG_WL_SECTOR_SIZE_4096=y
CONFIG_WL_SECTOR_SIZE=4096

#
# Wi-Fi Provisioning Manager
#
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16

#
# Supplicant
#
CONFIG_WPA_WPS_WARS=
CONFIG_WPA_DEBUG_PRINT=

```

akshar001, 2021-02-17:
This sdkconfig utilises PSRAM with MBEDTLS

me-no-dev, 2021-02-18:
duplicate?

akshar001, 2021-02-18:
Duplicate @me-no-dev

closed 2021-02-18 by akshar001

ayhamalji2, 2025-04-08:
hello I have just came across your config file solution is it possible to know how did you generate that and if differs from model to model. I am using esp32-s3-wroom-1-n16r8 so if the file differs how can I change it so it can fit my model ?