espressif/arduino-esp32 #4828: Unnecessary compile error when CONFIG_BLUEDROID_ENABLED is not defined

william-ferguson-au, 2021-02-18:
### Hardware:
Board:							ESP32 Wrover
Core Installation version:			ESP-IDF 4.3, Arduino-ESP32 v4.2
Computer OS:                                         Windows 10

### Description:
Trying to b Build esp-idf app with arduino-esp32 as a component.
When I do I get a compilation failure

```
../components/arduino-esp32/libraries/WiFiProv/src/WiFiProv.cpp: In function 'void get_device_service_name(prov_scheme_t, char*, size_t)':
../components/arduino-esp32/libraries/WiFiProv/src/WiFiProv.cpp:59:23: error: 'WIFI_PROV_SCHEME_BLE' was not declared in this scope
     if(prov_scheme == WIFI_PROV_SCHEME_BLE) {
                       ^~~~~~~~~~~~~~~~~~~~
../components/arduino-esp32/libraries/WiFiProv/src/WiFiProv.cpp:59:23: note: suggested alternative: 'WIFI_PROV_SCHEME_MAX'
     if(prov_scheme == WIFI_PROV_SCHEME_BLE) {
                       ^~~~~~~~~~~~~~~~~~~~
                       WIFI_PROV_SCHEME_MAX
.
```

This is occurring because I have configured BT and BLE off as I only intend to support Soft AP provisioning (and in fact I'm not even using the Arduino-sp32 provisioning because using the esp-idf directly is so easy). Because CONFIG_BLUEDROID_ENABLED is not defined, the  WIFI_PROV_SCHEME_BLE is not declared so it cannot be referenced in WifiProv.cpp

But the crazy thing is, WIFI_PROV_SCHEME_BLE doesn't need to be referenced because both paths of the if and #if generate the same code.

me-no-dev, 2021-02-18:
I see you already provided a fix :) will merge soon!

tsctrl, 2021-02-21:
i am getting the same issue, thank s for the quick fix

r2d2r01, 2021-03-26:
File components\arduino-esp32\tools\sdk\esp32\include\config\sdkconfig.h:
#define CONFIG_BT_BLUEDROID_ENABLED 1
#define CONFIG_BLUEDROID_ENABLED CONFIG_BT_BLUEDROID_ENABLED

closed 2021-07-11 by stale[bot]