espressif/arduino-esp32 #4830: Beacon Timeout IN STA mode WIFI

kayleejacob1, 2021-02-18:
I have a program that is send data to AWS but I keep getting a beacon timeout. The looping is it working fine then out of know where it disconnects with beacon timeout.  This program has been working fine i just add more data in the upload. 

[I][AWS_IOT.cpp:1455] aws_iot_yeld_task(): looping
[I][AWS_IOT.cpp:1455] aws_iot_yeld_task(): looping
[I][AWS_IOT.cpp:1455] aws_iot_yeld_task(): looping
[I][AWS_IOT.cpp:1455] aws_iot_yeld_task(): looping
E (49455) aws_iot: erro with ssl -76
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:348] _eventCallback(): Reason: 200 BEACON_TIMEOUT
E (49458) aws_iot: failed! mbedtls_ssl_write returned -0x4e
[I][AWS_IOT.cpp:533] disconnectCallbackHandler(): MQTT Disconnect
[I][AWS_IOT.cpp:542] disconnectCallbackHandler(): Auto Reconnect is enabled, Reconnecting attempt will start now
Disconnected from internet!
[I][AWS_IOT.cpp:1444] aws_iot_yeld_task(): Reconnecting
E (50587) aws_iot: failed! mbedtls_net_connect returned -0x52
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 4 - STA_CONNECTED
Connected to internet!
[I][AWS_IOT.cpp:1444] aws_iot_yeld_task(): Reconnecting
[D][WiFiGeneric.cpp:332] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:376] _eventCallback(): STA IP: 192.168.20.9, MASK: 255.255.255.0, GW: 192.168.20.1
[I][AWS_IOT.cpp:1444] aws_iot_yeld_task(): Reconnecting
E (54124) aws_iot: failed! mbedtls_ssl_handshake returned -0x10
[I][AWS_IOT.cpp:1444] aws_iot_yeld_task(): Reconnecting
[I][AWS_IOT.cpp:1444] aws_iot_yeld_task(): Reconnecting

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2021-02-18:
`WiFi.setSleep(false);`
I'm not sure that modem sleep is causing what you are seeing, as your log could just as easily be caused by a poor WiFi connection that keeps getting disconnected.  Do you have the problem less frequently if you move or reorient the device?

kayleejacob1, 2021-02-18:
I'm am within 2 feet of the modem is i got to an older version of the code no issue but my differences in the code are just a few lines i commented out still issue

kayleejacob1, 2021-02-18:
I stand corrected it doe seem to be hardware issue i change antennas seem to fix the issue.  I have been using this board for over a year no issues but i guess things break.  I just overlooked it thanks.

kayleejacob1, 2021-02-18:
So it was a fluke that the antenna was causing it because it is back.  What other things could be causing this Beacon Timeout Error.

kayleejacob1, 2021-02-18:
I have checked the network with analyzer and the worst signal from AP is -25 average -18 to the ESP chip

kayleejacob1, 2021-02-18:
I tried the above WiFi.setSleep(false); nothing I tried turning it on as well.  I also used the Max Sleep Function and set the interval beacon to 500ms when i did this started to miss sends. 

What core is the WIFI driver on could i be over using the processor?

What if i am using the PsRAM alot could this be causing issue if so is there a work around?

kayleejacob1, 2021-02-19:
I am going to reopen this with clear information

closed 2021-02-19 by kayleejacob1

chipweinberger, 2021-09-15:
Did you ever fix this issue? I'm hitting it too.

MichielfromNL, 2023-02-03:
same with me. Using wificlientsecure in 2.0.0 (embedded in PLatformio current) 
I have sketch with 2 SSL connections. One works flawless, the other one keeps hitting this issue.

`22:28:49.499 > [177138][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
22:28:49.507 > [177145][E][ssl_client.cpp:37] _handle_error(): [start_ssl_client():278]: (-76) UNKNOWN ERROR CODE (004C)
22:28:49.517 > [177148][E][WifiClientSecure.cpp:135] connect(): start_ssl_client: -76
22:28:49.523 > [177155][W][HTTPClient.cpp:1469] returnError(): error(-1): connection refused
22:28:49.530 > [177158][E][Opensky.cpp:162] GetData(): UNKNOWN ERROR CODE (004C)
22:28:49.536 > [177164][E][main.cpp:651] loop(): Opensky Failed code 5, retry
22:28:49.542 > [177174][W][WiFiGeneric.cpp:955] _eventCallback(): Reason: 202 - AUTH_FAIL`

chipweinberger, 2023-02-04:
Just to update, I no longer hit this anymore. I don't recall ever actively fixing it. Im now on v4.4.3.

MichielfromNL, 2023-02-05:
Turned out to be stalled hardware @ my end.
Powered my esp off/on instead of reboot, and problem went away.