espressif/arduino-esp32 #4833: WebServer Slow Performance

9ae8sdf76, 2021-02-19:
### Hardware:
Board:							ESP32 wroom
Core Installation version:			**How can I find out?**
IDE name:						Platform.io 5.1.0 (via CLion 2020.3.2)
Flash Frequency:					40Mhz
PSRAM enabled:                                    no
Upload Speed:					115200
Computer OS:                                        Mac OSX Catalina (10.15.7 (19H512))
Browsers:                                               Chrome (88.0.4324.150); Firefox (85.0.2 (64-bit)); Safari (iOS 14.4)

### Description:
WebServer in WiFi station mode is extraordinarily slow to the point it's unusable. WebServer in WiFi AP mode works as expected and is fast. 

The following are the things I've tried:

* Set the [`HTTP_MAX_*_WAIT`](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.h#L44-L47) times down to zero
* Manually add the [`delay(1)` to `WebServer::handleClient()`](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.cpp#L284-L289)
* [Commented out the block of code](https://github.com/espressif/arduino-esp32/blob/master/libraries/WebServer/src/WebServer.cpp#L316-L321) dealing with Chrome browsers

Originally the website wouldn't load at all in Chrome, and the `HTTP_MAX_*_WAIT` change [resolved the connection reset errors](https://github.com/espressif/arduino-esp32/issues/4230) I was receiving on my external resources. (Please revisit that ticket)

My website is currently 236kb in total, spread across 5 HTML pages. The largest file is a Javascript file (bootstrap and friends) and comes in at 177kb. It can take anywhere between 30-50s to load that one file. My time-to-first-byte typically is around 5s. I understand the requests are sequential and single-threaded, but in AP mode the page and resources load in about 1.2s consistently and across all browsers. This appears to only happen when the WiFi is connected in station mode. 

### Sketch:
This will take a little time to create. I am currently using [ConfigManager](https://github.com/nrwiersma/ConfigManager) to create the web portal, and while it is mostly a thin proxy around WebServer, I will create a simple example that demonstrates the problem.

### Debug Messages:
#### Station Mode
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
Waiting for WiFi to connect.[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.42.71, MASK: 255.255.255.0, GW: 192.168.42.1

Connected to nebakanezer with 192.168.42.71
Webserver enabled on port: 80
[V][WebServer.cpp:287] handleClient(): New client
[V][WebServer.cpp:287] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /wifi.html search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.42.71
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: keep-alive
[V][Parsing.cpp:219] _parseRequest(): headerName: Pragma
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: Cache-Control
[V][Parsing.cpp:220] _parseRequest(): headerValue: no-cache
[V][Parsing.cpp:219] _parseRequest(): headerName: DNT
[V][Parsing.cpp:220] _parseRequest(): headerValue: 1
[V][Parsing.cpp:219] _parseRequest(): headerName: Upgrade-Insecure-Requests
[V][Parsing.cpp:220] _parseRequest(): headerValue: 1
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/88.0.4324.150 Safari/537.36
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept
[V][Parsing.cpp:220] _parseRequest(): headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Encoding
[V][Parsing.cpp:220] _parseRequest(): headerValue: gzip, deflate
[V][Parsing.cpp:219] _parseRequest(): headerName: Accept-Language
[V][Parsing.cpp:220] _parseRequest(): headerValue: en-US,en;q=0.9
[V][Parsing.cpp:219] _parseRequest(): headerName: sec-gpc
[V][Parsing.cpp:220] _parseRequest(): headerValue: 1
[V][Parsing.cpp:247] _parseArguments(): args: 
[V][Parsing.cpp:230] _parseRequest(): Request: /wifi.html
[V][Parsing.cpp:231] _parseRequest():  Arguments: 
[V][WebServer.cpp:287] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /css/index.css search: 
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.42.71
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[... 338 lines cut ...]
[V][Parsing.cpp:113] _parseRequest(): method: GET url: /hotspot-detect.html search:
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: captive.apple.com
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: close
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: CaptiveNetworkSupport-390.60.1 wispr
[V][Parsing.cpp:247] _parseArguments(): args:
[V][Parsing.cpp:230] _parseRequest(): Request: /hotspot-detect.html
[V][Parsing.cpp:231] _parseRequest():  Arguments:
[E][WebServer.cpp:618] _handleRequest(): request handler not found
Unknown URL: captive.apple.com
[V][WebServer.cpp:287] handleClient(): New client
[V][Parsing.cpp:113] _parseRequest(): method: GET url: / search:
[V][Parsing.cpp:219] _parseRequest(): headerName: Host
[V][Parsing.cpp:220] _parseRequest(): headerValue: 192.168.1.1:80
[V][Parsing.cpp:219] _parseRequest(): headerName: Connection
[V][Parsing.cpp:220] _parseRequest(): headerValue: close
[V][Parsing.cpp:219] _parseRequest(): headerName: User-Agent
[V][Parsing.cpp:220] _parseRequest(): headerValue: CaptiveNetworkSupport-390.60.1 wispr
[V][Parsing.cpp:247] _parseArguments(): args:
[V][Parsing.cpp:230] _parseRequest(): Request: /
[V][Parsing.cpp:231] _parseRequest():  Arguments:
[W][SPIFFS.cpp:54] begin(): SPIFFS Already Mounted!
```

9ae8sdf76, 2021-02-24:
Is there not anybody on the team willing to help me with my issue?

drws, 2021-03-01:
You can get the SDK version from a running program with [ESP.getSdkVersion()](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/Esp.h).

For starters, you can check for memory leaks/heap fragmentation with ESP.getFreeHeap() and/or ESP.getMaxAllocHeap() by comparing their output in both modes.

You could also try the [esp-idf v4.2 branch](https://github.com/espressif/arduino-esp32/tree/idf-release/v4.2) and don't forget to post results if you do :)

9ae8sdf76, 2021-03-01:
@drws Thank you for the reply! Below is the information you requested:

```
SDK Version: v3.3.4-428-gd8082b7f3-dirty
Free / Max Heap: 127212 / 66440
```

Subsequent requests to a simple index page:
```
Free / Max Heap: 123340 / 66440
Free / Max Heap: 116960 / 66440
Free / Max Heap: 115588 / 66440
Free / Max Heap: 115376 / 66440
```

Unfortunately, when upgrading to v4.2 I get the error [`xtensa-esp32-elf-g++: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'`](https://github.com/espressif/esp-idf/issues/1879), so I cannot go any further with that branch at the moment.

9ae8sdf76, 2021-03-08:
@drws any ideas on how to proceed?

RafigRzayev, 2021-03-10:
@9ae8sdf76 I also had slow internet issues, maybe this will help you https://github.com/espressif/arduino-esp32/issues/4529 

Check the instructions in the very bottom.

9ae8sdf76, 2021-03-11:
@RafigRzayev FYI, if I understand the rules for includes, you can copy the `sdkconfig.h` into your project, set an `-I .` in your build, and the compiler will use your copy over the vendor version. 

But, unfortunately for me, despite making the changes you suggested; it did improve things slightly, as before making the changes it would take ~30s to download 69.6 kB (gzip compressed HTML and assets), and after making the changes it now takes ~20s. Slight improvement, but I would imagine it should be more instantaneous. 

I'm happy to continue to proverbially throw things at the wall to see what sticks. :)

RafigRzayev, 2021-03-11:
@9ae8sdf76 Hello. What you said is true for esp-idf, but not true for arduino. Any change that you do in sdkconfig won't have any affect on your build, because the libraries are already precompiled. That's why you didn't have effect in the speed. Please try the way I suggested (there are two instructions in the post, follow the last one) and it should work. Your download should take just a second after that process.

9ae8sdf76, 2021-03-11:
@RafigRzayev I don't use the Arduino IDE. I use PlatformIO and the arduino framework. That said, I've spent the past 4 hours attempting to reproduce your instructions for the Arduino IDE and have failed to get it to work as you've described. 

Using your first instructions, when trying to retrieve a web page over the WiFi using the WebServer, I get a `Guru Meditation Error` and the backtrace points to an issue in lwip (`tcpip_thread -> lwip_netconn_do_delconn -> lwip_netconn_do_close_internal` specifically). 

When attempting to use your updated instructions, the build fails because it cannot find the `app_main` symbol. Outside of being on a Mac, I've followed your instructions very closely and do not understand enough to know how to fix the issue. 

At this point I feel like I've given this process a good faith effort, but that effort is far too complicated and convoluted to be considered a reasonable workaround for me. I don't find replacing vendor libraries a sustainable endeavor.

I do really appreciate your help and I'm still open to suggestions; especially if there is a way to get PlatformIO to pickup the sdkconfig changes, then I'm all for it so long as I can continue to use the arduino framework.

RafigRzayev, 2021-03-11:
Hello again @9ae8sdf76 ,

Since you are using Mac, the process of generating updated libraries should be easier for you than me regarding sdkconfig for Arduino IDE, because there is an official espressif tool called lib-builder. It is supported on Linux and Mac - that's why I had to struggle for Windows and make long instructions.

Here are the links:
 https://github.com/espressif/esp32-arduino-lib-builder 
https://github.com/espressif/arduino-esp32/blob/master/docs/lib_builder.md

>  I don't find replacing vendor libraries a sustainable endeavor. 

I think you shouldn't perceive updating libraries as a custom "tweak". What I mean is that in esp-idf it is actually normal process to use your custom settings for sdkconfig, there is an official tool with GUI for it. I'm just saying it to emphasize that it is pretty normal to use your custom sdkconfig. 

When it comes to Arduino, the libraries come precompiled, that's the library build tool in the above links  is the way to go in my opinion. The default sdkconfig parameters on Arduino IDE will allow you max 50kB/s(from my own experience). 

> Using your first instructions

Sorry for this one, but first instruction is not that good, I think it is better to follow second instruction. 

> When attempting to use your updated instructions, the build fails because it cannot find the app_main symbol

There is a specific configuration to enable and disable arduino main loop. It must be the origin of this error if I am not mistaken. Try to reverse the current configuration for this one. It must be under Arduino configurations. 

The lib-builder tool experience also changes for me after arduino updates, I was not able to generate custom libraries on later version myself, was getting many different errors. I think you encountered the same problems.  That's why, if your problem is not fixed by disabling/enabling arduino main loop, I suggest you following:
I will share custom libraries that I have generated for myself (during writing that WiFi speed post). You can replace your libraries with them just to be sure that it indeed solves your problem. And afterwards if it suits your needs, you can try get help here from espressif employees, or this esp32-arduino chat https://gitter.im/espressif/arduino-esp32 regarding the proper way to build the libraries.

Please note that you need to download specific version of Arduino IDE so that it matches with my custom libraries. Download the version 1.0.5 - RC4. In the original link with wifi instructions, it is explained how to download this particular version, I think this link should work for Additional Board Manager URL - https://github.com/espressif/arduino-esp32/releases/download/1.0.5-rc4/package_esp32_dev_index.json. Here is a dropbox link to my libraries: https://www.dropbox.com/s/qf8vsexax03zeg6/lib.rar?dl=0

Please let me know once you are finished with the dropbox link (or in case you don't need it), so that I can remove it. 

I also went trough this pain when dealing with sdkconfig, so don't give up :)

lbernstone, 2021-03-11:
@RafigRzayev If you want to use lib-builder on Windows, I generate Docker containers that will work on Windows. https://hub.docker.com/r/lbernstone/esp32-arduino-lib-builder

drws, 2021-03-11:
Some additional ideas to check out:
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-buffer-usage
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#how-to-improve-wi-fi-performance
- https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/lwip.html#performance-optimization

Also this issue is a few days older than arduino-esp32 v1.0.5 release. Have you tried your code with the latest version of the framework?

closed 2021-07-09 by stale[bot]