espressif/arduino-esp32 #4834: The Watchdog got triggered.

SocketNet, 2021-02-19:
### Hardware:
Board:							Esp32
Core Installation version:			3.0.0
IDE name:						Arduino framework + Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                      no
Upload Speed:				        115200
Computer OS:                                          Mac OSX

### Description:
Interrupt wdt timeout

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>
#include <Preferences.h>
Preferences fs;

cleanWifiConfig()
{
    fs.remove("ssid");
    fs.remove("psk");
    Serial.println("Restarting...");
    ESP.restart();
}

void buttonISR()
{
 delay(50);
  if (!digitalRead(KEY_BUILTIN))
  {
    cleanWifiConfig();
  }
}

void setup() {
    fs.begin("WifiConfig");
    pinMode(KEY_BUILTIN, INPUT_PULLUP);
    attachInterrupt(KEY_BUILTIN, buttonISR, ONLOW);
}

void loop() {
}
```

### Debug Messages:
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)
Core 0 register dump:
PC      : 0x40089858  PS      : 0x00060034  A0      : 0x80089fc5  A1      : 0x3ffbbfc0  
A2      : 0x3ffbf334  A3      : 0xb33fffff  A4      : 0x00060023  A5      : 0x003fffff  
A6      : 0x00060021  A7      : 0x00000000  A8      : 0x0000cdcd  A9      : 0x0000cdcd  
A10     : 0x0000abab  A11     : 0x00060023  A12     : 0x00060021  A13     : 0x3ffbc070  
A14     : 0x00000003  A15     : 0x00060623  SAR     : 0x00000018  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x40089858:0x3ffbbfc0 0x40089fc2:0x3ffbbff0 0x4008bc00:0x3ffbc010 0x4008bbb6:0x3ffbc030 0x4008c90b:0x00000000

Core 1 register dump:
PC      : 0x4008bd5e  PS      : 0x00060534  A0      : 0x8008af1b  A1      : 0x3ffbe460  
A2      : 0x3ffb7ddc  A3      : 0x3ffbc768  A4      : 0x00000001  A5      : 0x00000001  
A6      : 0x00060523  A7      : 0x00000000  A8      : 0x3ffbc768  A9      : 0x3ffbc768  
A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000020  A13     : 0x3ffbc1c8  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000002  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008bd5e:0x3ffbe460 0x4008af18:0x3ffbe480 0x400891a3:0x3ffbe4a0 0x400dc406:0x3ffbe4e0 0x400dc4c6:0x3ffbe500 0x40086b82:0x3ffbe520 0x400870b6:0x3ffbe540 0x400e6415:0x3ffbe5b0 0x400e4ce5:0x3ffbe5d0 0x400e5133:0x3ffbe5f0 0x400e3ac9:0x3ffbe660 0x400e44f2:0x3ffbe6b0 0x400e33ce:0x3ffbe710 0x400d602c:0x3ffbe750 0x400d123c:0x3ffbe770 0x400d126a:0x3ffbe790 0x40080fb1:0x3ffbe7b0 0x400847a5:0x3ffbe7d0 0x40147f93:0x3ffbc650 0x400dad87:0x3ffbc670 0x4008ab91:0x3ffbc690 0x4008939d:0x3ffbc6b0

Rebooting...

chegewara, 2021-02-19:
With `ONLOW`, when you press button then ISR will be triggered multiple times, but you have delay(50) in ISR callback.
Also try backtrace encoder.

lbernstone, 2021-02-19:
ISRs cannot have delays.  They must execute quickly.
```
bool buttonPressed = false;

void buttonISR() {
    buttonPressed = true;
}

void setup() {
    attachInterrupt(KEY_BUILTIN, buttonISR, ONLOW);
}

void loop() {
    if (buttonPressed) {
       cleanWiFiConfig();
       buttonPressed = false;
    }
    delay(50);
}
```

closed 2021-02-20 by SocketNet