espressif/arduino-esp32 #4835: HardwareSerial esp32-Wrover-kit Uart2

francescot92, 2021-02-19:
hi, 
I try to use UART2 of the esp32-Wrover-E with Arduino IDE. i know that the UART0 (pin 3 RX pin 1 TX) is for programing UART1(pin 9 RX pin 10 TX) for flash and UART2 (pin 16 RX and pin 17 TX) is free, but in this case these pins (16 and 17) are used to PSRAM.

From datasheet any pins can be used for the UART throuth the library HardwareSerial.h but it seem not work.

i try to communicate two micro throuth UART. i show you the sketch:

this a part of HardwareSerial.cpp:

void HardwareSerial::begin(unsigned long baud, uint32_t config, int8_t rxPin, int8_t txPin, bool invert, unsigned long timeout_ms)
{
    if(0 > _uart_nr || _uart_nr > 2) {
        log_e("Serial number is invalid, please use 0, 1 or 2");
        return;
    }
    if(_uart) {
        end();
    }
    if(_uart_nr == 0 && rxPin < 0 && txPin < 0) {
        rxPin = 3;
        txPin = 1;
    }
    if(_uart_nr == 1 && rxPin < 0 && txPin < 0) {
        rxPin = RX1; //RX1
        txPin = TX1; //TX1
    }
    if(_uart_nr == 2 && rxPin < 0 && txPin < 0) {
        rxPin = 33; // pin used to UART2
        txPin = 32;
    }

my sketch:
#include <HardwareSerial.h>

HardwareSerial myUart(2);

String messaggio = "NULL";
char terminatore = '\r';

void setup()
{
  // Setup Serial connection:
  Serial.begin(115200);
  myUart.begin(115200);
  delay(1000);
}

void loop
{
  if(myUart.available()>0)
    {
      Serial.print("available:ok\n");
      messaggio = Serial2.readStringUntil(terminatore);
     }
......

I don’t get anything.
Can you help me??

lbernstone, 2021-02-19:
You can assign uart devices to any I/O pins you like.
```
gps_rx=21;
gps_tx=22;
Serial1.begin(9600, SERIAL_8N1, gps_rx, gps_tx);
```
The external device tx pin is going to be the rx pin on the mcu (and vice versa).

francescot92, 2021-02-19:
i use the 115200 bound rate.
i know but i don't get anything. why?? is there anything wrong??

francescot92, 2021-02-19:
I get characters different from what I expect why?

chegewara, 2021-02-19:
Few things:
- UART1 is not used for flash,
- you dong have to create new HardwareSerial object, there are 3 already in arduino( Serial, Serial1, Serial2),
- your code:
```
void loop
{
if(myUart.available()>0)   <--------myUart
{
Serial.print("available:ok\n");
messaggio = Serial2.readStringUntil(terminatore); <---------Serial2
}
```
- i dont know what kind gps do you have, but most usually is using baudrate 9600, so worth to try

francescot92, 2021-02-19:
i don't use gps, i'm sorry the mistake. it is only copy trouble. 
now it work but I get characters different from what I expect why?
i expect for example: Hello get random characters 

i show it in serial port:

17:27:15.657 -> }Ս⸮

chegewara, 2021-02-19:
It looks like bitrate does not match. You can try to set it to 0 in begin.

francescot92, 2021-02-19:
i have tried it but change the result:

17:45:57.645 -> ⸮

francescot92, 2021-02-19:
I noticed that those characters I received with the mass disconnected while, with the mass connected.

francescot92, 2021-02-19:
mass = GND

chegewara, 2021-02-19:
Yes, it is important to connect GND between 2 devices.

francescot92, 2021-02-19:
Yes i know but with the gnd connetted between two device i don’t get anything. Has rx pins needed the pull up?? Mybe that is the reason...

chegewara, 2021-02-19:
Time to use scope, logic analyzer and maybe try to swap pins, with the same code.

allexotecnologia, 2021-04-26:
Been looking through the docs of the WROVER E module, but haven't found the definition of UART1 and UART2 interfaces.. Can someone inform me if they're still available in the module? We are familiar with the HardwareSerial definitions, but in all docs of othr modules, we found the definition in one or another pin. In this one, haven't found them.

We've been using WROVER B, but are looking to upgrade to the newest version.

chegewara, 2021-04-26:
WROVER E is just new revison, nothing changed with UARTs.

allexotecnologia, 2021-04-28:
@chegewara Thanks for the info. Yet, is not that clear for me. So basically the WROVER-E module has the same 3 UART pattern on the pinout, is just not explicitly said in the docs?

Thanks in advance!

chegewara, 2021-04-28:
> So basically the WROVER-E module has the same 3 UART pattern on the pinout,

Yes

closed 2021-07-13 by stale[bot]

almeidab99, 2024-05-17:
Good night. I am trying to connect an as608 fingerprint sensor to the ESP32Wrover-E, according to some specifications that I read online, they indicated the following UART pins:
**UART1_TX GPIO_NUM_27
UART1_RX GPIO_NUM_14**
------------------------------------------
**UART2_TX GPIO_NUM_2
UART2_RX GPIO_NUM_15**
------------------------------------------
I'm not entirely sure friends, could you help me identify the correct UART pins on the ESPWrover-E board, so I can communicate with my fingerprint sensor?

The source code to read fingerprints, which I am using, is the following:

`#include <Adafruit_Fingerprint.h>

#define UART_RX_PIN 15 // GPIO 15  para RX
#define UART_TX_PIN 14 // GPIO 14  para TX
Adafruit_Fingerprint finger = Adafruit_Fingerprint(&Serial2);

void setup()
{
  Serial.begin(57600);
  Serial2.begin(57600, SERIAL_8N1, UART_RX_PIN, UART_TX_PIN);  //Se utiliza para la comunicación con el sensor de huellas dactilares de Adafruit. 
  while (!Serial);
  delay(100);
  Serial.println("\n\n Prueba de detección de huellas");
  // establecer la velocidad de datos para el puerto serie del sensor
  //finger.begin(57600);

  if (finger.verifyPassword()) {
    Serial.println("¡Sensor de huellas dactilares encontrado!");
  } else {
    Serial.println("No se encontró el sensor de huellas dactilares :(");
    
    while (1) 
    { 
      delay(1); 
    }

  }

  finger.getTemplateCount();
  Serial.print("El sensor contiene "); Serial.print(finger.templateCount); Serial.println(" plantillas");
  Serial.println("Esperando un dedo válido...");
}

void loop()                     // se ejecuta una y otra vez
{
  
  getFingerprintIDez();
  delay(50);            // no es necesario ejecutar esto a toda velocidad.
    
}

uint8_t getFingerprintID() {
  uint8_t p = finger.getImage();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Imagen capturada");
      break;
    case FINGERPRINT_NOFINGER:
      Serial.println("No se detectó dedo");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Error de comunicación");
      return p;
    case FINGERPRINT_IMAGEFAIL:
      Serial.println("Error de imagen");
      return p;
    default:
      Serial.println("Error desconocido");
      return p;
  }

  // ¡Éxito OK!

  p = finger.image2Tz();
  switch (p) {
    case FINGERPRINT_OK:
      Serial.println("Imagen convertida");
      break;
    case FINGERPRINT_IMAGEMESS:
      Serial.println("Imagen demasiado desordenada");
      return p;
    case FINGERPRINT_PACKETRECIEVEERR:
      Serial.println("Error de comunicación");
      return p;
    case FINGERPRINT_FEATUREFAIL:
      Serial.println("No se pudieron encontrar características de la huella dactilar");
      return p;
    case FINGERPRINT_INVALIDIMAGE:
      Serial.println("No se pudieron encontrar características de la huella dactilar");
      return p;
    default:
      Serial.println("Error desconocido");
      return p;
  }

  // ¡OK convertido!
  p = finger.fingerFastSearch();
  if (p == FINGERPRINT_OK) {
    Serial.println("¡Se encontró una coincidencia de huella!");
  } else if (p == FINGERPRINT_PACKETRECIEVEERR) {
    Serial.println("Error de comunicación");
    return p;
  } else if (p == FINGERPRINT_NOTFOUND) {
    
    Serial.println("No se encontró ninguna coincidencia");
    return p;
    
  } else {
    Serial.println("Error desconocido");
    return p;
  }

  // ¡se encontró una coincidencia!
  Serial.print("ID encontrado #"); Serial.print(finger.fingerID);
  Serial.print(" con una confianza de "); Serial.println(finger.confidence);

  return finger.fingerID;
}

// devuelve -1 si falla, de lo contrario, devuelve el ID #
int getFingerprintIDez() {
  uint8_t p = finger.getImage();

  if (p != FINGERPRINT_OK) 
  return -1;
  
  p = finger.image2Tz();
  if (p != FINGERPRINT_OK)
  return -1;

  p = finger.fingerFastSearch();
  if (p != FINGERPRINT_OK)  
  return -1;

  // ¡se encontró una coincidencia!
  Serial.print("ID encontrado #"); Serial.print(finger.fingerID);
  Serial.print(" con una confianza de "); Serial.println(finger.confidence);
  
  return finger.fingerID;

}
`