espressif/arduino-esp32 #4837: How to add pcnt_get_event_status

calkoe, 2021-02-19:
Hello,

the Method pcnt_get_event_status is missing in this SDK, 
the PCNT ISR Handler ist quite useless without it.

Is there a way to add id ?

Method is available in the ESP-IDF:
https://github.com/espressif/esp-idf/blob/b0150615dff529662772a60dcb57d5b559f480e2/components/driver/pcnt.c

lbernstone, 2021-02-19:
arduino-esp32 is currently based on idf 3.2/3.3
https://github.com/espressif/esp-idf/blob/release/v3.3/components/driver/include/driver/pcnt.h
The experimental branch (idf-release/v4.2) has the function you want.

calkoe, 2021-02-19:
Thank you verry much !

calkoe, 2021-02-19:
Haw can i bring this to work in PlatformIO ?

I've updated changed the platform_packages to Version 4.2 but now i'm reciving this error:
`xtensa-esp32-elf-g++: error: unrecognized command line option '-mfix-esp32-psram-cache-strategy=memw'`
 
**platformio.ini:**
[env:nodemcu-32s]
platform                = espressif32
platform_packages =
  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#idf-release/v4.2
  framework-espidf @ https://github.com/espressif/esp-idf.git#release/v4.2
board                   = nodemcu-32s
framework               = arduino
upload_speed            = 921600
;build_flags             = -O2
monitor_speed           = 115200
board_build.f_cpu       = 160000000L

I think i need to update the xtensa-esp32-elf-g++, how can i do this ?

closed 2021-07-08 by stale[bot]