espressif/arduino-esp32 #4842: Error when reinitializing WiFi after deinitialization with WiFi.mode( WIFI_MODE_NULL )

GiovanniCmpaner, 2021-02-20:
Previous issue: [4421](https://github.com/espressif/arduino-esp32/issues/4421)
Even after fix [80e9e42](https://github.com/espressif/arduino-esp32/commit/80e9e42c3b7a8ad646cd8803c9f4f84e35c9ebb1), I'm still getting
```
[ 10769][V][WiFiGeneric.cpp:315] _arduino_event_cb(): AP Started
...
[ 10800][E][WiFiGeneric.cpp:983] mode(): Could not set mode! 12289
```
This line: https://github.com/espressif/arduino-esp32/blob/beedeea4541116106b38fc5c3a03821cdf6fe288/libraries/WiFi/src/WiFiGeneric.cpp#L568
Should logically be: 
`lowLevelInitDone = !(esp_wifi_deinit() == ESP_OK);`

cardtap, 2021-04-21:
Hi @GiovanniCmpaner,
Did you ever solve this problem? I am facing the same issue of 
[E][WiFiGeneric.cpp:983] mode(): Could not set mode! 12289
When switching from WPA2 standard network to WPA2 enterprise network...

Thanks!

GiovanniCmpaner, 2021-04-21:
As I indicated on the first post, you need to modify the mentioned line locally on your source code.
The downside is, whenever you update or sync the framework libraries, you will have to modify it again.

closed 2021-07-08 by stale[bot]

garageeks, 2021-07-24:
The issue appears to be present in version 2.0.0RC1.
This is an extract of my wi-fi initialization code (yeah needs some cleanup):

```
void loop () {
	if(WiFi.status() != WL_CONNECTED || httpFailedCount > 5) enableWiFi();
}

void enableWiFi () {

		if(httpFailedCount > 5) {
			
			Serial.println("Too many failed transmission, restart Wi-Fi");
			wifiResetCounter++;
			WiFi.mode(WIFI_OFF); 
			delay(2000);
		}

		Serial.print("Connecting to AP...");

		int wait_time = 0;
			
		delay(1);
		WiFi.mode(WIFI_STA);
		if(routerIsRebooting || httpFailedCount > 5) {
			if(routerIsRebooting) routerIsRebooting = false;
			WiFi.begin(ssid, password);
			delay(150);
			char devicename[20]="Station-";
			char tempValue[5];
			itoa(stationID,tempValue,10);
			strcat(devicename, tempValue);
			WiFi.setHostname(devicename);
		} else {
			WiFi.reconnect();
		}  
		delay(500);

		while (WiFi.status() != WL_CONNECTED) {
			delay(500);
			Serial.print(".");
			if(wait_time > 10) break;
			wait_time++;
		}

		if(WiFi.status() == WL_CONNECTED) {     //WiFi AP is available, go ahead
			Serial.println("succeeded");

		} else {                                //WiFi not available, increment failed counter
			Serial.println("failed");
			httpFailedCount++;
			WiFi.mode(WIFI_OFF);            //switch it off since it didn't connect. Will try again later   
		}
	}	
}  
```

I get these errors:

```
Connecting to AP...............failed
Connecting to AP...[152878][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
............failed
Connecting to AP...[183923][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
............failed
Connecting to AP...[213842][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
............failed
Connecting to AP...[244885][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
............failed
Connecting to AP...[274802][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
............failed
Too many failed transmission, restart Wi-Fi
Connecting to AP...[307855][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
[307855][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
[307856][E][WiFiSTA.cpp:162] begin(): STA enable failed!
............failed
Too many failed transmission, restart Wi-Fi
Connecting to AP...[337805][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
[337805][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
[337806][E][WiFiSTA.cpp:162] begin(): STA enable failed!
............failed
Too many failed transmission, restart Wi-Fi
Connecting to AP...[368891][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
[368891][E][WiFiGeneric.cpp:996] mode(): Could not set mode! 12289
[368892][E][WiFiSTA.cpp:162] begin(): STA enable failed!
............failed
```

With the change suggested by OP in WiFiGeneric.cpp, line 580, it just works with no errors
```
static bool wifiLowLevelDeinit(){
    if(lowLevelInitDone){
    	lowLevelInitDone = **not(**esp_wifi_deinit() == ESP_OK**)**;		//change by nick https://github.com/espressif/arduino-esp32/issues/4842
    }
    return true;
}
```