espressif/arduino-esp32 #4844: analogRead() breaks WiFi connection on AI-Thinker ESP-12K (ESP32-S2)

letung3105, 2021-02-21:
### Hardware:
Board:							AI-Thinker ESP-12K (ESP32-S2)
Core Installation version:			idf-release/v4.2
IDE name:						Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    yes
Upload Speed:					921600
Computer OS:                                        Manjaro 20.2

### Description:

I'm trying to get some analog inputs while using WiFi with the ESP32S2, but the WiFi keeps disconnect and could not reconnect in a few iterations after an `analogRead` is performed. The value returned from `analogRead` also appeared to be bogus.

The following configuration is used for PlatformIO:
1. The platform packages is a fork of `idf-release/v4.2` that applies fixes to i2c initialization as specified [here](https://github.com/espressif/arduino-esp32/pull/4776#issuecomment-779496327)

```
[env:nodemcu-esp32s2-aithinker-esp12k]
platform = espressif32
platform_packages = framework-arduinoespressif32 @ git@github.com:letung3105/arduino-esp32.git#idf-release/v4.2
framework = arduino
board = nodemcu-32s
board_build.mcu = esp32s2
upload_speed = 921600
monitor_speed = 115200
```

### Sketch:

```cpp
#include <Arduino.h>
#include <FS.h>
#include <SPIFFS.h>
#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <Wire.h>

const int TEST_GPIO = 10;
WiFiClientSecure wifiClient;

void setup()
{
  pinMode(TEST_GPIO, INPUT);
  Serial.begin(115200);
  while (!Serial)
    ;
  if (!SPIFFS.begin())
  {
    return;
  }

  WiFi.begin("ssid", "password");
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(500);
  }
  Serial.println("connected!");

  File awsRootCACert = SPIFFS.open("/creds/aws/rootca-cert.pem");
  File awsClientCert = SPIFFS.open("/creds/aws/device-cert.crt");
  File awsClientKey = SPIFFS.open("/creds/aws/device-key.key");

  wifiClient.loadCACert(awsRootCACert, awsRootCACert.size());
  wifiClient.loadCertificate(awsClientCert, awsClientCert.size());
  wifiClient.loadPrivateKey(awsClientKey, awsClientKey.size());
}

void loop()
{
[... 111 lines cut ...]
[ 63092][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 63099][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 63107][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 65174][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: MySSID, BSSID: 00:00:00:00:00:00, Reason: 201
[ 65174][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 65182][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 65189][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
Reconnecting...
[ 66191][V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 97799
[ 66191][V][ssl_client.cpp:61] start_ssl_client(): Starting socket
[ 66195][E][WiFiGeneric.cpp:1168] hostByName(): DNS Failed for a2p23x3rl0tb19-ats.iot.ap-southeast-1.amazonaws.com
[ 66205][E][WiFiClientSecure.cpp:132] connect(): start_ssl_client: -1
[ 66211][V][ssl_client.cpp:259] stop_ssl_socket(): Cleaning SSL connection.
UNKNOWN ERROR CODE (0001)
[ 67256][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: MySSID, BSSID: 00:00:00:00:00:00, Reason: 201
[ 67257][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 67264][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 67271][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 69338][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: MySSID, BSSID: 00:00:00:00:00:00, Reason: 201
[ 69339][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 69346][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 201 - NO_AP_FOUND
[ 69353][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0

[......]
```

lbernstone, 2021-02-21:
esp32-s2 is much less tolerant of high voltage than esp32.  Make sure you don't supply >3V6 on any pin.
This works fine for me:
```
void setup() {
  Serial.begin(115200);
  pinMode(17, OUTPUT);
  pinMode(10, INPUT);
}

void loop() {
  dacWrite(17, 0);
  delay(100);
  Serial.printf("analogRead: %d\n", analogRead(10));
  delay(50);
  dacWrite(17, 128);
  delay(100);
  Serial.printf("analogRead: %d\n", analogRead(10));
  delay(50);
  dacWrite(17, 255);
  delay(100);
  Serial.printf("analogRead: %d\n", analogRead(10));
  delay(50);
}
```

```
analogRead: 296
analogRead: 5352
analogRead: 8191
analogRead: 297
analogRead: 5349
analogRead: 8191
```

letung3105, 2021-02-21:
The problem appears to happen with every pins on ADC1

lbernstone, 2021-02-21:
Confirmed.  Also happens with adc1_get_raw.  I'll try a full IDF test to see if the problem is upstream.

lbernstone, 2021-02-26:
@me-no-dev I don't know the new stack well enough to even troubleshoot this.  Here's simple code that will cause the WiFi to fail after ~60 seconds.  A simple test in IDF (I can give you code if you would like) does not disconnect or kill the WiFi connection.
```
#include <WiFi.h>
#include <driver/adc_common.h>

void setup() {
  Serial.begin(115200);
  WiFi.begin();
  WiFi.waitForConnectResult();
  adc1_config_channel_atten(ADC1_CHANNEL_9, ADC_ATTEN_DB_11);
}

void loop() {
  delay(100);
  adc1_get_raw(ADC1_CHANNEL_9);
}
```

Debug:
```
[   906][V][WiFiGeneric.cpp:273] _arduino_event_cb(): STA Started
[   907][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 2 - STA_START
[   909][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[  3020][V][WiFiGeneric.cpp:285] _arduino_event_cb(): STA Connected: SSID: larryb, BSSID: c0:56:27:6a:f5:c6, Channel: 8, Auth: WPA2_PSK
[  3022][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[  3268][V][WiFiGeneric.cpp:295] _arduino_event_cb(): STA Got New IP:192.168.1.121
[  3269][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  3272][D][WiFiGeneric.cpp:844] _eventCallback(): STA IP: 192.168.1.121, MASK: 255.255.255.0, GW: 192.168.1.1
[ 69201][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: larryb, BSSID: c0:56:27:6a:f5:c6, Reason: 200
[ 69202][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 69209][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
[ 69217][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
[ 71311][V][WiFiGeneric.cpp:290] _arduino_event_cb(): STA Disconnected: SSID: larryb, BSSID: 00:00:00:00:00:00, Reason: 201
[ 71312][D][WiFiGeneric.cpp:795] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[ 71319][W][WiFiGeneric.cpp:816] _eventCallback(): Reason: 201 - NO_AP_FOUND
```

me-no-dev, 2021-03-01:
I have not updated the S2 branch recently, I suspect a fix has been pushed :) will update these days and we will see

newHeiko, 2021-03-14:
Just stumbled across this: There seems to be an upstream bug for this as well: https://github.com/espressif/esp-idf/issues/6682

Looking forward to a fix,
Heiko

MS71, 2021-03-15:
Same with adc1 ch0, ch1,ch2 and ch3

lienbacher, 2021-03-22:
I've also just run into this problem. Is there any fix for this yet?

newHeiko, 2021-04-27:
FWIW, it works for me (tm) with 5d9b98c9b04e591044078ac2304da95e8c7e997b - that's 2.0.0-alpha1 and also the first to be installed through the arduino boards manager.

Heiko

closed 2021-07-13 by stale[bot]