espressif/arduino-esp32 #4846: Wifi shiel not present

thomazitos, 2021-02-21:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
Describe your system( Hardware, computer, O/S, core version, environment).
Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 CAM AI THINKER
Core Installation version:	                1.6.13
IDE name:						Arduino IDE?
Flash Frequency:					40Mhz
PSRAM enabled:                                      yes
Upload Speed:					9600
Computer OS:                                        Windows 10

### Description:
When the board arrived here, first thing I tried was the camera web server sketch. I worked just fine. After some time, I reinstalled the arduino IDE. Now, when I try to run the exemple ScanNetworks sketch the message display is "WIFI shiel not present". I have tried EVERYTHING, what makes me think the board is broken(?). I would apreciate a lot if some one could help.

### Sketch: /*

 This example  prints the Wifi shield's MAC address, and
 scans for available Wifi networks using the Wifi shield.
 Every ten seconds, it scans again. It doesn't actually
 connect to any network, so no encryption scheme is specified.

 Circuit:
 * WiFi shield attached

 created 13 July 2010
 by dlf (Metodo2 srl)
 modified 21 Junn 2012
 by Tom Igoe and Jaymes Dec
 */

#include <SPI.h>
#include <WiFi.h>

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(9600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  // check for the presence of the shield:
  if (WiFi.status() == WL_NO_SHIELD) {
    Serial.println("WiFi shield not present");
    // don't continue:
    while (true);
  }

  String fv = WiFi.firmwareVersion();
  if (fv != "1.1.0") {
    Serial.println("Please upgrade the firmware");
  }

  // Print WiFi MAC address:
  printMacAddress();
}

void loop() {
  // scan for existing networks:
  Serial.println("Scanning available networks...");
  listNetworks();
  delay(10000);
}

void printMacAddress() {
  // the MAC address of your Wifi shield
  byte mac[6];

  // print your MAC address:
  WiFi.macAddress(mac);
  Serial.print("MAC: ");
  Serial.print(mac[5], HEX);
  Serial.print(":");
  Serial.print(mac[4], HEX);
  Serial.print(":");
  Serial.print(mac[3], HEX);
  Serial.print(":");
  Serial.print(mac[2], HEX);
  Serial.print(":");
  Serial.print(mac[1], HEX);
  Serial.print(":");
  Serial.println(mac[0], HEX);
}

void listNetworks() {
  // scan for nearby networks:
  Serial.println("** Scan Networks **");
  int numSsid = WiFi.scanNetworks();
  if (numSsid == -1) {
    Serial.println("Couldn't get a wifi connection");
    while (true);
  }

  // print the list of networks seen:
  Serial.print("number of available networks:");
  Serial.println(numSsid);

  // print the network number and name for each network found:
  for (int thisNet = 0; thisNet < numSsid; thisNet++) {
    Serial.print(thisNet);
    Serial.print(") ");
    Serial.print(WiFi.SSID(thisNet));
    Serial.print("\tSignal: ");
    Serial.print(WiFi.RSSI(thisNet));
    Serial.print(" dBm");
    Serial.print("\tEncryption: ");
    printEncryptionType(WiFi.encryptionType(thisNet));
  }
}

void printEncryptionType(int thisType) {
  // read the encryption type and print out the name:
  switch (thisType) {
    case ENC_TYPE_WEP:
      Serial.println("WEP");
      break;
    case ENC_TYPE_TKIP:
      Serial.println("WPA");
      break;
    case ENC_TYPE_CCMP:
      Serial.println("WPA2");
      break;
    case ENC_TYPE_NONE:
      Serial.println("None");
      break;
    case ENC_TYPE_AUTO:
      Serial.println("Auto");
      break;
  }
}
void setup() {
}

void loop() {
}

```

lbernstone, 2021-02-21:
You don't have a WiFi Shield, you have an esp32.  Use examples from esp32 (eg WiFiScan.ino).

thomazitos, 2021-02-21:
I can't find this example. Can you explain the meaning of this message?

lbernstone, 2021-02-21:
https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiScan/WiFiScan.ino

thomazitos, 2021-02-21:
I uplouded every librery on this file. But I keep reciving this error:Arduino: 1.8.9 (Windows 10), Placa:"AI Thinker ESP32-CAM"

In file included from C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiSTA.h:28:0,

                 from C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFi.h:32,

                 from C:\Program Files (x86)\Arduino\libraries\WiFi\examples\WiFiScan\WiFiScan.ino:6:

C:\Program Files (x86)\Arduino\libraries\WiFi\src/WiFiGeneric.h:31:39: fatal error: wifi_provisioning/manager.h: No such file or directory

compilation terminated.

exit status 1
Erro compilando para a placa AI Thinker ESP32-CAM.
 I apreciate your help and patience. I hope we solve this problem! (:

me-no-dev, 2021-02-22:
you must have installed this repository wrong. If you used Git instead of the board manager, then your hardware folder where this repo is installed must be in `C:\Program Files (x86)\Arduino\espressif\esp32`. If it is not in a folder called `esp32`, then Arduino IDE will pick the wrong platform and will include the wrong WiFi lib.

closed 2021-02-22 by me-no-dev